gtxaspec/wz_mini_hacks #85: ssh inbound is fine after setting up the keys... scp outbound doesn't work... see my simple fix

rickgitdone, 2022-06-14:
scp outbound... the temp fix

[root@wyzeV3cam:configs]# scp [local file] user@IP:[file]
/opt/wz_mini/tmp/.bin/dbclient: No such file or directory
lost connection

[root@WyzeV3:configs]# cp /opt/wz_mini/bin/dbclient /opt/wz_mini/tmp/.bin/

[root@wyzeV3cam:configs]# scp  [local file] user@IP:[file]
/opt/wz_mini/tmp/.bin/dbclient: Warning: failed creating //.ssh: Read-only file system

Host 'IP' is not in the trusted hosts file.
(ssh-ed25519 fingerprint SHA256:...)
Do you want to continue connecting? (y/n) y
user@IP's password:                                     << outbound connection Now works >>

gtxaspec, 2022-06-15:
this is a bug in the `default_options.h` section of dropbear, I will fix this.

gtxaspec, 2022-06-16:
fixed in latest release