gtxaspec/wz_mini_hacks #88: iptables / netfilter module to support "-m state"?

claudobahn, 2022-06-14:
I can't seem to use a command like the following: `iptables -A INPUT -i wlan0 -m state --state ESTABLISHED,RELATED -j ACCEPT`, which I believe I've tracked down to a missing kernel module, since even `iptables -A INPUT -m state --state NEW` fails with:
```
iptables: No chain/target/match by that name.
```

Unfortunately I'm not quite sure which module is missing, hopefully you've got an idea. If not, I can try to do some more research.

I do have `ENABLE_IPTABLES="true"` in my `wz_mini.conf` file.

gtxaspec, 2022-06-15:
looking into this now

gtxaspec, 2022-06-15:
You are right, the `connection tracking` module is enabled on the t20 side, but not the t31.  I'll release an update tonight with the updated kernel and modules.

gtxaspec, 2022-06-15:
this may not be possible.  there appears to be a bug in either the kernel version from the SDK, or the wifi driver.  When the connection tracking module is enabled, the wifi driver crashes.  When it's compiled into the kernel, it still crashes.

the state module you need depends on the connection tracking module :\

gtxaspec, 2022-06-15:
good news, I found an alternate driver that works, https://linux-sunxi.org/Wifi#RTL8189FTV

gtxaspec, 2022-06-15:
initial testing the new driver works, let me do some further testing to double check

gtxaspec, 2022-06-16:
fixing this breaks compatibility with the PANv2, waiting on a fix... lets delay this for now, in the mean time I will modify wz_mini to prepare for this.

gtxaspec, 2022-06-22:
good news! I have new wifi drivers for rtl8189ftv working on V2, v3, and the atbm603x chipsets, all working with connection tracking now... some more testing and then we can give this a whirl

gtxaspec, 2022-06-23:
the https://github.com/gtxaspec/wz_mini_hacks/tree/wz_mini_wifi branch has full support for iptables/conntrack/etc.

make sure not to disable the new drivers in the config, otherwise the kernel will crash on boot with the old drivers.

gtxaspec, 2022-06-25:
available now in current release

closed 2022-06-25 by gtxaspec