gtxaspec/wz_mini_hacks #100: New V2/V3 wifi driver testing wanted!

gtxaspec, 2022-06-17:
Hi,

the latest releases include a new wifi driver, 8189fs, that replaces the rtl8189ftv driver included by the factory.  

this can be enabled in the config by setting `ENABLE_RTL8189FS_DRIVER="true"`

This driver works on the V2 and V3 only for now.

I would appreciate some feedback, if you find it improved performance, or makes it worse.  It seems to work better for me, both in throughput and connectivity.

archandanime, 2022-06-17:
Run from V3:
ENABLE_RTL8189FS_DRIVER="false"
`[root@WCV3:mmc]# ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1): 56 data bytes
64 bytes from 192.168.10.1: seq=0 ttl=64 time=6.785 ms
64 bytes from 192.168.10.1: seq=1 ttl=64 time=35.036 ms
64 bytes from 192.168.10.1: seq=2 ttl=64 time=2.056 ms
64 bytes from 192.168.10.1: seq=3 ttl=64 time=44.395 ms
64 bytes from 192.168.10.1: seq=4 ttl=64 time=1.534 ms
64 bytes from 192.168.10.1: seq=5 ttl=64 time=27.162 ms
64 bytes from 192.168.10.1: seq=6 ttl=64 time=17.707 ms
64 bytes from 192.168.10.1: seq=7 ttl=64 time=6.825 ms
64 bytes from 192.168.10.1: seq=8 ttl=64 time=10.741 ms
64 bytes from 192.168.10.1: seq=9 ttl=64 time=23.203 ms
64 bytes from 192.168.10.1: seq=10 ttl=64 time=9.135 ms
64 bytes from 192.168.10.1: seq=11 ttl=64 time=15.106 ms
64 bytes from 192.168.10.1: seq=12 ttl=64 time=1.454 ms
64 bytes from 192.168.10.1: seq=13 ttl=64 time=60.342 ms
64 bytes from 192.168.10.1: seq=14 ttl=64 time=30.049 ms
64 bytes from 192.168.10.1: seq=15 ttl=64 time=15.366 ms
64 bytes from 192.168.10.1: seq=16 ttl=64 time=29.637 ms
64 bytes from 192.168.10.1: seq=17 ttl=64 time=46.333 ms
64 bytes from 192.168.10.1: seq=18 ttl=64 time=22.135 ms
64 bytes from 192.168.10.1: seq=19 ttl=64 time=3.294 ms
64 bytes from 192.168.10.1: seq=20 ttl=64 time=6.921 ms
64 bytes from 192.168.10.1: seq=21 ttl=64 time=21.406 ms
64 bytes from 192.168.10.1: seq=22 ttl=64 time=84.365 ms
64 bytes from 192.168.10.1: seq=23 ttl=64 time=5.362 ms
64 bytes from 192.168.10.1: seq=24 ttl=64 time=7.379 ms
64 bytes from 192.168.10.1: seq=25 ttl=64 time=35.423 ms
64 bytes from 192.168.10.1: seq=26 ttl=64 time=14.244 ms
64 bytes from 192.168.10.1: seq=27 ttl=64 time=19.382 ms
64 bytes from 192.168.10.1: seq=28 ttl=64 time=6.173 ms
^C
--- 192.168.10.1 ping statistics ---
29 packets transmitted, 29 packets received, 0% packet loss
round-trip min/avg/max = 1.454/20.998/84.365 ms`
ENABLE_RTL8189FS_DRIVER="true"
`[root@WCV3:~]# ping 192.168.10.1
PING 192.168.10.1 (192.168.10.1): 56 data bytes
64 bytes from 192.168.10.1: seq=0 ttl=64 time=18.097 ms
64 bytes from 192.168.10.1: seq=1 ttl=64 time=27.193 ms
64 bytes from 192.168.10.1: seq=2 ttl=64 time=47.498 ms
64 bytes from 192.168.10.1: seq=3 ttl=64 time=12.492 ms
64 bytes from 192.168.10.1: seq=4 ttl=64 time=1.268 ms
64 bytes from 192.168.10.1: seq=5 ttl=64 time=6.910 ms
64 bytes from 192.168.10.1: seq=6 ttl=64 time=56.284 ms
64 bytes from 192.168.10.1: seq=7 ttl=64 time=6.889 ms
64 bytes from 192.168.10.1: seq=8 ttl=64 time=7.948 ms
64 bytes from 192.168.10.1: seq=9 ttl=64 time=1.052 ms
64 bytes from 192.168.10.1: seq=10 ttl=64 time=23.370 ms
64 bytes from 192.168.10.1: seq=11 ttl=64 time=18.362 ms
64 bytes from 192.168.10.1: seq=12 ttl=64 time=43.314 ms
64 bytes from 192.168.10.1: seq=13 ttl=64 time=4.172 ms
64 bytes from 192.168.10.1: seq=14 ttl=64 time=11.032 ms
64 bytes from 192.168.10.1: seq=15 ttl=64 time=2.132 ms
64 bytes from 192.168.10.1: seq=16 ttl=64 time=1.054 ms
64 bytes from 192.168.10.1: seq=17 ttl=64 time=4.835 ms
64 bytes from 192.168.10.1: seq=18 ttl=64 time=1.062 ms
64 bytes from 192.168.10.1: seq=19 ttl=64 time=9.109 ms
64 bytes from 192.168.10.1: seq=20 ttl=64 time=4.444 ms
64 bytes from 192.168.10.1: seq=21 ttl=64 time=6.125 ms
64 bytes from 192.168.10.1: seq=22 ttl=64 time=4.657 ms
64 bytes from 192.168.10.1: seq=23 ttl=64 time=1.352 ms
64 bytes from 192.168.10.1: seq=24 ttl=64 time=21.227 ms
64 bytes from 192.168.10.1: seq=25 ttl=64 time=3.590 ms
64 bytes from 192.168.10.1: seq=26 ttl=64 time=1.358 ms
64 bytes from 192.168.10.1: seq=27 ttl=64 time=2.444 ms
64 bytes from 192.168.10.1: seq=28 ttl=64 time=1.133 ms
64 bytes from 192.168.10.1: seq=29 ttl=64 time=13.797 ms
64 bytes from 192.168.10.1: seq=30 ttl=64 time=1.875 ms
64 bytes from 192.168.10.1: seq=31 ttl=64 time=5.299 ms
64 bytes from 192.168.10.1: seq=32 ttl=64 time=3.864 ms
64 bytes from 192.168.10.1: seq=33 ttl=64 time=22.386 ms
64 bytes from 192.168.10.1: seq=34 ttl=64 time=3.266 ms
^C
--- 192.168.10.1 ping statistics ---
35 packets transmitted, 35 packets received, 0% packet loss
round-trip min/avg/max = 1.052/11.454/56.284 ms
`

gtxaspec, 2022-06-17:
@archandanime what model camera is this on?

tavicu, 2022-06-17:
Hei. I have updated one camera to the latest release and tried this feature.
The camera is very close to the router, don't know if that makes a difference.

But in my case it was worse with the feature on.

ENABLE_RTL8189FS_DRIVER="false"
```
--- 192.168.50.1 ping statistics ---
21 packets transmitted, 21 packets received, 0% packet loss
round-trip min/avg/max = 1.300/10.796/35.859 ms
```

ENABLE_RTL8189FS_DRIVER="true"
```
--- 192.168.50.1 ping statistics ---
21 packets transmitted, 21 packets received, 0% packet loss
round-trip min/avg/max = 1.665/24.603/122.322 ms
```

skilau, 2022-06-17:
I have a few v3's, of different vintages.

I have one with the atbm603x_wifi_sdio chipset with a MAC starting with: D0:3F:27:
I have one with the rtl8189ftv chipset with a MAC starting with 7C:78:B2:

Both MACs are assigned to Wyze of course, but I wonder if the change in chipset follows the change in MAC.

Anyway, I will give the new driver a whirl on my rtl8189ftv one.

yeahme49, 2022-06-17:
> I have a few v3's, of different vintages.
>
> I have one with the atbm603x_wifi_sdio chipset with a MAC starting with: D0:3F:27: I have one with the rtl8189ftv chipset with a MAC starting with [...]

I just checked one of my V3 cameras, it has rtl8189ftv chipset and the MAC starts with d0:3f:27 so it doesn't appear that the MAC address corresponds to the chipset

virmaior, 2022-06-18:
all 5 of mine have rtl8189ftv , I have MAC addresses in both ranges. (not currently using wifi so not in a situation to test them)

archandanime, 2022-06-18:
> @archandanime how old is this v3?

it's lastest firmware(4.36.9.139)
I have dd partition images of the v4.36.3.19 if you want me to test. I just don't know how to re-flash them

rickgitdone, 2022-06-19:
I loaded the new realtek module and don't see much of a difference in the network performance... on 4.36.9.139

clee604, 2022-06-19:
I tested new driver on a V2, pings looked about the same, but some performance differences in iperf3 (newer driver seems to have more bandwidth)

driver = false
[image]

driver = true
[image]

gtxaspec, 2022-06-23:
this branch https://github.com/gtxaspec/wz_mini_hacks/tree/wz_mini_wifi includes new drivers for both the rtl8189FTV and the atbm6032i, so it should work across all camera models.  

feel free to test, I will roll this into the master branch soon.  both drivers perform much better than the stock drivers, as well as the older 8189fs I was using.

gtxaspec, 2022-06-23:
this branch has newer drivers than master at the moment.

yes, stock driver is the original factory driver.

gtxaspec, 2022-06-23:
already planned 😁

capncybo, 2022-06-23:
Ahhh cr@p!!!
I can't "easily" play with this new branch remotely from work.
I'm probably going to have to do some real work instead... Dammit ;-)

capncybo, 2022-06-24:
Testing & so far I AGREE... " both drivers perform much better than the stock drivers, as well as the older 8189fs"
Noticing improved performance, thx.

gtxaspec, 2022-06-25:
available now in current release

closed 2022-06-25 by gtxaspec