gtxaspec/wz_mini_hacks #102: RTPS: Cannot init device:/dev/video2

michaelgaultjr, 2022-06-17:
Hello,

I'm having some trouble getting RTSP working on my Wyze Cam v3 Spotlight camera (Firmware version 4.61.0.3).

`wz_init.log` and `wz_post.log` and didn't have any errors, but I found this error in `wz_user.log`

```
[NOTICE] /root/CLEAN/v4l2rtspserver/main.cpp:297
        Version: 0.3.3-4-ga43e094-dirty live555 version:2022.04.26
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:135
        /dev/video2 support output
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
        /dev/video2 support capture
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
        /dev/video2 support read/write
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
        /dev/video2 support streaming
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video2: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video2: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:192
        /dev/video2: Cannot get format Invalid argument
[ERROR] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:83
        Cannot init device:/dev/video2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
        Device /dev/video2+ event=n
```

Manually running `LD_LIBRARY_PATH=/opt/wz_mini/lib /opt/wz_mini/bin/v4l2rtspserver -U username:password -P 8554 /dev/video2` gives the same error.

Full `wz_user.log`
```
+ echo 'welcome to wz_user.sh'
welcome to wz_user.sh
+ echo 'PID 119'
PID 119
+ '[[' -e /tmp/dbgflag ]]
+ export 'WZMINI_CFG=/opt/wz_mini/wz_mini.conf'
+ '[' -f /opt/wz_mini/wz_mini.conf ]
+ source /opt/wz_mini/wz_mini.conf
+ WEB_CAM_ENABLE=false
+ WEB_CAM_BIT_RATE=8000
+ WEB_CAM_FPS_RATE=25
+ CUSTOM_HOSTNAME=WCV3SPOTLIGHT
+ ENABLE_USB_ETH=false
+ ENABLE_USB_ETH_MODULES=asix
+ ENABLE_USB_DIRECT=false
+ USB_DIRECT_MAC_ADDR=02:01:02:03:04:08
+ ENABLE_USB_RNDIS=false
+ ENABLE_IPV6=false
+ ENABLE_WIREGUARD=false
+ WIREGUARD_IPV4=
+ WIREGUARD_PEER_ENDPOINT=
+ WIREGUARD_PEER_PUBLIC_KEY=
+ WIREGUARD_PEER_ALLOWED_IPS=
+ WIREGUARD_PEER_KEEP_ALIVE=
+ ENABLE_IPTABLES=false
+ ENABLE_NFSv4=true
+ ENABLE_RTL8189FS_DRIVER=false
+ REMOTE_SPOTLIGHT=false
+ REMOTE_SPOTLIGHT_HOST=0.0.0.0
+ RTSP_LOGIN=[redacted]
+ RTSP_PASSWORD=[redacted]
+ RTSP_PORT=8554
+ RTSP_HI_RES_ENABLED=false
[... 332 lines cut ...]
+ event=n
+ directory=/tmp
+ file=28.mp4
+ date
Fri Jun 17 21:28:01 UTC 2022
+ '[[' 28.mp4 '==' img ]]
+ event=n
+ directory=/tmp
+ file=29.mp4
+ date
Fri Jun 17 21:29:01 UTC 2022
+ '[[' 29.mp4 '==' img ]]
+ event=n
+ directory=/tmp
+ file=30.mp4
+ date
Fri Jun 17 21:30:01 UTC 2022
+ '[[' 30.mp4 '==' img ]]
+ event=n
+ directory=/tmp
+ file=31.mp4
+ date
Fri Jun 17 21:31:01 UTC 2022
+ '[[' 31.mp4 '==' img ]]
```

gtxaspec, 2022-06-17:
need more logs.  run gather_wz_logs.sh, you can check the logs there or share here so we can analyze then

michaelgaultjr, 2022-06-17:
Not exactly sure what to look for so here's the logs zip

[log_gather_2022-06-17_T2206.tar.gz](https://github.com/gtxaspec/wz_mini_hacks/files/8931537/log_gather_2022-06-17_T2206.tar.gz)

gtxaspec, 2022-06-18:
the problem is your firmware version, go on the wyze website and download the latest firmware, flash it, and it should work.  there's an issue using the more with the old rtsp firmware version

michaelgaultjr, 2022-06-18:
> a

That worked, however I cannot view the stream. I'm using VLC with this URL `rtsp://[username]:[password]@192.168.1.[redacted]:8554/unicast` and it just times out and doesn't give any errors, any other URL fails.

closed 2022-06-18 by michaelgaultjr