gtxaspec/wz_mini_hacks #98: Wyze cam v3 rtsp stops responding

jemershaw, 2022-06-17:
I have a few cams that are having problems with rtsp. They seem to have the rtsp process running but when connecting I get a black screen. Wyze app doesn't have a problem viewing the video. I'll try to see if I can trace this but I'm running on the latest firmware and release.

dudududodododedede, 2022-06-17:
How are you viewing the feed? Have you tried VLC? What do the logs from VLC say? What do the logs from wz_mini show?

gtxaspec, 2022-06-17:
not sure if you run audio, but when you get the black screen, does audio still go through?

jemershaw, 2022-06-17:
The audio still goes through. I'll have to test on my computer to get the vlc output but from my phone. It's weird because I have 6 cameras and only two so far are having the problem.

jemershaw, 2022-06-18:
@gtxaspec is there a way to restart the services without rebooting via ssh?

gtxaspec, 2022-06-18:
not at the moment, the entire camera has to be restarted for things to work right typically

jemershaw, 2022-06-18:
I killed the rtsp process and then run `/bin/sh /opt/wz_mini/etc/init.d/wz_init.sh` and everything worked for an unknown time. Trying to figure out what the video stops displaying but the audio still works.

gtxaspec, 2022-06-18:
what firmware version and camera model?

gtxaspec, 2022-06-18:
on one of the ones that you are getting no video on, do this:

1. pull the memory card
2. format it fresh FAT32
3. download the latest release
4. don't restore any old config files or anything, do everything new from scratch
5. once you have configured your stuff, put the sd card back in
6. boot the camera, test the video.

if you still get no video, after you experience the blackout video, run `gather_wz_logs.sh` and upload the logs here so I can take look to see whats going on.

jemershaw, 2022-06-20:
I'm running 4.36.9.139 and it's a cam v3. I'll try what you suggested hopefully tomorrow! Thanks for the info and help!

gtxaspec, 2022-06-28:
Did you ever resolve this? Working well on my end.

gtxaspec, 2022-06-30:
closing due to lack or information, open a new ticket if you still have issues!

closed 2022-06-30 by gtxaspec