gtxaspec/wz_mini_hacks #108: running "cmd aplay .wav file" results in "error"

archandanime, 2022-06-19:
There's no other error messages beside "error"
the music.wav is playable on my Arch linux laptop
> [root@WCV3:mmc]# cmd aplay music.wav
> error

If any information I need to provide, please tell.

gtxaspec, 2022-06-19:
do you have rtsp enabled?

gtxaspec, 2022-06-19:
if you do, run `logread | grep command`  and see if you get any entries relating to aplay

gtxaspec, 2022-06-19:
it also needs to be 1 ch, s16le pcm format, 16000hz if on v3, 8000hz if on v2.

archandanime, 2022-06-19:
yes, RTSP is enabled

> [root@WCV3:mmc]# logread | grep command
> May 14 03:22:39 iCamera: [command] v4l2_device_path = /dev/video2
> May 14 03:22:39 iCamera: [command] video product 640x320Unable to set V4L2 device video format: -1
> May 14 03:22:39 iCamera: [command] v4l2_device_path = /dev/video1
> May 14 03:22:39 iCamera: [command] video product 1920x1080[sdk,0686]dbg: chn:0 sdkaudio_algorithm_init(16000,2)
> Jun 19 10:26:11 iCamera: [command] video capture ch0 on
> Jun 19 10:26:11 iCamera: [command] audio capute on
> Jun 19 10:29:06 iCamera: [command] aplay: file:time_to_say_goodbye.wav
> Jun 19 10:29:06 iCamera: [command] aplay err: fopen time_to_say_goodbye.wav failed!
>

gtxaspec, 2022-06-19:
can you try to play another file? try something in /opt/wz_mini/usr/share/audio

archandanime, 2022-06-19:
with RTSP off, the "cmd aplay .wav 100" doesn't give any error message but there's no audio out either though I put my ears very close to the speaker
same results with files in /opt/wz_mini/usr/share/audio

gtxaspec, 2022-06-19:
ah, i think the issue is you have to provide a full path to the audio file, try `cmd aplay /opt/wz_mini/usr/share/audio/binbin_v3.wav 70`

archandanime, 2022-06-19:
Running cmd aplay /opt/wz_mini/usr/share/audio/binbin_v3.wav 70 works!
However running my own audio file time_to_say_goodbye.wav makes a very high pitch noise and the music doesn't play at all

gtxaspec, 2022-06-19:
Looks like its a file format error if your file doesn't play, let me dig up the ffmpeg command for conversion:

`ffmpeg -i in.wav -acodec pcm_s16le -ac 1 -ar 16000 out.wav`

archandanime, 2022-06-19:
it works flawlessly now!
Just convert the audio file as you said and scp to the camera SD card, it works!

gtxaspec, 2022-06-19:
awesome! enjoy.  You can also play streaming audio too:

https://github.com/gtxaspec/wz_mini_hacks/discussions/74

archandanime, 2022-06-19:
The README.md should be updated to avoid future same bug report I think

gtxaspec, 2022-06-19:
pull request merged!

closed 2022-06-19 by gtxaspec

HeyItsDaddy, 2024-10-18:
Side note for anyone else that may happen across this:  If you accidentally play a bad file, every file after that (even good ones) will throw an error.  Just simply type reboot to reboot your camera and it'll fix it right up, though.