gtxaspec/wz_mini_hacks #110: Custom Script Not Executing

jacoblukewood, 2022-06-19:
Using a V3 camera I am not able to get a custom script to run on boot.

The script is I am attempting to run is the motion detection web hook caller from this issue: https://github.com/gtxaspec/wz_mini_hacks/issues/23#issuecomment-1136712191_

The CUSTOM_SCRIPT variable is set to  "/media/mmc/scriptToRun.sh"

Is it correct to use an absolute path?

archandanime, 2022-06-21:
Did you forget to chmod +x the /media/mmc/scriptToRun.sh file? :p

gtxaspec, 2022-06-28:
Did you ever get this working? Testing the latest version, it works well.

gtxaspec, 2022-06-30:
closing due to lack or information, open a new ticket if you still have issues!

closed 2022-06-30 by gtxaspec

jacoblukewood, 2022-07-02:
> Did you forget to chmod +x the /media/mmc/scriptToRun.sh file? :p

I had made the script executable via the root account using SSH however am still not seeing the script execute.

This is the script I am attempting to execute from boot at /media/mmc/motiondetection.sh:
`#!/opt/wz_mini/bin/bash

while true
do
  ret=`cmd waitmotion 1000 | grep "timeout" | wc -l`

  if [ $ret -eq 0  ] ; then
    echo "Reporting motion!"
    wget --post-data="motion=1" http://10.0.1.90:11080/endpoint/21/public/ -O/dev/null 2>/dev/null
  else
    #optionally actively report timeouts
    echo "No motion found!"
  fi
done`

And this is the .conf file at /media/mmc/wz_mini/wz_mini.conf:
`#wz_mini configuration

#### W E B CAMERA###
##THIS MODE DISABLES EVERYTHING AND IT WILL
## WORK AS A WEB CAMERA FOR YOUR PC ***ONLY***
WEB_CAM_ENABLE="false"
WEB_CAM_BIT_RATE="8000"
WEB_CAM_FPS_RATE="25"

#####NETWORKING#####
HOSTNAME="WCV3"

ENABLE_USB_ETH="false"
ENABLE_USB_ETH_MODULES="asix"

ENABLE_USB_DIRECT="false"
USB_DIRECT_MAC_ADDR="02:01:02:03:04:08"

ENABLE_USB_RNDIS="false"

ENABLE_IPV6="false"

ENABLE_WIREGUARD="false"
WIREGUARD_IPV4=""
WIREGUARD_PEER_ENDPOINT=""
WIREGUARD_PEER_PUBLIC_KEY=""
WIREGUARD_PEER_ALLOWED_IPS=""
WIREGUARD_PEER_KEEP_ALIVE=""

ENABLE_IPTABLES="false"

ENABLE_NFSv4="false"

#####ACCESSORIES#####
REMOTE_SPOTLIGHT="false"
REMOTE_SPOTLIGHT_HOST="0.0.0.0"

#####VIDEO STREAM#####
RTSP_LOGIN="admin"
RTSP_PASSWORD=""
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"
RTSP_HI_RES_FPS="30"
RTSP_HI_RES_MAX_BITRATE="4096"
RTSP_HI_RES_TARGET_BITRATE="2048"
RTSP_HI_RES_ENC_PARAMETER="2"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_FPS=""
RTSP_LOW_RES_MAX_BITRATE=""
RTSP_LOW_RES_TARGET_BITRATE=""
RTSP_LOW_RES_ENC_PARAMETER=""

#####RTMP STREAMING#####
#### REQUIRES RTSP ####
### AND RTSP AUDIO ###
RTMP_STREAM_ENABLED="false"
RTMP_STREAM_FEED="video1_unicast"
RMTP_STREAM_SERVICE="youtube"
RTMP_STREAM_DISABLE_AUDIO="false"
RTMP_STREAM_YOUTUBE_KEY=""
RTMP_STREAM_TWITCH_KEY=""
RTMP_STREAM_FACEBOOK_KEY=""

#####GENERAL#####
ENABLE_SWAP="true"
ENABLE_USB_STORAGE="false"
ENABLE_EXT4="false"
ENABLE_CIFS="false"
DISABLE_FW_UPGRADE="true"
AUDIO_PROMPT_VOLUME="50"
ENABLE_MP4_WRITE="false"

#####DEBUG#####
#drops you to a shell via serial, doesn't load app_init.sh
DEBUG_ENABLED="false"
DEBUG_INITRAMFS_ENABLED="false"
DEBUG_PASSWORD="false"

#####SCRIPTING#####
CUSTOM_SCRIPT_PATH="/media/mmc/motiondetection.sh"
`

jacoblukewood, 2022-07-02:
If I run the script from the command line `/media/mmc/motiondetection.sh` then it works successfully.

gtxaspec, 2022-07-02:
if you check the wz_user.log, what does it say about the custom script

jacoblukewood, 2022-07-02:
`/media/mmc/wz_mini/log/wz_user.log`

```
+ echo 3
+ '[' -f /media/mmc/motiondetection.sh ]
+ echo 'custom script not found'
custom script not found
+ echo 'wz_user.sh done'
log level:500
```

gtxaspec, 2022-07-02:
the /media/mmc path may not be created yet by iCamera when your script is attempting to execute, change it to `/opt/motiondetection.sh`

jacoblukewood, 2022-07-02:
I've modified the path and the I'm no longer seeing the above error in the log, however, the script still does not seem to function.

Is there a way I can delete the old log files?

jacoblukewood, 2022-07-02:
Could it be because the script is an infinite loop but doesn't continue to run after wz_user.sh completes?

gtxaspec, 2022-07-02:
the script is run with & which shoves in the background...
 add `sleep 10` right before the while loop statement to delay execution a bit and we if that works.

jacoblukewood, 2022-07-02:
Success! Thank you!