gtxaspec/wz_mini_hacks #115: Is it too late to add Overlayfs support?

archandanime, 2022-06-21:
Do you still keep the plan to add Overlayfs support?(mount / as overlayfs)
I think that using it makes the code shorter(no more mount --bind files) but realized it's a lot of work to modify existing code.
Also is it done simply by modifying atomcams_tool/initramfs_skeleton/init so the kernel mounts the / as overlayfs before switch_root to it?

archandanime, 2022-06-21:
Sounds wonderful!

virmaior, 2022-06-21:
@archandanime I'd think the bigger problem is how the T31 works. The T31 is a combination of MIPS and RISC-V (http://www.ingenic.com.cn/en/?product/id/20.html). I don't know enough to answer definitively but my understanding is that some parts of what's running are proprietary and interact with parts of the part processor for which there is no open code (see https://www.cnx-software.com/2020/04/26/ingenic-t31-ai-video-processor-combines-xburst-1-mips-and-risc-v-lite-cores/ )

gtxaspec, 2022-06-21:
it's the same older version of overlayfs.  I'll start work on adding overlayfs soon

gtxaspec, 2022-06-21:
> @gtxaspec from my understanding, it's not possible to compile a newer version kernel to work with current(Wyze's) firmware because it's incompatible with old packages, is it correct? Is that why in developer_environment.md you said we must use 3.10.x kernel option for building the kernel? Another question, what can we possibly do if Wyze makes their firmware SDK public? does it make backporting easier?

The SDK released by the original design manufacturer (ODM), hualai, is based off kernel 3.10.14.  This is the version which is provided to their customer, wyze, as well as anyone else who uses a T31 SoC in their designs.  The SDK is available online from multiple sources, you can google for it. 

Ingenic's newer platform, X1000/X1500E has support in the mainline kernel (5.x), and **_may_** be bootable on the T31, though I haven't tried.  I do plan to test it at some point.  

If it **_is_** possible to boot the newer kernel, the next step would be to verify that everything works, like SDIO/MMC, GPIO, i2c, etc.  If the kernel boots but everything else is broken, it doesn't really get us anywhere.

gtxaspec, 2022-06-21:
> @archandanime I'd think the bigger problem is how the T31 works. The T31 is a combination of MIPS and RISC-V (http://www.ingenic.com.cn/en/?product/id/20.html). I don't know enough to answer [...]

the only camera which seems to use the risc-v core is the outdoor camera.  I don't have the SDK for the risc-v stuff yet, so progress is slow on this.

Every other camera just uses the regular SDK which is already out there.  If you have the knowledge, you can really do anything with the camera's.  The SDK includes everything you need to.

Ideally the best thing would be to take the latest kernel, 5.19, and work on the source code to get it booting on the T31.  Getting the right people together that have the knowledge and time to do this is the trick though.

mach7, 2023-02-13:
> the only camera which seems to use the risc-v core is the outdoor camera. I don't have the SDK for the risc-v stuff yet, so progress is slow on this.
>
> Every other camera just uses the regular SDK [...]

Where would someone might start to obtain the knowledge to help on something like this?

archandanime, 2023-10-03:
#631