gtxaspec/wz_mini_hacks #116: upgrade-run.sh failure

shamlord, 2022-06-21:
One of my cameras had a problem resolving/reaching github during a wz_mini upgrade.  I deleted the `Upgrade` directory and rerunning the upgrade script a couple times, without success. I tried rebooting the camera, but upon reboot the camera tried upgrading from a nonexistant download of `wz_mini.zip`, rm -rf'ing the old directories from `/opt/wz_mini/`.  I don't recall if I deleted the `Upgrade` directory the last time before the reboot.  I suggest adding a check in the script to ensure the upgrade files are present before deleting the existing files.  I pulled the SD card and found that it only contained the following empty folders and log file:
```
.\.ota\
.\time_lapse\
.\wz_mini\
.\upgrade_wz_mini.log
```

Log attached! [upgrade_wz_mini.log](https://github.com/gtxaspec/wz_mini_hacks/files/8947000/upgrade_wz_mini.log)

archandanime, 2022-06-21:
The upgrade-run.sh has does check the upgrade archive before upgrading!
(upgrade-run.sh):
```
...
echo "Download latest master"
wget --no-check-certificate https://github.com/gtxaspec/wz_mini_hacks/archive/refs/heads/master.zip -O /opt/Upgrade/wz_mini.zip; sync
    
echo "Extract archive"
unzip /opt/Upgrade/wz_mini.zip -d /opt/Upgrade/

echo "Verify file integrity"
cd /opt/Upgrade/wz_mini_hacks-master
md5sum -c file.chk

if [ $? -eq 0 ]; then
        echo "files OK"
        install_upgrade_script
else
        echo "Failure: archive has corrupted files"
        echo "Delete failed upgrade dir"
        rm -rf /opt/Upgrade
        exit 1
fi
        
}
...
  ```
That's why there are repeated errors at the beginning of the log file
But somehow the upgrade-run.sh script skipped some lines of code, really weird.
(upgrade_wz_mini.log):
```
Download latest master
--2022-06-21 07:45:51--  https://github.com/gtxaspec/wz_mini_hacks/archive/refs/heads/master.zip
Resolving github.com... failed: Try again.
wget: unable to resolve host address 'github.com'
Extract archive
Archive:  /opt/Upgrade/wz_mini.zip
unzip: short read
Verify file integrity
/opt/wz_mini/usr/bin/upgrade-run.sh: cd: line 28: can't cd to /opt/Upgrade/wz_mini_hacks-master: No such file or directory
md5sum: file.chk: No such file or directory
Failure: archive has corrupted files
+ GPIO=63
+ '[['  '==' false ]]
+ echo 'not HL_PAN2'
not HL_PAN2
+ '[' -b /dev/mtdblock9 ]
+ '[['  '==' true ]]
+ insmod /opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/kernel/audio.ko 'spk_gpio=63' 'alc_mode=0' 'mic_gain=0'
+ /opt/wz_mini/bin/audioplay_t31 /opt/wz_mini/usr/share/audio/upgrade_mode.wav 50
[INFO]        : Playing the audio /opt/wz_mini/usr/share/audio/upgrade_mode.wav file.
[INFO]        : Volume is set to 50.
```

shamlord, 2022-06-21:
Right.

I did see the upgrade download hash verification in the logs of my other cameras which had a successful upgrade. On this camera though, the upgrade script would fail and then exit when I attempted to run it. The errors were that either Github was unreachable, or the zip file was incomplete.

Trying to troubleshoot the connectivity issue, I changed a setting on my camera network and then rebooted the camera. I was surprised when the camera played the upgrade announcement. 

It makes sense that there is a second part of the upgrade script that runs after the reboot. The output from which appears to start at line 73 with the output: `+ GPIO=63`. 

Either the trigger for that post-reboot upgrade script should not be added to the wz_mini startup routine until after the download is verified, or the post-reboot script should reverify that the files are at least present. Or maybe some other solution.

gtxaspec, 2022-06-21:
latest release includes some fixes to upgrade-run.sh

You will have to manually downloaded the latest version and replace the one located in `wz_mini/bin/upgrade-run.sh`

shamlord, 2022-06-22:
> there are issues upgrading from early versions of the script to the recent ones, which shouldnt happen in the recent versions... I will use this feedback and run some additional checks and verifications. stay tuned.

I don't believe my issue was caused by the difference between the older and newer hack revisions. I'm not certain, but I believe I was running the May 25th or May 27th version of wz_mini on the cameras before the update. All are on the same Wyze firmware version. Before running the update script, I did manually replace the `wz_mini.conf` files on each camera with the most recent version on Github as of yesterday. I then ran the update script, and all four of my cameras were experiencing difficulty reaching Github directly.

I suspect the connectivity issue was caused by my wifi AP. I had the update scripts running on all four cameras and was troubleshooting the connectivity issue. I decided to restart my AP, and unexpectedly, three out of the four cameras started downloading and were able to complete the download during the AP shutdown. Those cameras then successfully completed their updates without issue. It seems the fourth camera was also able to initiate a download, but just didn't complete it in time before the AP rebooted, ultimately causing the update issue. 

Regardless of the reason for the interrupted download and the resulting update failure, it's no big deal to just manually update the SD card. I just wanted to share the errors in the update script log to help improve the project.

gtxaspec, 2022-06-23:
fixed in https://github.com/gtxaspec/wz_mini_hacks/commit/cf02d1aeb96fafde1b19d5eb25a22b705e536df5

thanks for the logs!

closed 2022-06-23 by gtxaspec