gtxaspec/wz_mini_hacks #124: Kernel compliation for overlayfs testing

archandanime, 2022-06-24:
I'm ready to apply the [overlayfs patch](https://github.com/adilinden-oss/overlayfs-patches/blob/master/overlayfs.v18-3.10-rc7.patch), I also complied T31 kernel from [mnakada's repo](https://github.com/mnakada/atomcam_tools) after replacing mnakada's built-in init by your v3_init. Was that all I need to do to make the v3 cam boot successfully?
I ask in case there are some missing steps.
Thank you.

gtxaspec, 2022-06-24:
yes, use the t31 kernel config from my repo, modify the init script in `src/initramfs_skeleton/init` and then `make linux-rebuild`

I have a sample init that I was testing if you want it too.

archandanime, 2022-06-24:
yes, I also replaced the kernel.config, I forgot to tell, sorry about that.
I would love to see your init file.

gtxaspec, 2022-06-24:
```
#!/bin/sh
# devtmpfs does not get automounted for initramfs
##V2

set -x

mount -t devtmpfs devtmpfs /dev
mount -t proc proc /proc
mount -t sysfs sysfs /sys

sleep 1

echo 43 > /sys/class/gpio/export
echo in > /sys/class/gpio/gpio43/direction

#wait for sd card init
sleep 3

mkdir -p /sdcard
mount -t vfat /dev/mmcblk0p1 /sdcard -o rw,umask=0000,dmask=0000

mkdir /v3
mount -t squashfs /dev/mtdblock2 /v3

if [ ! -f /sdcard/wz_mini/etc/init.d/v3_init.sh ]; then

echo "v3_init not found, booting stock"

mount --move /dev /v3/dev
mount --move /sys /v3/sys
mount --move /proc /v3/proc

umount /sdcard

exec busybox switch_root /v3 /linuxrc

else

mkdir -p /v3/dev
mkdir -p /v3/sys

mount --move /dev /v3/dev
mount --move /sys /v3/sys
mount --move /proc /v3/proc

#mkdir -p /v3/media/mmc
mount --move /sdcard /v3/opt

#exec busybox switch_root /v3 /linuxrc
exec busybox switch_root /v3 /opt/wz_mini/etc/init.d/v3_init.sh

fi
```
its old, so rename the scripts accordingly.

gtxaspec, 2022-06-24:
no, the kernel command line is not the same as the stock.  It specifies the initramfs.

archandanime, 2022-06-24:
ummm do you have a file where you wrote down the commandline?

gtxaspec, 2022-06-24:
its in the .config in the wz_mini repo under src

archandanime, 2022-06-24:
I get it now, it's already there :), replace the kernel.config and it's done

archandanime, 2022-06-24:
@gtxaspec the factory_t31_ZMC6tiIDQN kernel is loaded by uboot instead of being written to the device so there's no risk bricking it, right? I just want to make sure, again

gtxaspec, 2022-06-24:
right, the camera is impossible to brick, unless you mess with u-boot, `/dev/mtd1`

if any other partition gets damaged, just flash demo-wcv3.bin and everything is back to stock.

archandanime, 2022-06-24:
Procedure:
1. Replace initramfs-skeleton/init by init_overlay (the one you sent above)
2. Replace kernel.config
3. run "sudo make" at atomcam_tools
```
(atomcam_tools) $ ls target
factory_t31_ZMC6tiIDQN  rootfs_hack.ext2
```
4. scp the factory_t31_ZMC6tiIDQN to /media/mmc/
Result: The device fails to boot, I ssh'ed to router(it runs Openwrt), the camera is not connected to the router.
I'm not at home to check if the status light is on or what's inside the SD card

gtxaspec, 2022-06-24:
did you change the script names in the init? they changed, its not v3_init anymore, wz_init.sh

archandanime, 2022-06-24:
the built-in init? - Yes
oh the line:
```
	exec busybox switch_root /overlay /opt/wz_mini/etc/init.d/v3_init.sh
```
I didn't change it to wz_init.sh

archandanime, 2022-06-24:
@gtxaspec the camera still doesn't boot.
I changed
```
	exec busybox switch_root /overlay /opt/wz_mini/etc/init.d/v3_init.sh
``` to
```	exec busybox switch_root /overlay /opt/wz_mini/etc/init.d/wz_init.sh
```
then complied the kernel but it didn't boot.
I also tried using the old kernel-built-in init but it still doesn't boot

gtxaspec, 2022-06-24:
how are you recompiling? did you `docker exec` into the docker instance?

archandanime, 2022-06-24:
I ran sudo `docker-compose exec builder bash` then `make linux-rebuild`

gtxaspec, 2022-06-24:
ok that works too.  I use `docker exec -it <id> /bin/bash`  Did put the wz_mini kernel config in `/src/kernel.config` ?

gtxaspec, 2022-06-24:
this is the latest stock init:

```
#!/bin/sh

set -x

# devtmpfs does not get automounted for initramfs
mount -t devtmpfs devtmpfs /dev
mount -t proc proc /proc
mount -t sysfs sysfs /sys

#Required delay on T20/T31
sleep 1

echo "Check for T20 platform"

if [ -b /dev/mtdblock10 ]; then

	echo "Found T20"

	#T20: Need to export the sd card enable gpio manually
	echo 43 > /sys/class/gpio/export
	echo in > /sys/class/gpio/gpio43/direction

	#T20: Insert required delay for sd card init
	sleep 3

else
	echo "Not T20"
fi

mkdir -p /sdcard
mount -t vfat /dev/mmcblk0p1 /sdcard -o rw,umask=0000,dmask=0000

initram_init() {

mkdir /wz
mount -t squashfs /dev/mtdblock2 /wz

if [ ! -f /sdcard/wz_mini/etc/init.d/wz_init.sh ]; then

	echo "wz_mini not found, booting stock"

	mount --move /dev /wz/dev
	mount --move /sys /wz/sys
	mount --move /proc /wz/proc

	umount /sdcard

	exec busybox switch_root /wz /linuxrc

else

	echo "Loading wz_mini..."

	mkdir -p /wz/dev
	mkdir -p /wz/sys

	mount --move /dev /wz/dev
	mount --move /sys /wz/sys
	mount --move /proc /wz/proc

	#mkdir -p /wz/media/mmc
	mount --move /sdcard /wz/opt

	#exec busybox switch_root /wz /linuxrc
	exec busybox switch_root /wz /opt/wz_mini/etc/init.d/wz_init.sh

fi

}

if [ -f /sdcard/wz_mini/wz_mini.conf ]; then

	source /sdcard/wz_mini/wz_mini.conf

	if [[ "$DEBUG_INITRAMFS_ENABLED" == "true" ]]; then
		/bin/sh
		#exit 0
	else
		echo "initramfs debug disabled"
		initram_init
	fi
else
		initram_init
fi
```
try using that one just to make sure you can compile a working kernel

archandanime, 2022-06-24:
I tried the code above to initramfs-skeleton/init, it still doesn't boot.
there's only the LED red above the cam

archandanime, 2022-06-24:
I think I can make serial connection to the camera, I'll solder and give you the bootlog

archandanime, 2022-06-24:
I got the log!
these are last lines of the log
```
[    0.748035] ALSA device list:
[    0.751094]   #0: Loopback 1
^@[    0.755996] Freeing unused kernel memory: 484K (80507000 - 80580000)
[    0.762592] Failed to execute /init

Please press Enter to activate this console. [    0.963813] mmc0: new high speed SDHC card at address 5048
[    0.969696] mmcblk0: mmc0:5048 SD16G 14.4 GiB 
[    0.975660]  mmcblk0: p1
```
full log is [here](https://github.com/archandanime/Log-files/blob/main/2022-06-24_wz_mini_boot_01.log)

archandanime, 2022-06-24:
I tried compiling the kernel so many times without success.
always stuck at
```
[    0.748100] ALSA device list:
[    0.751160]   #0: Loopback 1
^@[    0.756052] Freeing unused kernel memory: 484K (80507000 - 80580000)
[    0.762647] Failed to execute /init

Please press Enter to activate this console. [    0.963819] mmc0: new high speed SDHC card at address 5048
```
how exactly did you compile your kernel, please? I think I'm not doing it the right way, what I do:
```
git clone https://github.com/mnakada/atomcam_tools.git
cd atomcam_tools
sudo systemctl start docker
sudo docker-compose exec builder bash
make linux-rebuild
cp output/images/uImage.lzma /src
```
then I copy the uImage.lzma to SD card and rename to factory_t31_ZMC6tiIDQN
I don't know if I'm doing wrong :(
maybe you want to try compiling it and send me so we can test it out

archandanime, 2022-06-24:
it's at `CONFIG_INITRAMFS_SOURCE="/openmiko/build/buildroot-2016.02/output/images/initramfs.cpio"`

archandanime, 2022-06-24:
it's 644, let's see if it work with 755

archandanime, 2022-06-24:
it works!!!!!!!!

gtxaspec, 2022-06-24:
=D

archandanime, 2022-06-24:
wait, I think the init works but I have kernel panic here

gtxaspec, 2022-06-24:
in the kernel, disable netfilter and try again

archandanime, 2022-06-24:
changed to `CONFIG_NETFILTER=n`, compiling again

gtxaspec, 2022-06-24:
you can also use `make linux-menuconfig` to bring up the standard Linux menu config

gtxaspec, 2022-06-24:
there may be a typo in the .config file? or formatting error?

gtxaspec, 2022-06-24:
does /openmiko/build/buildroot-2016.02/output/images/initramfs.cpio exist?

archandanime, 2022-06-24:
wait, is `CONFIG_NETFILTER=n` correct? or I should leave `CONFIG_NETFILTER=`?
 /openmiko/build/buildroot-2016.02/output/images/initramfs.cpio doesn't exist

gtxaspec, 2022-06-24:
just comment the line out,don't set to n

archandanime, 2022-06-24:
```
mkdir -p /openmiko/build/buildroot-2016.02/output/images/
touch /openmiko/build/buildroot-2016.02/output/images/initramfs.cpio
```
it's compiling

archandanime, 2022-06-24:
```
[    0.307334] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.313868] dwc2 dwc2: ID PIN CHANGED!
[    0.328174] Kernel panic - not syncing: junk in compressed archive
[    0.334374] Rebooting in 3 seconds..Restarting after 4 ms
```
that trick didn't help

gtxaspec, 2022-06-24:
if you leave netfilter, do you get that error still?

archandanime, 2022-06-24:
yes, my the issue on my computer

gtxaspec, 2022-06-24:
hmm... ok try disabling the new wifi drivers in wz_mini.conf

gtxaspec, 2022-06-24:
i updated https://github.com/gtxaspec/wz_mini_hacks/blob/master/src/kernel/t31-config.conf 

if you use that config, add the overlay, and it should work with the new wifi drivers (i think)

archandanime, 2022-06-24:
it's my computer acting weird. I'll try it as soon as the compiler works again

archandanime, 2022-06-24:
and thank you!

gtxaspec, 2022-06-24:
no worries! knowledge for all! =D

archandanime, 2022-06-24:
@gtxaspec 
with t3-config.conf as init
```
[    1.773712] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
mount: mounting /sdcard/rootfs_hack.ext2 on /newroot failed: No such file or directory
mount: mounting /proc on /newroot/proc failed: No such file or directory
BusyBox v1.24.1 (2020-08-26 19:40:44 UTC) multi-call binary.

Usage: switch_root [-c /dev/console] NEW_ROOT NEW_INIT [ARGS]

Free initramfs and switch to another root fs:
chroot to NEW_ROOT, delete all in /, move NEW_ROOT to /,
execute NEW_INIT. PID must be 1. NEW_ROOT must be a mountpoint.

	-c DEV	Reopen stdio to DEV after switch
[    1.843866] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[    1.843866] 
[    1.853306] Rebooting in 3 seconds..Restarting after 4 ms
```

gtxaspec, 2022-06-24:
the initramfs file is the atomcam_tools version, you may have to replace it again

gtxaspec, 2022-06-25:
you also may want to delete the docker image completely and start over

archandanime, 2022-06-25:
I think it should work now.
the `src/buildscripts/make_initramfs.sh` generates `initramfs.cpio` at `/atomtools/build/buildroot-2016.02/output/images`
but the `make` command looks for it at `/openmiko/build/buildroot-2016.02/output/images/` so I made the `/openmiko/build/buildroot-2016.02/output/images/` and copied the `initramfs.cpio` to there. Then the compilation succeeded. Its size is 2168078 bytes which looks very 'legit', the kernels I complied earlier were only around 1880473.
I think that initramfs was not properly embedded in to my previous kernels

archandanime, 2022-06-25:
it worked
the initramfs init was executed
```
[    0.794430]   #0: Loopback 1
^@[    0.799350] Freeing unused kernel memory: 496K (80544000 - 805c0000)
+ mount -t devtmpfs devtmpfs /dev
+ mount -t proc proc /proc
+ mount -t sysfs sysfs /sys
+ sleep 1
[    1.014907] mmc0: new high speed SDHC card at address 5048
[    1.020780] mmcblk0: mmc0:5048 SD16G 14.4 GiB 
[    1.026743]  mmcblk0: p1
+ echo Check for T20 platform
Check for T20 platform
+ [ -b /dev/mtdblock10 ]
+ echo Not T20
Not T20
+ mkdir -p /sdcard
+ mount -t vfat /dev/mm[    1.818261] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
cblk0p1 /sdcard -o rw,umask=0000,dmask=0000
+ [ -f /sdcard/wz_mini/wz_mini.conf ]
+ source /sdcard/wz_mini/wz_mini.conf
```
but it ended up with kernel panic, I think it failed to switch to new root
[log file](https://github.com/archandanime/Log-files/blob/main/2022-06-25_wyze-v3_boot01.log)

archandanime, 2022-06-25:
kernel with Overlayfs init
[log file](https://github.com/archandanime/Log-files/blob/main/2022-06-25_wyze-v3_boot02.log)
```
/init: line 55: syntax error: unexpected end of file (expecting "fi")
```
I thought you commented it on purpose

archandanime, 2022-06-25:
kernel with Overlayfs init, this time #fi is uncommented
[log file](https://github.com/archandanime/Log-files/blob/main/2022-06-25_wyze-v3_boot03.log)
```
+ mkdir /upper
+ mkdir /overlay
+ mkdir /rootfs
+ mkdir /sdcard
+ sleep 1
[    1.034909] mmc0: new high speed SDHC card at address 5048
[    1.040792] mmcblk0: mmc0:5048 SD16G 14.4 GiB 
[    1.046749]  mmcblk0: p1
+ mount -t tmpfs -o size=1 tmpfs /upper
+ mount -t tmpfs -o size=1 tmpfs /overlay
+ mount -t squashfs /dev/mtdblock2 /rootfs
+ mount -t vfat /dev/mmcblk0p1 /sdcard -o rw,umask=0000,dmask=0000
[    1.945162] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
+ mount -t overlayfs overlayfs -olowerdir=/rootfs,upperdir=/upper /overlay
mount: mounting overlayfs on /overlay failed: No such device
+ mkdir /overlay/rootfs
+ mkdir /overlay/overlay
+ mkdir /overlay/upper
+ mount --move /sdcard /overlay/opt
mount: mounting /sdcard on /overlay/opt failed: No such file or directory
+ mount --move /rootfs /overlay/rootfs
+ grep INITRAMFS_DBG_ENABLED\=
+ cat /overlay/opt/wz_mini/run_mmc.sh
cat: can't open '/overlay/opt/wz_mini/run_mmc.sh': No such file or directory
+ [[ == INITRAMFS_DBG_ENABLED="true" ]]
sh: INITRAMFS_DBG_ENABLED="true": unknown operand
+ mount --move /dev /overlay/dev
mount: mounting /dev on /overlay/dev failed: No such file or directory
+ mount --move /sys /overlay/sys
mount: mounting /sys on /overlay/sys failed: No such file or directory
+ mount --move /proc /overlay/proc
mount: mounting /proc on /overlay/proc failed: No such file or directory
+ exec busybox switch_root /overlay /opt/wz_mini/etc/init.d/wz_init.sh
switch_root: can't execute '/opt/wz_mini/etc/init.d/wz_init.sh': No such file or directory

[    2.064974] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000100
[    2.064974] 
[    2.074589] Rebooting in 3 seconds..
```

archandanime, 2022-06-25:
I think the Overlayfs is not in the kernel, though `CONFIG_OVERLAYFS_FS=y`

gtxaspec, 2022-06-25:
did you disable the new wifi drivers?

archandanime, 2022-06-25:
how do I... do it? I looked for `8189` in kernel.config but I couldn't find

gtxaspec, 2022-06-25:
in wz_mini.conf

archandanime, 2022-06-25:
it's still stuck at kernel panic after I changed to `ENABLE_RTL8189FS_DRIVER="false"`

gtxaspec, 2022-06-25:
you disabled netfilter? that's the cause of the crash

archandanime, 2022-06-25:
yes, I left it empty `CONFIG_NETFILTER=`

gtxaspec, 2022-06-25:
comment it out ##

archandanime, 2022-06-25:
I commented it, still kernel panic

gtxaspec, 2022-06-25:
in wz mini conf, enable debug mode

archandanime, 2022-06-26:
it drops me a shell but I can't enter commands
```
+ mount --move /sys /wz/sys
+ mount --move /proc /wz/proc
+ mount --move /sdcard /wz/opt
+ exec busybox switch_root /wz /opt/wz_mini/etc/init.d/wz_init.sh

WCV3 login: 
```

[59 more replies]