gtxaspec/wz_mini_hacks #127: Add wiki page

archandanime, 2022-06-24:
I think it's a good idea make to wiki pages so it's easier to read and the README.md won't be too long.
It's easier to add more contents too.

archandanime, 2022-06-28:
@gtxaspec I can't add wiki pages or make pull request for wiki pages from my cloned repo

gtxaspec, 2022-06-28:
from `https://github.com/gtxaspec/wz_mini_hacks.wiki.git` ?

archandanime, 2022-06-29:
let me try it again

tensiondriven, 2022-09-10:
As a counterpoint, having the documentation in source control makes a lot of sense to me.

I agree that the readme could be improved and split out into multiple files, but I do not think divorcing the docs from git would be an improvement.

The good news is that any one of us could update the docs and submit a pull request to improve this with minimal technical expertise.

gtxaspec, 2023-09-26:
wiki added.

closed 2023-09-26 by gtxaspec