gtxaspec/wz_mini_hacks #128: Issues after running update-run.sh

tavicu, 2022-06-24:
Hi,

It was the new release and I tried updating the cameras with `update-run.sh`.
I don't know why but after `Rebooting into UPGRADE MODE` the cameras are indeed rebooting but I don't hear any sound of `Upgrade mode` that I hear before.

Once the camera boots it works in the wyze app but wz_mini looks like it was not loaded because I can't connect to ssh and also the RTSP feeds are down.
Then I tried to unplug the camera from the power and plug it again but the same thing.

The only option is to remove the SD card and write the wz_mini again.

This is a very big problem because I'm planning to install the cameras on my house and they will not be easy reachable. Removing the SD card will be a pain in the ass and I will probably avoid updating the cameras.

How can I help you to debug this?
It happened on all my 5 cameras.

tavicu, 2022-06-24:
Here is the update log:
```
wz_mini, this will download the latest version from github and upgrade your system.  Are you sure? [y/N] Create Upgrade staging directory
Create backup files directory
Download latest master
--2022-06-24 12:27:44--  https://github.com/gtxaspec/wz_mini_hacks/archive/refs/heads/master.zip
Resolving github.com... 140.82.121.4
Connecting to github.com|140.82.121.4|:443... connected.
WARNING: cannot verify github.com's certificate, issued by 'CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1,O=DigiCert Inc,C=US':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/gtxaspec/wz_mini_hacks/zip/refs/heads/master [following]
--2022-06-24 12:27:44--  https://codeload.github.com/gtxaspec/wz_mini_hacks/zip/refs/heads/master
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
WARNING: cannot verify codeload.github.com's certificate, issued by 'CN=DigiCert TLS Hybrid ECC SHA384 2020 CA1,O=DigiCert Inc,C=US':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: '/opt/.Upgrade/wz_mini.zip'

     0K .......... .......... .......... .......... ..........  376K
    50K .......... .......... .......... .......... ..........  489K
   100K .......... .......... .......... .......... ..........  589K
   150K .......... .......... .......... .......... ..........  493K
   200K .......... .......... .......... .......... ..........  525K
   250K .......... .......... .......... .......... ..........  492K
   300K .......... .......... .......... .......... ..........  424K
   350K .......... .......... .......... .......... ..........  628K
   400K .......... .......... .......... .......... ..........  557K
   450K .......... .......... .......... .......... ..........  736K
   500K .......... .......... .......... .......... ..........  580K
   550K .......... .......... .......... .......... ..........  754K
   600K .......... .......... .......... .......... .......... 1.04M
   650K .......... .......... .......... .......... ..........  506K
   700K .......... .......... .......... .......... ..........  568K
   750K .......... .......... .......... .......... ..........  744K
   800K .......... .......... .......... .......... ..........  617K
   850K .......... .......... .......... .......... .......... 1.01M
   900K .......... .......... .......... .......... ..........  650K
   950K .......... .......... .......... .......... ..........  632K
  1000K .......... .......... .......... .......... .......... 1.03M
  1050K .......... .......... .......... .......... ..........  524K
  1100K .......... .......... .......... .......... ..........  795K
  1150K .......... .......... .......... .......... ..........  992K
  1200K .......... .......... .......... .......... ..........  916K
  1250K .......... .......... .......... .......... .......... 1.22M
  1300K .......... .......... .......... .......... ..........  708K
  1350K .......... .......... .......... .......... ..........  599K
  1400K .......... .......... .......... .......... .......... 1.13M
  1450K .......... .......... .......... .......... ..........  696K
  1500K .......... .......... .......... .......... ..........  525K
  1550K .......... .......... .......... .......... ..........  619K
  1600K .......... .......... .......... .......... ..........  792K
  1650K .......... .......... .......... .......... ..........  459K
  1700K .......... .......... .......... .......... ..........  612K
  1750K .......... .......... .......... .......... .......... 2.38M
  1800K .......... .......... .......... .......... ..........  655K
  1850K .......... .......... .......... .......... ..........  656K
  1900K .......... .......... .......... .......... ..........  543K
  1950K .......... .......... .......... .......... ..........  585K
  2000K .......... .......... .......... .......... ..........  822K
  2050K .......... .......... .......... .......... ..........  367K
  2100K .......... .......... .......... .......... ..........  690K
  2150K .......... .......... .......... .......... ..........  588K
  2200K .......... .......... .......... .......... ..........  603K
  2250K .......... .......... .......... .......... ..........  620K
  2300K .......... .......... .......... .......... ..........  628K
  2350K .......... .......... .......... .......... ..........  765K
  2400K .......... .......... .......... .......... ..........  691K
[... 1216 lines cut ...]
+ '[' -f /opt/wz_mini/tmp/.T20 ]
+ insmod /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/audio.ko 'spk_gpio=63' 'alc_mode=0' 'mic_gain=0'
insmod: can't read '/opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/audio.ko': No such file or directory
+ /opt/wz_mini/bin/audioplay_t31 /opt/Upgrade/wz_mini_hacks/SD_ROOT/wz_mini/usr/share/audio/upgrade_mode.wav 50
Segmentation fault
+ rmmod audio
rmmod: remove 'audio': No such file or directory
+ echo 'UPGRADE MODE'
UPGRADE MODE
+ echo 'Verify extracted file integrity'
Verify extracted file integrity
+ cd /opt/Upgrade/wz_mini_hacks
/opt/wz_mini/bin/upgrade-run.sh: cd: line 107: can't cd to /opt/Upgrade/wz_mini_hacks: No such file or directory
+ md5sum -c file.chk
md5sum: file.chk: No such file or directory
+ '[' 1 -eq 0 ]
+ echo 'Failure: Extracted files may be corrupt.  Aborting upgrade.'
Failure: Extracted files may be corrupt.  Aborting upgrade.
+ echo 'Delete failed upgrade directory'
Delete failed upgrade directory
+ rm -rf /opt/Upgrade
+ exit 1
+ reboot

```

tavicu, 2022-06-24:
I can see in the logs that those lines are duplicated: Maybe the script is run multiple times?

```
Exit old script
Resume upgrade-run, latest version
Resume upgrade-run, latest version
Backup user config
Backup user config
Rebooting into UPGRADE MODE
Rebooting into UPGRADE MODE
```

archandanime, 2022-06-24:
Sorry about inconvenience, the file has been modified recently without testing.
> Maybe the script is run multiple times?
it wasn't run multiple times but was because the script ran itself again...
```
echo "Launching latest upgrade-script"
/opt/wz_mini/bin/upgrade-run.sh backup_begin &
```
I think gtxaspec will make releases soon so people won't put experimental code into their devices

gtxaspec, 2022-06-24:
the duplicate messages are a result of the file descriptor nor being closed properly I believe, its a bug.  I did some testing before releasing, In my testing I had no issues but obviously something may have been missed :o: 

I'll double check today

gtxaspec, 2022-06-24:
Looks like this is an issue with the new upgrade script, upgrading from the old one.   it doesn't affect new installs, just upgrades that contain the old script.  the fix is to install the upgrade script before upgrading.  I'll have to figure out how to avoid this situation in the future.

gtxaspec, 2022-06-24:
fixed in latest release.

tavicu, 2022-06-26:
Awesome. Still have 2 cameras which I didn't update. I will try it today :)

tavicu, 2022-06-27:
I tried something.

After everything was working I run the `sync` and `reboot` commands manually from the SSH.
The camera restarts but wz_mini it's not loaded.

Power cycle it makes it loading wz_mini.

Then I tried to run only `reboot` command and same thing. It reboots the camera and doesn't load wz_mini.
So it appears there is a problem with `reboot` command on 3 of my cameras.

gtxaspec, 2022-06-27:
@tavicu which model camera?

tavicu, 2022-06-27:
> @tavicu which model camera?

Wyze v3, all of them. 3 were purchased 8 months ago (all 3 have this problem) and two were purchased about one month ago.

tavicu, 2022-06-28:
I have formated the card and setup wz_mini from the scratch.
Booted the camera, everything was ok.

Then I run the update-run.sh script and it started the update process.
Then the camera reboots and nothing. No message, no access to ssh. The wz_mini was not loaded.
Here is the update log: https://pastebin.com/uCgACMaC

Unplug the camera from the power and back in I can hear the message: "wz_mini initializing upgrade mode"
Camera reboots, wz_mini not loaded.
Logs after this step: https://pastebin.com/WAiSpsTW

Unplug the camera again and hear message: "wz_mini initializing first boot, extracting swap, please wait"
Camera reboots, wz_mini is loaded and everything works.
No logs after this step.

Run the `reboot` command from SSH.
Camera reboots, wz_mini not loaded.
Unplug camera and and back again, wz_mini is loaded.

It appears that the problem is when rebooting. It doesn't load wz_mini.

tavicu, 2022-06-28:
Ok. As I said in previous messages I have 5 cameras in total. 
3 are older, 2 newer.
The problem can be replicated on the 3 older cameras.

The other difference between them is the SD Card. Older one have a card from Hama and the newer ones have a card from Kingston.
Same speed, same capacity (32GB), same way to format them.

I've made a writing/read test and the Hama cards are even better, reaching 50MB/s writing speeds versus 22MB/s of Kingston.

But I tried a Kingston card on one of the 3 cameras with problem. And guess what, everything works. After running `reboot` the camera restarts and wz_mini loads.
Run the `upgrade-run.sh` script and the camera restarts and hear messages, all good without having to unplug the camera from the power.

I then got the old SD Card (hama), format it and copy everything from the newer card to this one. Booted the camera with it, wz_mini was loaded. Run the `reboot` command and wz_mini was not loading.
Even tried multiple times because I could not believe it.

So the problem appears to come from the SD card. But I don't understand why, only the manufacturer differs.

virmaior, 2022-07-06:
can you put one of the non-functioning cards into a computer and tell us whether it's formatted FAT32 or EXFAT? (first link for how to do so on windows: https://www.diskpart.com/articles/check-sd-card-format-0310.html )

tavicu, 2022-07-06:
It was formatted at FAT32.

I will close this issue for now since it's not related to update-run.sh
If you want to debug and have any indications on how to do it, I will be happy to open a new issue and continue there.
Let me know :)

closed 2022-07-06 by tavicu