gtxaspec/wz_mini_hacks #133: add function to libcallback to stop motor controls on panv2

gtxaspec, 2022-06-25:
find the function, hook it to stop motor init.

gtxaspec, 2022-06-25:
`local_sdk_motor_init`

endertable, 2022-06-25:
Hi @gtxaspec Forgive my ignorance, just asking because I am confused, but why exactly would one want to stop motor controls on a PTZ (minus the Z)?  Since you are working on it I know there would be a specific reason, I just can't think of it.

gtxaspec, 2022-06-25:
it was a request 
https://github.com/gtxaspec/wz_mini_hacks/discussions/132

gtxaspec, 2022-07-01:
added in https://github.com/gtxaspec/wz_mini_hacks/commit/d87d43241fbd04e4b7304e2aca94d42dbab088dd

closed 2022-07-01 by gtxaspec

claudobahn, 2022-07-11:
@gtxaspec noticed you removed all the sources for libcallback as part of the commit for this (https://github.com/gtxaspec/wz_mini_hacks/commit/d87d43241fbd04e4b7304e2aca94d42dbab088dd). Any chance you could restore that?

Seems like a useful piece of source code for expanding functionality.