gtxaspec/wz_mini_hacks #136: RTSP video display distorted with fast moving objects in frame

goldenhub77, 2022-06-27:
All V3 cameras with current firmware, running RTSP 20FPS Max bitrate 2048, target 1024 get distortion/artifacts with fast moving objects. Impacts video quality but not audio.

clee604, 2022-06-27:
May be similar to the issue I posted here, no resolution though:
https://github.com/gtxaspec/wz_mini_hacks/discussions/121

goldenhub77, 2022-06-27:
@clee604 Yes, this is exactly what I'm experiencing. It is not a connectivity issue because I have perfect signals and it only occurs when using the RTSP server loaded with wz_mini_hacks.

gtxaspec, 2022-06-27:
if you change the encoder mode, to 4 or 8, does this still happen?

goldenhub77, 2022-06-27:
@gtxaspec Yes, I tried 4 and 8 and it still occurs.

goldenhub77, 2022-06-27:
@gtxaspec How ever Wyze programmed RTSP on their firmware it works perfect but has other bugs with their accessories like the spotlight. Is there anyway you can assess the RTSP firmware if I share it?

gtxaspec, 2022-06-27:
We have their firmware, and we are unable to implement it the way they are doing it without the original source code.

One other suggestion, if you disable "motion tagging" in the app does it still occur as well?

clee604, 2022-06-27:
I disabled motion tagging as one of the first things after implementing rtsp because I hated the green boxes and it still occurs.

gtxaspec, 2022-06-27:
It's strange, because I can not reproduce this on my end.  The way I test (rtsp stream)

WiFi or ethernet
place the camera infront of a big tv and then run some fast motion > no artifacts
place the camera in front of my and then I move around really fast > no artifacts
place camera with phone upclose for testing > no artifacts

gtxaspec, 2022-06-27:
I was able to get some artifacts just now only by changing the router to a wifi channel that is congested.  The artifacts still do not happen on wired ethernet.  I will continue to investigate this as a software issue on the camera

gtxaspec, 2022-06-27:
another suggestion (for testing) disable continuous or motion recording to micro sd in the app, to test.  I've found when the microsd card is slow or failing, iCamera does not like that and problems occur.

clee604, 2022-06-27:
similar results here: disabling sd card recording does not make a difference.
@gtxaspec  how are you viewing the RTSP stream? I noticed if I have HW acceleration on (using tinycam) there's artifacts. However using SW decoding or VLC there's no artifacts, but instead it frame skips.

clee604, 2022-06-27:
using VLC on my pc i don't really notice any artifacts or frameskips.. 

VLC on my ipad shows some frameskips...

Maybe some kind of error correction is better on the PC VLC?

gtxaspec, 2022-06-28:
Ok, following up, does this happen on the 360p stream as well?

goldenhub77, 2022-06-28:
@gtxaspec Yes

goldenhub77, 2022-06-28:
@skilau I am on Wifi at 20fps. I do not get any tearing or artifacts while using the RTSP firmware directly from Wyze but the audio is distorted.   This implementation has perfect audio but the video is distorted.

clee604, 2022-06-28:
I tested on both ethernet and wifi, and FPS at 15, 20, and blank; all the same issue for me

Vanterax, 2022-06-28:
I have the same problem with three v3 cams, but they're all on Wifi.  The RTSP stream was prefect before updating to the "06-24-22" update, but now I get lots of artifacts and corruption in the video feed.  Of my three v3 cams, I put two back on the Wyze RTSP firmware and now they're streaming fine.  I'm leaving one of the latest wz_mini to troubleshoot.  I have them set to 15fps and it's hard to keep a stable feed.

gtxaspec, 2022-06-28:
if you revert to this release: https://github.com/gtxaspec/wz_mini_hacks/tree/cf02d1aeb96fafde1b19d5eb25a22b705e536df5

do the rtsp problems go away?

gtxaspec, 2022-06-28:
[8189fs.ko.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9003450/8189fs.ko.gz)
Here is an updated build of the wifi driver for the V3 (only v3), if anyone wants to test, rename the old one `8189fs.ko.old` and replace it with this one.  It's located in `/opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/8189fs.ko`

Vanterax, 2022-06-28:
Reverted to the previous release fixes my issue with the RTSP feed.  It's solid and consistent without artifacts or skipping.
I have a 4th spare V3 that I'll be using to test the new wifi driver, but I have to wait for a new SD card coming from Amazon tomorrow. :)

gtxaspec, 2022-06-28:
[8189fs_v2.ko.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9003521/8189fs_v2.ko.gz)
@clee604 here is a new driver for the v2 if you want to test.  the path to replace the driver is `/opt/wz_mini/lib/modules/3.10.14/extra/8189fs.ko`

clee604, 2022-06-28:
thanks @gtxaspec, currently on my V2 I'm rolling back and testing older releases to see if I can get ethernet to work again first. Then I'll test the rtsp stream stability and the new wifi driver after. 

My V3s are in use so I don't want to touch until I'm confident i have enough time to test and fix it they stop working

gtxaspec, 2022-06-28:
thanks for the feedback and testing everyone!  I have almost every camera, and with every release I make sure I test every model camera I have both wifi and Ethernet.  So hopefully we can track this down and figure out what's going on

clee604, 2022-06-28:
I didn't think the wifi driver would change anything on my asix ethernet connected V3 but I tried anyways. no improvement with the artifacts. I will try rolling back the V3 to the above posted version later to see if it improves artifacts.

gtxaspec, 2022-06-28:
@clee604 if i can compile a new kernel with some changes would you be able to test it on your v2?

clee604, 2022-06-28:
@gtxaspec yes, no problem

gtxaspec, 2022-06-28:
@clee604 

copy these to the camera
via ssh, run the commands:

first, replace the wifi driver with this one:
[8189fs.ko.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9005646/8189fs.ko.gz)
extract it and replace it `/opt/wz_mini/lib/modules/3.10.14/extra/8189fs.ko`

then, 
copy this to the camera:
[kernel.bin.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9005589/kernel.bin.gz)
and run the commands to flash the kernel:
```
gzip -d kernel.bin.gz
flashcp -v kernel.bin /dev/mtd1
sync
reboot
```

clee604, 2022-06-28:
@gtxaspec done, what should I be testing post flash?

gtxaspec, 2022-06-28:
problems with RTSP

Vanterax, 2022-06-28:
It just dawned on me that I might have missed an important detail.  I upgraded using the upgrade-run.sh script and then uploaded a new  wz_mini.conf file with all the proper settings.  Turns out that might have been my problem.  By installing wz_mini fresh, I don't have an artifact problem anymore.

clee604, 2022-06-28:
@gtxaspec ah, I was just about to comment that the ethernet doesn't work with this kernel either on my V2.

My V2 didn't have artifacts with RTSP but the video stream did crash lately with the new versions. But that generally happens overnight. So far with the new wifi driver and kernel nothing seems out of the ordinary, I'll leave it on and see how stable it is.

gtxaspec, 2022-06-28:
Ethernet broken...hmm
okay let me look into some stuff for ya

clee604, 2022-06-28:
Think I spoke too soon.. seen a few artifacts with the new V2 kernel which I never had before with the V2 cam..

If i want to roll back can I just reflash demo.bin?

clee604, 2022-06-28:
Back to the ethernet connected V3 - another thing I just noticed when I pulled some recordings off the SD card via SCP, the audio does not sync with the video. I'm not sure if this started happening today when I dropped to 15fps for testing or it was always like this...

gtxaspec, 2022-06-28:
> Think I spoke too soon.. seen a few artifacts with the new V2 kernel which I never noticed before with the V2 cam..
> 
> If i want to roll back can I just reflash demo.bin?

that's right

clee604, 2022-06-28:
> 

Thanks, I'll probably roll back to the last version where ethernet works for my V2 for the time being except when testing.

goldenhub77, 2022-06-30:
@gtxaspec @skilau I reverted to the previous build https://github.com/gtxaspec/wz_mini_hacks/tree/cf02d1aeb96fafde1b19d5eb25a22b705e536df5 and the artifact issue appears to be resolved. I updated the WiFi driver out of curiosity and it caused frame drops so I reverted back to the old driver.

gtxaspec, 2022-06-30:
I've compiled the latest version of wz_mini in a branch without nf-conntrack (v3/panv2 ONLY)
https://github.com/gtxaspec/wz_mini_hacks/tree/wz_mini-no_conntrack
if anyone wants to test it and see if the issues are resolved, go ahead =D

clee604, 2022-06-30:
is there a parameter with the upgrade script to select a different branch now? I saw you mentioned it was done in the other thread but wasn't sure if it was available yet

gtxaspec, 2022-06-30:
edit the file `upgrade-run.sh` and change:

`BRANCH="master"`
to
`BRANCH="wz_mini-no_conntrack`

then run it

clee604, 2022-06-30:
tried twice, but getting 60 out of 232 computed checksums did not match error when branch is changed BRANCH="wz_mini-no_conntrack

gtxaspec, 2022-07-01:
oops one sec. ok its fixed try now

clee604, 2022-07-01:
got this error at the end.. not sure if it actually continued to upgrade or not...
/opt/wz_mini/bin/upgrade-run.sh: line 11: can't create /dev/fd/64: nonexistent directory

gtxaspec, 2022-07-01:
thats just a logging error, shouldn't matter.   it should have rebooted the camera.

if you ssh into the system and run md5sum /opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/8189fs.ko

it should give you `a19d7ac413fcb57fe25f4bc5889739dc`

clee604, 2022-07-01:
it gives me d52dd5f273ed68807dfddfd99940e6f9 so i assume it didn't upgrade

gtxaspec, 2022-07-01:
yeah strange, i just tested on one of my camera's and it worked without error.  you may have to do a manual upgrade.

clee604, 2022-07-01:
i'll take a look at it later when i'm home.. one thing i noticed is when i check the upgrade-run.sh script it is back to "master" for the branch

goldenhub77, 2022-07-02:
@gtxaspec Your changes on the no_conntrack branch resolved the artifact issue but are still dropping frames occasionally. When testing on VLC for about 5-10 minutes the video completely freezes but audio continues. This does not occur using the Wyze RTSP firmware under the same conditions so I know it is not a network related issue.

clee604, 2022-07-02:
I tried the no_conntrack branch on my ASIX connected V3 and didn't boot up correctly (no ssh or rtsp stream, though I think the camera is on). wz_user log had some missing directory for the wifi driver error but I forgot to copy log before I just wiped it with a fresh install of the latest version.

I have not yet a chance to test it on a wifi connected V3 as they are a bit more difficult to get to pull the SD for a fresh install of the no_conntrack branch after the OTA update failed

goldenhub77, 2022-07-02:
@clee604 I recommend wiping your Wyze and updating by downloading the firmware from Wyze and using an SD card.

clee604, 2022-07-02:
I don't have the same video freezing issue as you and the artifacts are a nuisance but not a dealbreaker at the moment.

The wyze RTSP firmware works well but doesn't have ethernet support and I'm a bit concerned about the security of the older RTSP firmware

gtxaspec, 2022-07-02:
> @gtxaspec Your changes on the no_conntrack branch resolved the artifact issue but are still dropping frames occasionally. When testing on VLC for about 5-10 minutes the video completely freezes but [...]

thanks for the info.  still gathering some data, we may have a resolution soon.

goldenhub77, 2022-07-06:
@gtxaspec any updates?

gtxaspec, 2022-07-06:
still working, I'll have a test branch up soon to see.  stand by

mcondren, 2022-07-19:
Just checking in on this. I am also seeing this issue with motion. Sometimes it creates artifacts, sometimes it drops most of the frames that include the motion and appears to freeze for a few seconds. 20fps.

gtxaspec, 2022-07-21:
still experiencing this with the latest release?

mcondren, 2022-07-21:
I turned off one of the two streams, reduced my bit rate by 3/4’s, and reduced my frame rate as well and it mostly went away. Perhaps the camera just can’t handle what I was doing, or my Wi-Fi network couldn’t.

gtxaspec, 2022-08-20:
@goldenhub77 

I run my devices at 30fps, with the following settings:
```
RTSP_HI_RES_MAX_BITRATE="4096"
RTSP_HI_RES_TARGET_BITRATE="4096"
RTSP_HI_RES_ENC_PARAMETER="2"
```
the camera seems to work fine.  I don't use rtsp, but it shouldn't be an issue as long as you have good wifi.