gtxaspec/wz_mini_hacks #145: The wz_user.sh script has r8152 USB Ethernet driver auto detection wrong

jdesai61, 2022-06-30:
The auto recognition doesn't work for r8152 adapter because the value  '0bda:8152' is also listed under cdc_ether (which takes precedence)
```
                        '1004:61aa' | '046d:c11f' | '1410:b001' | '1410:9010' | '413c:8195' |\
                        '413c:8196' | '413c:819b' | '16d5:650a' | '12d1:14ac' | '0bda:8152' )
                                echo "Loading USB CDC Ethernet driver..."
                                modprobe cdc_ether
```
This is a minor issue because I can work around it by setting auto detect off and manually specifying r8152

gtxaspec, 2022-06-30:
from what I remember, 8152 based adapters could use either driver.  @jdesai61, will you confirm your adapter does not work with cdc_ether alone?  Can you supply the brand and model as well?

skilau, 2022-06-30:
Interesting.
It looks like the Device ID ('0bda:8152') for your device can be used with either driver:

grep USB_DEVICE * -r |grep 8152
cdc_ether.c:	USB_DEVICE_AND_INTERFACE_INFO(REALTEK_VENDOR_ID, 0x8152, USB_CLASS_COMM,
r8152.c:	{USB_DEVICE(VENDOR_ID_REALTEK, PRODUCT_ID_RTL8152)},

grep PRODUCT_ID_RTL8152 * -r
r8152.c:#define PRODUCT_ID_RTL8152		0x8152

Both drivers appear to support  '0bda:8152'...
Wonder if it matters what driver is used?

If one driver is preferred over another in this case, we could always either remove the id from the one that is least preferred, or move the cdc_ether "case" above the other.

jdesai61, 2022-06-30:
There is a discrepancy in documentation (or script wz_user.sh)...
```
        # Manually load any other Ethernet Drivers if asked for
        if [[ "$ENABLE_USB_ETH_MODULE_MANUAL" != "" ]]; then
                for i in $(echo "$ENABLE_USB_ETH_MODULES" | tr "," "\n")
                do
                        insmod $KMOD_PATH/kernel/drivers/net/usb/$i.ko
                done
        fi
```
In above code the ENABLE_USB_ETH_MODULES should change to ENABLE_USB_ETH_MODULE_MANUAL so I can setup
ENABLE_USB_ETH_MODULE_MANUAL="a,b,c"
or the documentation should state that ENABLE_USB_ETH_MODULE_MANUAL needs to be set to true and then use a second variable ENABLE_USB_ETH_MODULES (not listed in sample file) should contain the list of modules. 

However, even after fixing this, manual loading using "insmod" doesn't work for me, perhaps because there is a difference between modprobe and insmod.  Removing the first '0bda:8152' and using automatic loading works.

gtxaspec, 2022-07-02:
fixed in https://github.com/gtxaspec/wz_mini_hacks/commit/f1c0e3725e1af99166c9eaab31149e68f5292082

closed 2022-07-02 by gtxaspec