gtxaspec/wz_mini_hacks #147: Strange issue with ethernet driver

jdesai61, 2022-06-30:
First of all, thanks for this excellent painless way to hack with Wyze camera *in my case v3). I am able to turn ethernet on (r8152 based adapter) and the camera seems to function well. However, once I enable ethernet, I can only access camera from Wyzeapp,. I can't ping, ssh or use RTSP streaming etc unless I disable ethernet.  However, the Wyze app on my phone is able to see the stream properly even with ethernet enabled. I have tried enabling bonding but the result is the same.

Perhaps there is something strange going with my network because the mini hack obtains the IP using Wifi and then switches to ethernet using same IP and MAC.   Since I can't ssh once ethernet is enabled, it is difficult to debug the issue or provide more details.

gtxaspec, 2022-06-30:
you can pull the sd card and check wz_mini/logs

the ethernet is brought up in wz_user.log

check there to see if anything stands out.   Unless you having bonding enabled, the ethernet adapter should pull a different IP since the MAC is different...

gtxaspec, 2022-06-30:
What I would recommend for debugging, set everything up using wifi, ssh in, then load the module manually, as in:
`modprobe cdc_ether`
`ifconfig eth0 up`
`udhcpc -i eth0`
`ifconig`
and see if you can get an IP on the ethernet adapter.

jdesai61, 2022-06-30:
ok will do

jdesai61, 2022-06-30:
I think not getting ethernet setup was due to following issue:

  https://github.com/gtxaspec/wz_mini_hacks/issues/145#issuecomment-1171721181

Combined with the fact that even when camera gets ethernet IP and gives up WiFi IP, the wyze app continues to show old WiFi ip.   Closing this issue.

closed 2022-06-30 by jdesai61