gtxaspec/wz_mini_hacks #156: wyze car comparability and fixes

gtxaspec, 2022-07-04:
needs kernel with USB to serial module enabled for the car

endertable, 2022-07-04:
So does the Wyze car use standard V2 FW?  I was under the impression that use a special version of the V2 FW.  Could never find a link.

gtxaspec, 2022-07-05:
it does use a a special firmware and iCamera version.  I have reverse engineered the control protocols so it should work with any camera via rtsp, just need to figure out how to implement controls via an app or web page.

gtxaspec, 2022-07-05:
serial commands: 
```
idle=[aa][55][43][06][29][80][80][00][02][71]
left=[aa][55][43][06][29][76][81][00][02][68]
right=[aa][55][43][06][29][8a][81][00][02][7c]
forward=[aa][55][43][06][29][80][e3][00][02][d4]
reverse=[aa][55][43][06][29][80][36][00][02][27]
light on=[aa][55][43][04][1e][01][01][65]
light off=[aa][55][43][04][1e][02][01][66]

echo -ne "\xaa\x55\x43\x06\x29\x80\x80\x00\x02\x71" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x06\x29\x80\xe3\x00\x02\xd4" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x06\x29\x80\x36\x00\x02\x27" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x06\x29\x8a\x81\x00\x02\x7c" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x06\x29\x76\x81\x00\x02\x68" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x04\x1e\x01\x01\x65" > /dev/ttyUSB0
echo -ne "\xaa\x55\x43\x04\x1e\x02\x01\x66" > /dev/ttyUSB0
```

gtxaspec, 2022-07-05:
@endertable https://s3-us-west-2.amazonaws.com/wuv2/upgrade/HL_CAMTC/firmware/4.55.0.105.tar

endertable, 2022-07-05:
You are awesome!!   Got updated stuff off of it to run my "car" cam.  But "car" as in full sized car   :)

gtxaspec, 2022-07-07:
audio is broken in the car fw with libcallback, test with refactored libcallback

endertable, 2022-07-07:
What I like about the car FW is that you can use it as a standalone cam as it has AP mode built in.  It obviously starts the IOTC stream in AP mode ( IOTC_Initialize2() ) without needing internet unlike the online FWs.  For car FW you must use the Wyze Car app and the resolution in AP mode is horrible, or last least for me it is.  
The tinkering continues..

gtxaspec, 2022-07-14:
maybe those functions for AP standalone mode are present in the regular binary...

endertable, 2022-07-14:
The only binaries I have to add to regular FW (for over a year now) has been the hostapd binary which allows one to turn the Cam into a "hotspot" or AP mode.  Was using their old binary from their V2 RTSP and now got the new updated version from Car FW, hehe.  They don't include that binary in their "regular" FW.  If you touch "/configs/.toycar" it goes into AP mode after reboot and the iCamera will still start the IOTC streams ( IOTC_Initialize2() ) even without internet.  
I haven't figured out how to make the regular FW start the stream without internet.  So on we go...   :)

gtxaspec, 2022-07-14:
working in https://github.com/gtxaspec/wz_mini_hacks/commit/84c893e593678016359d8909c4addaf8c9855d65

closed 2022-07-14 by gtxaspec