gtxaspec/wz_mini_hacks #157: Rtsp no video issue using two streams

jemershaw, 2022-07-04:
Okay, I'm opening this #121 because I was mistaken it seems like there's some session timeout or something with icamera. I've been using this configuration with a reolink and the old wyze rtsp firmware and hadn't had any problems with the stream via rtsp-simple-server.

I currently have two streams on wyze and rtsp-simple-server proxies the connection to wyze on init. It seems like if the first connection from client -> rtsp-simple-server -> wyze-cam stays active then you get both audio/video but the second only gets audio only it's almost like a continuous stream eventually drops the video.

I can't seem to find the icamera sources to see if this is a known bug in that application or not. I can update the wyze logs if it happens.

clee604, 2022-07-04:
Is this on a V2 because I've been trying to figure out why the video feed crashes after some time. I also use rtsp-simple-server as a proxy (on demand, it opens and closes the stream as needed) and the V2 video stops working after some time (usually overnight or a few hours in the day). I see some small consistent file sizes after in the recordings which I assume are audio only.

This doesn't happen on my V3s though, even though its the same rtsp-simple-server proxy set up. They've been stable and video stream stays up.

jemershaw, 2022-07-05:
@clee604 can you run `/opt/wz_mini/bin/v4l2rtspserver -V` on the V3 camera?

 I'm running 4.36.9.139 wyze firmware and the v4l2rtspserver is 0.3.3-4-ga43e094-dirty.

clee604, 2022-07-05:
Also 0.3.3-4-ga43e094-dirty on both V2 and V3. 

I just noticed you mention you have two streams though, is that two RTSP streams on the camera (ie. you have both hi-res and low-res streams) or do you mean two client streams on the camera (ie. the rtsp server + another device)

clee604, 2022-07-05:
I was browsing some of the issues on another V2 firmware and found these two which seem similar to my issue
https://github.com/openmiko/openmiko/issues/62

For curiosity I ran windows disk scan and repair on some sd cards and windows indeed did pick up some errors. Not sure if it's due to the camera getting powered off or rebooting, it could be a cause for some of the issues?

@gtxaspec in this thread they have some kind of filesystem check for the Miko firmware, is something similar possible here? I'm not sure if it will fix anything, but the fact there's file system errors detected by windows can't be a good thing....
https://github.com/openmiko/openmiko/issues/70

endertable, 2022-07-05:
Wyze binaries run the following to remove the dirty bit and repair the FS, if needed:

 tf_prepare --blkdev /dev/mmcblk0p1 --strategy=2 --samplecnt=64 

Try that and see if it works for you..

clee604, 2022-07-05:
just so i'm clear, I run that in the command line via ssh on the device where I suspect the memory card is corrupted?

gtxaspec, 2022-07-05:
@endertable thats right!

@clee604, no, this command has to be run early in the boot process, it doesn't work when run manually.  But, I will include `fsck.vfat` in an upcoming release of wz_mini

jemershaw, 2022-07-09:
@gtxaspec I think you added the fsck to the release? That still didn't fix my problem I have to every few hours to log into my rtsp server and restart the ffmpeg session. I can capture logs if it would be helpful. I know it's not the command I'm using because it worked with the wyze rtsp firmware and my other cameras have no problems.

@clee604 I have high and low setup. I have frigate connect record high res and detect on low res which all connect through rtsp-simple-server.

jemershaw, 2022-07-10:
It actually seems like maybe the problem with the cameras. They were activated earlier this year and they are the only two having problems. Might have to see if I can rma it from wyze.

jemershaw, 2022-07-12:
Well it turned out the camera I moved ended up having the same problem but this time video1 doesn't work and video2 is black but audio works no problem.

jemershaw, 2022-07-12:
I'm able to recover by running `cmd video off` then `cmd video on` not sure what that does but is there a possibility at startup there is a sync problem between the os and hardware?

gtxaspec, 2022-08-20:
test the latest versions, they have improved the libcallback library. resolving for now

closed 2022-08-20 by gtxaspec