gtxaspec/wz_mini_hacks #161: Wyze Cam v3 no stream audio when loading wz_mini_hacks

mrsarge, 2022-07-05:
Brand new Wyze Cam v3.
If I boot it with the Micro SD card inserted, thus loading wz_mini_hacks, I have no audio in the Wyze app.
Also tried to enable the RTSP stream and I have no audio in there either.

Removed the Micro SD card, rebooted and audio is working properly in the Wyze app.

Firmware 4.36.9.139

gtxaspec, 2022-07-05:
Just tested, on 139, using the latest non beta wyze app (android) sound works fine with wz_mini.  
What platform? android or ios?
latest wz_mini release?

mrsarge, 2022-07-05:
Android.
Downloaded the master.zip yesterday.

mrsarge, 2022-07-05:
Since everything is fine when booting without the Micro SD, the app should be ok.
I also just tested the Speak feature and it doesn't work when using wz_mini.
It's like the speaker and microphone were disabled.

(I got the audio prompt from wz_mini when I did the first boot)

gtxaspec, 2022-07-05:
let me try a fresh install and check stand by

gtxaspec, 2022-07-05:
confirmed, seems like some sort of regression.  looking into it now

capncybo, 2022-07-05:
So that's why that "Rat-Bird" Magpie I tried to Scare off... never flew away. LOL

gtxaspec, 2022-07-05:
found the issue, new release soon

gtxaspec, 2022-07-06:
fixed in latest release, please test

mrsarge, 2022-07-06:
It works!

Thank you for the quick fix.

gtxaspec, 2022-07-07:
resolved for now =D

closed 2022-07-07 by gtxaspec