gtxaspec/wz_mini_hacks #163: cmd waitmotion issue & clarification

capncybo, 2022-07-06:
I only started testing "cmd waitmotion __" yesterday by using:
cmd waitmotion 20
no luck, so I tried longer timeouts... cmd waitmotion 60
Nope... cmd waitmotion 1000
Hmmmm...

Perhaps I was misunderstanding the example & functionality.
However I did discover that using... ctrl+c would break any future attempts to use this command, until after rebooting the camera.
+ the documentation didn't mention it but there seems to be a minimum timeout required as: >10

But regardless of these findings... I could not seem to generate any text output...
-Should this work with Continuous recording to microSD enabled, or do we have to select Events Only under Advanced settings?
-Under detection settings... will it work **without** any Detection Zone set?
-Is there a recommended/required minimum for MOTION DETECTION SENSITIVITY ex) 50?

And perhaps it was just fixed via the latest update, as I haven't tried since yesterday (mid-day)

gtxaspec, 2022-07-06:
you need to have motion tagging enabled in the app

capncybo, 2022-07-06:
In WYZE Android Application I have:
Settings > Detection Settings > MOTION DETECTION SENSITIVITY: Low 1 --- 100 High  (tired a few different values, 50, Nope)
Detection Zone: Off  (Note: I tried enabling every square, but no luck)
Settings > Event Recording > **Detects Motion**  off
Is this possibly what you mean by "motion tagging enabled in the app" ???

capncybo, 2022-07-06:
Oh for F__K's  sake... I'm really starting to HATE the non-intuitive, non-consistent, manner that certain app developers program & stick options anywhere.  Shouldn't that "Motion Tagging" option also be included & accessible somewhere within settings?
Sorry, Rant over... I know it's not your application.
But what was so wrong with the file menu structure, I mean in "most" counties we read from: Left to Right & Top - Down
Thanks for pointing out, my stupidity...
Jesus

capncybo, 2022-07-06:
But for Now, the only text output that I'm seeing is...
clear
No co-ordinates

gtxaspec, 2022-07-06:
theres a bug (or change) in the newer wyze implentation vs the atomcam, which is what the waitmotion function is based on, so you need to run it in a loop for now.

capncybo, 2022-07-07:
Thanks for all the help & explanations... I will endeavor to put the waitmotion function into a loop & retest.

gtxaspec, 2022-07-12:
fixed in latest libcallback release, upcoming wz_mini release

capncybo, 2022-07-12:
To clarify: Which Bug?
-the Timeout not working
-Ctrl-c Crashing future attempts to run: cmd waitMotion

gtxaspec, 2022-07-12:
timeout not working - fixed
ctcl-c crashing future attempts to run - you have to wait until the original waitMotion period you specified has expired to run it again.

gtxaspec, 2022-07-13:
out now in https://github.com/gtxaspec/wz_mini_hacks/commit/68b0e2092ea8e64b679fad60f1c61e12b514eeee

gtxaspec, 2022-08-20:
working as intended for now.

closed 2022-08-20 by gtxaspec