gtxaspec/wz_mini_hacks #171: MQTT?

guarddog13, 2022-07-10:
Can this turn on the Wyze Cam V3 as an MQTT client? Does it pub it's motion events?

gtxaspec, 2022-07-10:
see https://github.com/gtxaspec/wz_mini_hacks/discussions/75

guarddog13, 2022-07-11:
@gtxaspec thank you for the quick reply. I was looking at the reply from Matthew...does that script go into the camera's SDCARD /bin as a .bin file? This is what I would assume anyway...

I am using Agent DVR but it's very intensive to run 24/7 motion detection...and I'm so not paying for IFTTT. Using this I can trigger Agent's detector on and off via HA. This is even better than MQTT.

guarddog13, 2022-07-11:
If I could use the cmd to trigger based on the camera's smart detection I could elimate the agent dvr lol.

gtxaspec, 2022-07-11:
you would place it in a file called something like `motion.sh` and then give it execute permissions `chmod +x motion.sh` and then shove in somewhere on the sdcard

guarddog13, 2022-07-11:
Thats what I've done. I also seen the custom script command in the config so called its path there? I'm assuming that's correct?

gtxaspec, 2022-07-11:
if you are using the latest version just place it in `/opt/wz_mini/etc/rc_local.d/`

guarddog13, 2022-07-11:
Would I call for it in the config still? I put it at /opt/wz_mini/[my own file called Scripts] then I put that path in the configuration. That seems to be working. Is putting it in opt/wz_mini/etc/rc_local.d a better option or does it matter?

gtxaspec, 2022-07-11:
no need to put the path in if you put it in the rc_local.d

guarddog13, 2022-07-11:
I'll try it like that then. Thank you so much. Thank you for this, as well. This is amazing and so much faster than *cough* IFTTT.

guarddog13, 2022-07-11:
All setup on all 3 cams. Wow this is amazing btw. Was able to flash the regular firmware back, looks like I get to keep all features plus now my agent dvr is only consuming 20 percent of my CPU compared to 60 before. Gracias

closed 2022-07-13 by guarddog13