gtxaspec/wz_mini_hacks #174: Don't require a wifi connection to use usb-ethernet

gtxaspec, 2022-07-12:
### Discussed in https://github.com/gtxaspec/wz_mini_hacks/discussions/173

<div type='discussions-op-text'>

<sup>Originally posted by **UncleJohnsBnd** July 11, 2022</sup>
Does a Wi-Fi signal need to be available for the USB network adapter to be able to work?

I setup a brand new Cam3 with no SD card and got it working on the app.

Formatted 256gb SD to Fat32 and copied all the files in SD_ROOT over to the SD.  Which is one folder(ad everything in that folder) and one file.

Edited the wz_mini_config with NotePad++
ENABLE_USB_ETH="true"

Turned off camera

Turned off my 2.4 Wi-Fi

Put SD card in Wyze and started it up.  It just blinked for connection.... nothing on network adapter.

Turned off camera

Turned 2.4 Wi-Fi back on

Turned on camera

Camera connects

USB network adapter shows connected and data

Turn off 2.4 wifi and USB network card remains connected and everything runs

Just seems I can't get the usb network card to connect without there initally being wifi available.  Is that to be expected?</div>

gtxaspec, 2022-07-17:
@UncleJohnsBnd this should be fixed in the latest release today, can you test?

thanks

UncleJohnsBnd, 2022-07-18:
No... Not working.... I get a steady blue flashing light with USB Ethernet set to true with the 2.4 on or off.  App shows connected but no feed, steady fast flash blue.

If I disable USB Ethernet I then connect but on wifi only.

gtxaspec, 2022-07-19:
try doing a fresh install, working well for me on multiple cameras.

UncleJohnsBnd, 2022-07-19:
@gtxaspec I'll try again..... This is what I did. 

Did a re-format
Installed new version
Let camera boot up and connect
Shut down
Updated config for USB = true
Restart

Starts up..... Fast flash blue light...  Like it is stuck in loop. 

I'll let you know what happens this go around.

UncleJohnsBnd, 2022-07-19:
@gtxaspec  yes.... It does work.  I forgot I had added in an extension cable so I decided to remove it and now it works just fine.

I thought this extension would carry the power and data but apparently not.  Need to find one that does...

CY Micro USB 2.0 Type B 5Pin Male... https://www.amazon.com/dp/B00S6AZCHW?ref=ppx_pop_mob_ap_share

gtxaspec, 2022-07-21:
resolved! =D

closed 2022-07-21 by gtxaspec