gtxaspec/wz_mini_hacks #175: Won't Load

guarddog13, 2022-07-12:
I'm struggling quite a bit with this on installating it on my 4th camera. For some reason it just won't load the wz_mini. I was having this issue last night as well but it eventually stuck. Not sure what i can do to get it going on this camera.

capncybo, 2022-07-12:
Did you try & power down one of your other "working cameras"...
remove it's microSDcard & try inserting it & powering up the problematic one?

guarddog13, 2022-07-12:
Just tried that.. with a fresh install of the latest firmware still doesn't boot up. Red light turns on then off. Setup button doesn't respond either.

gtxaspec, 2022-07-12:
try a fresh install again., boot the camera, wait a while, if it still doesnt boot, pull the sd card and check the logs folder inside the wz_mini directory.  any logs there?

guarddog13, 2022-07-12:
@gtxaspec There's 2 files in there. One called .gitignore and one called wz_inti.log. 

This is what's in it. 

+ echo 'replace stock busybox'
replace stock busybox
+ mount --bind /opt/wz_mini/bin/busybox /bin/busybox
+ echo 'replace stock fstab'
replace stock fstab
+ mount --bind /opt/wz_mini/etc/fstab /etc/fstab
+ echo 'mount wz_mini tmpfs'
mount wz_mini tmpfs
+ mount -t tmpfs /opt/wz_mini/tmp
+ echo 'install busybox applets'
install busybox applets
+ mkdir /opt/wz_mini/tmp/.bin
+ /opt/wz_mini/bin/busybox --install -s /opt/wz_mini/tmp/.bin
+ '[' -b /dev/mtdblock9 ]
+ '[' -b /dev/mtdblock6 ]
+ mkdir /opt/wz_mini/tmp/configs
+ mount -t jffs2 /dev/mtdblock6 /opt/wz_mini/tmp/configs
+ sed -e 's#.*=\(\)#\1#'
+ grep PRODUCT_MODEL
+ cat /opt/wz_mini/tmp/configs/.product_config
+ touch /opt/wz_mini/tmp/.WYZE_CAKP2JFUS
+ touch /opt/wz_mini/tmp/.T31
+ umount /opt/wz_mini/tmp/configs
+ rm -rf /opt/wz_mini/tmp/configs
+ '[' -f /opt/wz_mini/tmp/.HL_PAN2 ]
+ '[' -f /opt/wz_mini/tmp/.WYZE_CAKP2JFUS ]
+ GPIO=63
+ '[' -e /opt/wz_mini/etc/.first_boot ]
+ echo 'first boot, initializing'
first boot, initializing
+ '[' -f /opt/wz_mini/tmp/.T20 ]
+ insmod /opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/audio.ko 'spk_gpio=63' 'alc_mode=0' 'mic_gain=0'
+ /opt/wz_mini/bin/audioplay_t31 /opt/wz_mini/usr/share/audio/init.wav 50
[INFO]        : Playing the audio /opt/wz_mini/usr/share/audio/init.wav file.
[INFO]        : Volume is set to 50.
+ rmmod audio
+ '[' -f /opt/wz_mini/etc/.first_boot ]
+ echo 'Set first_boot'
Set first_boot
+ touch /opt/wz_mini/etc/.first_boot
+ echo 'replace stock inittab'
replace stock inittab
+ mount --bind /opt/wz_mini/etc/inittab /etc/inittab
+ echo 'bind /etc/profile for local/ssh shells'
bind /etc/profile for local/ssh shells
+ mount --bind /opt/wz_mini/etc/profile /etc/profile
+ echo 'mounting global tmpfs'
mounting global tmpfs
+ mount -t tmpfs /tmp
+ echo 'mount system to replace factorycheck with dummy, to prevent bind unmount'
mount system to replace factorycheck with dummy, to prevent bind unmount
+ '[' -f /opt/wz_mini/tmp/.T31 ]
+ mount /dev/mtdblock3 /system
+ mount --bind /opt/wz_mini/bin/factorycheck /system/bin/factorycheck
+ touch /tmp/usrflag
+ echo 'create workspace directory'
create workspace directory
+ mkdir /opt/wz_mini/tmp/.storage
+ echo 'copy stock rcS'
copy stock rcS
+ cp /etc/init.d/rcS /opt/wz_mini/tmp/.storage/rcS
+ echo 'add wz_post inject to stock rcS'
add wz_post inject to stock rcS
+ sed -i '/^".*/aset -x' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/^# Run init script.*/i/opt/wz_mini/etc/init.d/wz_post.sh\n' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/sbin:/s/$/:\/opt\/wz_mini\/bin/' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/system\/\lib/s/$/:\/opt\/wz_mini\/lib/' /opt/wz_mini/tmp/.storage/rcS
+ '[' -f /opt/wz_mini/tmp/.T20 ]
+ echo 'Copy factory app_init.sh'
Copy factory app_init.sh
+ cp /system/init/app_init.sh /opt/wz_mini/tmp/.storage/app_init.sh
+ echo 'Replace factory app_init.sh path'
Replace factory app_init.sh path
+ sed -i '/\/system\/init\/app_init.sh/,+4d' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/Run init script.*/a /opt/wz_mini/tmp/.storage/app_init.sh\n' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/\/system\/init\/app_init.sh/,+2d' /opt/wz_mini/tmp/.storage/rcS
+ echo 'replace stock password'
replace stock password
+ cp /opt/wz_mini/etc/shadow /opt/wz_mini/tmp/.storage/shadow
+ '[[' false '==' true ]]
+ mount --bind /opt/wz_mini/tmp/.storage/shadow /etc/shadow
+ chmod 400 /etc/shadow
+ '[[' -e /opt/wz_mini/swap.gz ]]
+ '[' -f /opt/wz_mini/tmp/.T20 ]
+ insmod /opt/wz_mini/lib/modules/3.10.14__isvp_swan_1.0__/extra/audio.ko 'spk_gpio=63' 'alc_mode=0' 'mic_gain=0'
+ /opt/wz_mini/bin/audioplay_t31 /opt/wz_mini/usr/share/audio/swap.wav 50
[INFO]        : Playing the audio /opt/wz_mini/usr/share/audio/swap.wav file.
[INFO]        : Volume is set to 50.
+ rmmod audio
+ echo 'swap archive present, extracting'
swap archive present, extracting
+ gzip -d /opt/wz_mini/swap.gz

guarddog13, 2022-07-12:
On a sidenote I put the card back in the other camera and now the custom script I had in rc.local.d isn't working anymore.

gtxaspec, 2022-07-12:
do you hear the audio prompt on the camera that doesn't boot? is it the regular V3? or a floodlight or something else?

guarddog13, 2022-07-12:
It is the regular V3 and I did hear both audio prompts about first boot and extracting swap. It appears that it stops right at the extracting swap so I'm going to try to set that to false and see what happens.

gtxaspec, 2022-07-12:
try a different memory card if you can (i know you already tried one from another camera)

guarddog13, 2022-07-12:
Thats my next goal i have to either find one or go grab one. Wifey gonna grab one for me lol.

guarddog13, 2022-07-13:
New SD card, same behavior. I finally got it. I ended up doing everything completely clean lol. 
1. Put the firmware back on cam. 
2. Reformatted fat 32
3. Didn't edit any of the SD Root except the ssh file. Seems that edited the conf file before transferring causes issues.

closed 2022-07-13 by guarddog13