gtxaspec/wz_mini_hacks #177: cURL Support?

guarddog13, 2022-07-13:
I know I asked in a comment....(newb move lol). 

But figured I'd open a feature request too, as I it's fairly important to my goals. cURL support would open up all kinds of possiblilities to use motion for triggers it would open up things like maker api calls via Hubitat, could make POST requests to basically any home auto software that supports it. My idea is to skip even Home Assistant and have the camera talk directly to iSpy's api and their api doesn't respond to --post-data from wget. Thank you for being so responsive and keeping wz_mini updated. I updated all my cameras this morning and am working on selling off my Nest Cams and replacing them with thes wonderful little Wall-E looking things haha.

gtxaspec, 2022-07-13:
I thought I had included both wget and curl, oops.  I'll throw in curl for you.

guarddog13, 2022-07-13:
That's awesome cURL would be amazing.

gtxaspec, 2022-07-13:
added in https://github.com/gtxaspec/wz_mini_hacks/commit/7e56f96e443e1e02dcc7cba2ff743fe14bf7264c

closed 2022-07-14 by guarddog13