gtxaspec/wz_mini_hacks #179: Branch no-conntrack/video regression

jemershaw, 2022-07-14:
I'm just wondering what the status was for the no-conntrack branch?

I think it fixed some of my connection problems  with the v3 and the latest firmware.

I think I might have ran into something similar to https://github.com/gtxaspec/wz_mini_hacks/issues/161 but with the video being a black screen in rtsp(this was using the master branch)

gtxaspec, 2022-07-14:
@jemershaw issues even with today's latest release?

jemershaw, 2022-07-14:
I just installed it, it usually takes a few reboots etc before the video on startup has problems.

gtxaspec, 2022-07-14:
ok let me know, if you still have issues I will update the conntrack branch too.

jemershaw, 2022-07-14:
Yes it looks like it is still happening on the latest master release. My guess is because I have rtsp-simple-server trying to connect once it is listening.

gtxaspec, 2022-07-14:
@jemershaw and it only works with nocontrack?

jemershaw, 2022-07-14:
I think it was still happening on that branch. I only ran into the audio problem in the no-conntrack branch but I didn't test that much so let me redeploy.

jemershaw, 2022-07-14:
The video seems fine but the audio doesn't seem to want to work.

gtxaspec, 2022-07-14:
so:
master no video
no-conntrack video ok, audio broken

jemershaw, 2022-07-14:
@gtxaspec that is correct. This is for a V3 camera.

gtxaspec, 2022-07-14:
and it only happens when using `rtsp-simple-server`?

jemershaw, 2022-07-14:
I haven't been able to time it right to create a connection without rtsp-simple-server. It seems like the lastest master release over some time the rtsp session for the tcp connection loses video. Still unclear why.

gtxaspec, 2022-07-14:
have you tried using UDP instead of TCP?

jemershaw, 2022-07-15:
I didn't try, do you know if it is it listening by default?

gtxaspec, 2022-07-15:
it listens on udp and tcp by default

gtxaspec, 2022-07-21:
any update? still having issues?

jemershaw, 2022-07-22:
I'm actually doing mapping with ffmpeg so udp doesn't work most of the time because video sometimes is corrupted.

jemershaw, 2022-08-02:
I even tried bypassing rtsp-simple-server and just use frigate and sometimes it works and sometimes the videos don't get recorded. I had to resort to putting back on the stock rtsp version since that doesn't have any problems.

jemershaw, 2022-08-02:
Also some of the recordings are just a black screen. I'm not sure if anyone else is using frigate and is having this problem or not.

gtxaspec, 2022-08-20:
I would post in the discussions area for other frigate users or rtsp-simple-server users to see, and if possible share setup information to fix your issue.  resolving for now

closed 2022-08-20 by gtxaspec