gtxaspec/wz_mini_hacks #180: Keep wz_mini alive

SomebodySysop, 2022-07-14:
I just installed wz_mini_hacks on V3 cam.  I have only enabled usb_ethernet and rtsp.  It runs for a few hours, then for some reason stops (crashes?) and WyzeCam wifi returns.  In addition, I can no longer ssh into the unit without restarting it.

I'm wondering is there some way to keep wz_mini alive?  Some way for the server itself to detect when it has stopped, and restart it?

I do not know why it keeps crashing.

gtxaspec, 2022-07-14:
check the logs folder inside wz_mini/logs

you can also run `gather_wz_logs.sh` and upload the archive here for analysis

SomebodySysop, 2022-07-14:
Attached is my log .gz file.  

Suggestions on what I should be looking for in logs?

[log_gather_2022-07-14_T0423.tar.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9108457/log_gather_2022-07-14_T0423.tar.gz)

gtxaspec, 2022-07-14:
what ethernet adapter are you using? 
```
[   19.455533] usb 1-1: USB disconnect, device number 2
[   19.460761] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -143
[   19.529668] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   19.804859] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   19.964895] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to read reg index 0x0006: -19
[   19.985509] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0005: -19
[   20.053971] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.080456] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.130986] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.236045] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.302014] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   20.321487] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.414919] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   20.471621] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to read reg index 0x0001: -19
[   20.526274] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.606239] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to read reg index 0x0000: -19
[   20.666344] ax88179_178a 1-1:2.0 eth0: register 'ax88179_178a' at usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet, 46:ae:52:33:4f:3f
[   20.781869] ax88179_178a 1-1:2.0 eth0: unregister 'ax88179_178a' usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet
[   20.845206] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   20.869519] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   20.878850] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   20.910730] usbcore: registered new interface driver ax88179_178a
[   21.059124] cfg80211: Calling CRDA to update world regulatory domain
[   21.264879] usb 1-1: new high-speed USB device number 3 using dwc2
[   21.654353] usb 1-1: device v0b95 p1790 is not supported
[   22.325645] ax88179_178a 1-1:1.0 (unregistered net_device): Failed to read reg index 0x0040: -32
```

gtxaspec, 2022-07-14:
also, checkout today's release, there are some bug fixes.

SomebodySysop, 2022-07-14:
This adapter: https://www.amazon.com/dp/B07M5X9795

With this setting:

ENABLE_USB_ETH="true"
ENABLE_USB_ETH_MODULE_AUTODETECT="false"
ENABLE_USB_ETH_MODULE_MANUAL="ax88179_178a"

gtxaspec, 2022-07-14:
does it work if you use "cdc_ether" instead of the ax88179 driver?

SomebodySysop, 2022-07-14:
> does it work if you use "cdc_ether" instead of the ax88179 driver?

No.

SomebodySysop, 2022-07-14:
I was following the lead of the guy who said:

You beat me to it. I've just received an adapter like yours (but no brand) and it is exactly an ASIX AX88772B USB 2.0 Ethernet. Works by enabling usbnet.ko and asix.ko drivers.
https://github.com/gtxaspec/wz_mini_hacks/discussions/73#discussioncomment-2926703

gtxaspec, 2022-07-14:
try asix instead of ax88179_178a

SomebodySysop, 2022-07-14:
> try asix instead of ax88179_178a

I tried asix and also auto detect.  Neither works.  The only one that works is ax88179_178a

I did, however, download and copy the latest zip build.

SomebodySysop, 2022-07-14:
Crashed again this morning.  I modified wz_mini.conf:

ENABLE_USB_ETH_MODULE_MANUAL="asix, ax88179_178a, cdc_ether, r8152"

This was in dmesg.log:

[   28.671915] usbcore: registered new interface driver asix                                                                                          
[   29.316754] ax88179_178a 1-1:1.0 (unregistered net_device): Failed to read reg index 0x0040: -32                                                   
[   29.505226] ax88179_178a 1-1:1.0 eth0: register 'ax88179_178a' at usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet, f8:e4:3b:12:ba:dd             
[   29.570901] usbcore: registered new interface driver ax88179_178a                                                                                  
[   29.633146] usbcore: registered new interface driver cdc_ether                                                                                     
[   29.701953] usbcore: registered new interface driver r8152                                                                                         
[   29.768663] cfg80211: Calling CRDA for country: US                                                                                                 
[   30.451098] ax88179_178a 1-1:1.0 wlan0: Failed to read reg index 0x0040: -32                                                                       
[   30.577874] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready                                                                                    
[   32.362302] ax88179_178a 1-1:1.0 wlan0: ax88179 - Link status is: 1         

Drivers not compatible with this device?

SomebodySysop, 2022-07-15:
Swapped out with this ethernet adapter which I purchased years ago: https://www.amazon.com/gp/product/B00MYT481C/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

With this OTG cable: https://www.amazon.com/dp/B07QGZK4FF?psc=1&ref=ppx_yo2ov_dt_b_product_details

Been working for the past 24 hours with no crash or reported fails in dmesg.log

Thank you!

gtxaspec, 2022-07-17:
user resolved, closing! =D

closed 2022-07-17 by gtxaspec