gtxaspec/wz_mini_hacks #182: lamp socket accessory testers needed

gtxaspec, 2022-07-14:
lamp socket accessory testers needed!  does the socket work with wz_mini?

mcondren, 2022-07-14:
It does. Just installed this week. Been running for about 24 hours. I did set it to on 24/7 in the Wyze app and 16 hours later found the light off again, but I was able to turn it back on again in the app. Will continue to monitor.

billotronic, 2022-07-27:
Tested on and off via app and it works fine. Will report back when I have a chance to test motion.

Wyze V3 4.36.9.139
Socket 0.9.025
wz_mini_hacks 07-25-22 (no version numbers?)

Thank you for this awesome project.

billotronic, 2022-08-03:
Ok, best I can tell, the socket works 100% as expected via the wyze app in regards to turning the light on and off both manually and with motion.

gtxaspec, 2022-08-03:
@billotronic awesome, thanks

gtxaspec, 2022-08-03:
taking this as a win =D

closed 2022-08-03 by gtxaspec

billotronic, 2022-09-06:
I think this needs to be reopened and tested some more. I am trying to operate my v3 and not have it trigger the lamp socket at night and it seems impossible. I have a control v3 setup with no hacks and I am able to disable the socket as expected.

Lamp socket firmware: 0.9.0.27

Is there something I can look for in logs for this?

CCelian, 2024-07-31:
I'd be interested to see if others have figured out how to control the socket while in self-hosted mode. I find that setting a bulb to off doesn't keep it off--after a few days, the bulb will turn on after it hasn't made contact with the official Wyze app in a while.