gtxaspec/wz_mini_hacks #184: Waitmotion timing out periodically

guarddog13, 2022-07-14:
Any ideas how I could watch for this timeout to happen and then possibly have another script restart the timed out script so i don't have to continually monitor all the cameras and reboot them to get it going again?

guarddog13, 2022-07-14:
I downloaded the most recent release yesterday after you provided cURL support. Has there been a more recent update? 

Here's the script I'm using. 

#!/opt/wz_mini/bin/bash

while true
do
  ret='cmd waitmotion 1000 | grep "timeout" | wc -l'

  if $ret -eq 0 ; then
    echo "Reporting motion!"
    curl -q http://myAgentDVR.com/api/
  else
    echo "No motion found!"
  fi
done

It runs like it should to begin. Something it wasn't doing before....you had to trigger motion right away before and then they were timing out on a regular basis. Now they just randomly timeout and require a reboot to start the script again.

gtxaspec, 2022-07-14:
you could also try something like

#!/bin/sh

while true; do
```
ret=$(cmd waitmotion 10)

if [[ "$ret" == *"detect"* ]]; then
echo "motion detected"
#cooldown time
sleep 5
else
echo "no motion"
fi

done
```

guarddog13, 2022-07-14:
Ugh how do the others get it to format right? Not having proper indents is driving this newb insane lol. #!/bin/sh vs... #!/opt/wz_mini/bin/bash? Does that make a difference?

gtxaspec, 2022-07-14:
in the github box, for the formatting you use three backticks 

```
```
code
```
```

guarddog13, 2022-07-14:
Hey look at that...thank you!

guarddog13, 2022-07-14:
cmd waitmotion 10 or 1000? So every 10 seconds that echos the "Reporting Motion" and sends the POST to my server. It's returning timeout but still sending the commands after the if...as if the if doesn't exist. I may have formatted it incorrectly. How you have it formatted throws all kinds of errors and it exits. lol

guarddog13, 2022-07-15:
All my cameras are running great now with that code. A new problem though lol. The scripts won't auto load on boot. I have to go into putty and ```nohup ./path/to/script & them``` to get them going and keep them going.

gtxaspec, 2022-07-15:
i think theres a bug in the script execution logic, stand by

guarddog13, 2022-07-15:
You're amazing...otherwise now that you provided the script they work great. I just have to manually start the script...which isn't a problem as long as I don't lose power and the script doesn't crash.

capncybo, 2022-07-15:
To get it to work...
I had to remove the first * just before "detect"* on the "if" line + the OCD in me got me tinkering so he's mine, for the cut-n-paste type-people ;-)
` ` `
#!/bin/sh

while true; do
ret=$(cmd waitmotion 10)

if [[ "$ret" == "detect"* ]]; then
  printf "\nMotion-detected "
  date
  #Do Something here
  curl -s "http://__________________"  > /dev/null
  #cooldown time: but ONLY sleep (after a detection) 
  sleep 5
    else
    #print PERIOD(s) instead of: "No motion" it looks nicer ;-) or just remove it via #
    echo -n "."
fi

done

` ` `

guarddog13, 2022-07-15:
What I have now runs great when I get back later I'll take that first * out. Did that get them to start on boot for you? That's my only issue now...this made these cameras worth every penny lol.

gtxaspec, 2022-07-15:
latest update has a fix for the script execution.

guarddog13, 2022-07-16:
> latest update has a fix for the script execution.

I'm still having issues with this on all 3 cameras.

gtxaspec, 2022-07-16:
its probably launching too early then.  add `sleep 10` right above `while true` and see if that works?

gtxaspec, 2022-07-16:
it works for me with sleep 10...

guarddog13, 2022-07-16:
I'll see if I have to do that if I have more issues. I started them all fresh did a couple of reboots and now all good. You're likely right cause it's random if it works or not. Plus no more errors with that first * gone.

closed 2022-07-16 by guarddog13