gtxaspec/wz_mini_hacks #187: do not upgrade to 4.36.10.2163

gtxaspec, 2022-07-14:
this release has significant back end changes, and  breaks all libcallback functionality.

endertable, 2022-07-14:
Wyze actively looking for hacks, but this also happens because people complain to them that a hack does not work and they need tech assistance for it or need to return their bricked or non-working device.  :(

Reminder for us all:  Do NOT go to vendor for any issues with this hack.

capncybo, 2022-07-15:
This is TERRIBLE News, I've been worried something like this would eventually happen. for a while now.
So, I just saved myself a backup of the Last working firmware.
But In the immediate to prevent future pain, for us, RTSP users... shouldn't we configure:
DISABLE_FW_UPGRADE="true" 
&
I doubt you've had very much time to reverse engineer whatever it is they've just done but in your expert opinion:
Do you think moving forward...
This hack is BUGGERED or most likely going to be much more limited.
I am not completely shocked, or surprised, that this occurred as tech companies always try to protect their IP.
But I really hoped WYZE was somewhat more developer friendly.
What all these Home Automation companies seem to forget is...
Many Different Products have Come & Gone, & the same is true for their Parent Companies.
And after they've lulled you into cloud-based subscription-type contacts...
If they Go Under or Drop Support, we're left with a bunch of paperweights instead of functional devices.
Can't they just leave the cleaver Nerds Alone?

ryny24, 2022-07-15:
Does 4.36.10.2163 only break RTSP, or does the V3 not boot if RTSP is enabled on 4.36.10.2163?

I just tried to change DISABLE_FW_UPGRADE="true", rebooted but my camera didn't come back. I'm using ethernet. After power cycling, changing the setting back, no help. I even downloaded the latest wz_mini_hacks and started fresh, but still no good.  Luckily, I had a saved SD_ROOT from Jul 8th, this worked but I think maybe because my config didn't have RTSP enabled. I enable it, and it doesn't come back. The app is still reporting version 4.36.9.139, and it works when I remove the card.

Maybe my V3 upgraded, but is still reporting the old version? My camera is on a roof and I'm tired of climbing up and down. Tomorrow I'll flash 4.36.9.139 and try again.

gtxaspec, 2022-07-15:
4.36.10.2163 only breaks RTSP and any libcallback related functions, for now.  

Let me test the DISABLE_FW_UPGRADE. stand by

guarddog13, 2022-07-15:
Well I guess I'll be downgrading cameras as I buy them....although the last two I bought from both Best Buy and Amazon had firmwares several versions old lol. 

I always keep the stable firmware right on my desktop lol.

gtxaspec, 2022-07-15:
4.36.9.139 is a pretty solid firmware version anyway

guarddog13, 2022-07-15:
My family is enjoying the auto lights without talking to Google or finding a light switch they never use...I won't be losing 139 as I'm so not getting yelled at by a 4 year old because the kitchen didn't turn on for him lmao. I may even download a couple older firmwares...just in case.

gtxaspec, 2022-07-15:
enabling disable_fw_upgrade and then hitting upgrade in the app causes iCamera to crash...sheesh.  Something changed there.  For now, don't enable it, since the new FW is beta anyway.  I'll have a fix for the crash issue coming up

sideup66, 2022-07-15:
I hope they do not send a C&D to gtxaspec :( . Though how could they? we are not modifying or reverse engineering their code in any way. The chinese version of this camera without their proprietary binary. If that scenario happens I wonder if there is a way we can straight fork from what wyze started with into our own firmware for the camera entirely....Is there any kernel level changes in this?

gtxaspec, 2022-07-15:
it's really a game of cat and mouse, with most manufacturers.  They close the loopholes, new ones are found.  Projects stop, new ones start.  The cycle continues.

gtxaspec, 2022-07-15:
latest version fixed `DISABLE_FW_UPGRADE="true"`

here is the behavior:

once the variable is enabled, the watcher will wait and detect an incoming update, once an update is detected on the device, it will force an immediate reboot.  The update will not continue, as they are only pushed once to the camera upon request.

gtxaspec, 2022-07-15:
wyze has pulled the beta of 4.36.10.2163 due to bricking of cameras ...

clee604, 2022-07-15:
what were the advertised benefits of this beta firmware anyways?

capncybo, 2022-07-15:
> wyze has pulled the beta of 4.36.10.2163 due to bricking (can't connect to wifi?) of cameras ... I had no issues but it looks like a bunch of folks did

Is it wrong of me to enjoy seeing WYZE having to retract & retreat a little, LOL

endertable, 2022-07-16:
It looks like they created iCamera into a huge super binary putting alot of the libraries into the one file.  It got to over 4MB in size from the mere 1.9MB in version 4.36.9.131.  I wonder if that's their way of having libcallback not work so easily anymore.
When you crunch so much into one thing there is a chance you're gonna break something..

gtxaspec, 2022-07-16:
right, before the libraries were dynamically linked and loaded, now they are statically compiled in.

capncybo, 2022-07-16:
IMO Newer is NOT always "Better" & regarding making it into a single BIG Binary... YUCK !!!  But that will certainly protect WYZE's Intellectual Property.  My concerns with them moving in this direction is not only them making it harder (if not near impossible) to hack.  But with WYZE stuffing almost everything into their own binary... they can probably send a C&D Order & say Hands-off our IP.  However, I always get confused regarding Linux devices because from what I see... Isn't WYZE themselves already using significant portions of GNU/GPL software inside their devices?  Of course, individuals rarely have the means to pursue legally should any company start to violate such open-source type licensing.  But I have seen Governments in certain EU countries succeed in enforcing licensing violations.  To myself it looks an awful lot like changing portions of code & not sharing certain aspects... However, I'm certainly not a team of Hot Shot Lawyers specializing in software lawsuits.  YUCK.

sideup66, 2022-08-17:
Is this still shaping up to be a thing in the next firmware release of the camera? Im curious if we can workaround it by having the custom kernel possibly load a custom icamera binary based on the old firmware version and load our libraries? though I dont know what other functionality that would impact...unless it can simply be a part of enabling rtsp...otherwise use stock icamera binary...

rickgitdone, 2022-08-18:
my V3 CAM quit connecting to rtsp and then noticed rtsp wasn't an active process when SSH'd in.. reverted CAM back to 4.36.2.5 and let it come up and see it in Wyze app... I have rsyn'd my SD_ROOT to the latest github after using a git pull ... can't see rtsp on the process list nor using netstat ... are they able to reach back into older firmware to block rtsp? 

I am going to re-clone the repo to a new directory and re-do the SD Card from scratch to see if there is a conflict somewhere...

closed 2022-10-26 by gtxaspec

cirrusflyer, 2022-11-02:
I upgraded to 4.36.10.2864 thinking this was no longer an issue.  My bad.  I can still SSH to the cameras, but not feed.  What's the recommended solution?  Just rollback?

rkrueger11, 2022-12-17:
>  the event you do accidentally upgrade, you can always roll back to 4.36.9.139, the recommended release for now.

How can I do this when Wyze is no longer providing the 4.36 firmware?

tavicu, 2023-04-19:
> How can I do this when Wyze is no longer providing the 4.36 firmware?

You can download the firmware from here: https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware

pjuhl2313, 2023-04-23:
Is anyone having issues with ethernet to their Wyze cams? I can RTSP over Wifi fine and am using an older firmware. However when I enable ethernet, the cams do not want to connect. What adapters is everyone using?