gtxaspec/wz_mini_hacks #190: V2 uBoot experiment

endertable, 2022-07-15:
@gtxaspec One of my V2 cams had a unique md5sum for /dev/mtd0, so I took your V2_kernel.bin and placed it as factory_ZMC6tiIDQN on the SD card.  Not really sure if that is the way it would work or if you have to compile that kernel file a different way for SD card, but the cam is reacting to the factory file.  Unfortunately, in this case, it is rebooting.  Would you like a copy or is that boat not worth sailing?

closed 2022-07-15 by endertable

gtxaspec, 2022-07-15:
@endertable 

name it `factory`
and see what happens.  

also you can analyze uboot: `dd if=/dev/mtd0 of=/opt/mtd0.bin` and then run `strings /opt/mtd0.bin | grep factory` and see if you get any output

endertable, 2022-07-15:
Yep, I grepped 'factory' and I got: factory_ZMC6tiIDQN
I renamed it to 'factory' and it didn't start wyze_mini, just started normal but the filename above is what puts it into a reboot-loop.

I guess I should also mention, this cam does not have a wz_mini modified kernel loaded on it..

So should it just be as simple as putting your V2_kernel.bin on the SD card as factory_ZMC6tiIDQN or do those kernels have to be compiled differently to boot off the SD card?

gtxaspec, 2022-07-15:
send me the mtd0.bin file with the md5sum and I'll test it

gtxaspec, 2022-07-15:
@endertable to confirm, this is from a V2? what color?

endertable, 2022-07-15:
It is a white V2 cam and i used the following command:  

dd if=/dev/mtd0 of=/tmp/mtd0

Sorry the final file I sent you was based on my script's naming convention of {Camera Name}_{MAC address}_{MD5 hash}_uBoot but it is the file from above command.  Most of my V2's uBoot have an MD5sum of b1f6f38dbb48d4e9d8911833aff0bc99

If you prefer, if you want to send me a factory_ZMC6tiIDQN, I can do the testing if you want so you don't risk bricking a cam.

Whichever way...

gtxaspec, 2022-07-15:
just wanted to make sure =D  I have a v2 wired up with a flash chip wired up externally.   so if i need to reprogram the flash chip no big deal

gtxaspec, 2022-07-15:
it works fine, and boot wz_mini.  i tested this by erasing /dev/mtd2, where the kernel lives on the camera.  then renaming v2_kernel.bin as the factory file, it boots it fine.  what a find!

gtxaspec, 2022-07-15:
@endertable you don't have any PAN camera's do you? lol

endertable, 2022-07-15:
I do, if my memory serves me they had the sane uboot md5 as the other V2’s.  I can confirm that when I get back to my server.

endertable, 2022-07-16:
yes, the pancam V1 have the same md5sum as most of the V2's:  b1f6f38dbb48d4e9d8911833aff0bc99

gtxaspec, 2022-07-16:
nice.  now we just need someone to be a guinea-pig and test flash the bootloader on the pan... sadly I don't have one anymore I borrowed the ones I used to test

endertable, 2022-07-16:
So I imagine you are using the flasheraseall commands?  What exact commands did you use to put the boot onto mtd0?

gtxaspec, 2022-07-16:
just

`flashcp -v file.bin /dev/mtd0`

no need to do flash erase all.  flashcp does the erase.

endertable, 2022-07-16:
Ok, I did it to Pancam and rebooted and I verified it took by doing an md5sum on a dd out.  So at minimum I know it works with the stock firmware so no reason it shouldn't work with the factory file.  This cam also has a V1 sensor on it's back and I just got back from doing stuff and am on my way out to go do stuff so when I get back later I can test it with wz-mini boot.

endertable, 2022-07-17:
Strangest thing, is that the cam that I got the uBoot off of reacts to the factory file but it goes into a boot-loop.  The yellow light goes off for a second then comes on to reboot.   Sometimes it will just go on to boot the stock FW kernel on mtd2, but it absolutely refuses to boot wz_mini.  I used the same SD card on other V2's after I reflashed their uBoot and no issue, just the one cam.  No logs created or nothing.  

Very strange

gtxaspec, 2022-07-17:
wierd I put it on all the v2s I have tomorrow and test.  can you access the serial console on the one that doesn't work (if you have the ability to)?

endertable, 2022-07-20:
Doing serial console access As always sounded interesting. I’m sure there are tutorials out there. Do you have some links to some? Or do you have your own tutorial and software/parts list?

gtxaspec, 2022-07-20:
do you have a raspberry pi? it's the simplest solution if you do... otherwise I can whip up a tutorial with photos

endertable, 2022-07-20:
I do have a raspberry pi.

gtxaspec, 2022-07-23:
[image]

endertable, 2022-07-23:
Ok.  Did you solder or purchase some special clip pins.  And you use a special getty from pi or telnet to cam or is it a kermit proggy?

gtxaspec, 2022-07-24:
you can use something like these:
https://www.amazon.com/Connector-Arduino-Raspberry-Including-Headers/dp/B0815S6P8Y
you really don't have to solder the headers on,  use the male to female ones, and just poke them through while you are reading the terminal

you can use `tio` on the rpi to connect.  enable serial console in raspi-config

endertable, 2022-07-24:
Sweet!!  Part on order. You requested the output of uBoot.  Do I redirect output of tio to file or does it have an auto capture log file.  I am new this particular stuff but it sounds fun. 😀

endertable, 2022-08-02:
[image]

looks like they may have changed the hardware.  Mine looks different.   But i believe those are the UART pons/holes on the lower left of the main board?  One is a squarez

gtxaspec, 2022-08-02:
yes, those look like the serial pins.

endertable, 2022-08-02:
OK, I connected all pins to pi pins (6,8,10) and to the 3 holes, executed:   tio -b 115200 /dev/serial0  then plugged in cam.
Nada.  One of the holes has a square around it and in one schematic that was for 3.3v but then where would the ground be?!?

Another guy stated from left to right GND, TX, RX.  Is there a way to test which one is ground cuz I am surprised I haven't fried this yet, hehe.

endertable, 2022-08-03:
Here is the log from the ?new? V2 that had the uBoot.  You can see the reboot call in there that creates the boot loop.

[tio1.log](https://github.com/gtxaspec/wz_mini_hacks/files/9247192/tio1.log)

gtxaspec, 2022-08-03:
can you pull the card, is there a log file called something initramfs.log on the root of the sd card

endertable, 2022-08-03:
There was no such file anywhere on the SD card.  Or any other new file for that matter..

gtxaspec, 2022-08-03:
can you try formatting the sd card and doing a fresh install? or using another sd card if you have it preferably.

it may be an error loading files from the  sdcard

gtxaspec, 2022-08-14:
any update? @endertable

endertable, 2022-08-14:
Sorry I forgot to post, yes it must have been the card.  I actually just reformatted the same card and it worked...