gtxaspec/wz_mini_hacks #194: version 20220718 - issues

virmaior, 2022-07-18:
I just tried to upgrading one of my cameras to the latest version. I ran into the following issues:
1. wyze_hacks seems to be incompatible with the wz_init.sh code (rectified by adding a line to remove wyze_hacks code from rcS).
2. usb direct doesn't appear to be working. The camera defaulted to wifi. Bonding didn't work. was able to manually raise the usb direct connection via wifi.
3. `cmd jpeg` appears to be busted on the latest version. It just seems to stall whereas the earlier versions  have all worked.

For wz_init.sh, I changed it like so:

```
...
sed -i '/sbin:/s/$/:\/opt\/wz_mini\/bin/' /opt/wz_mini/tmp/.storage/rcS
sed -i '/system\/\lib/s/$/:\/opt\/wz_mini\/lib/' /opt/wz_mini/tmp/.storage/rcS
sed -i  '/if/,+2d' /opt/wz_mini/tmp/.storage/rcS
..
```
and that kills the wyze_hacks issue (or so I think) 
See
[rcS-fixed.txt](https://github.com/gtxaspec/wz_mini_hacks/files/9130989/rcS-fixed.txt)
[rcS-prerun.txt](https://github.com/gtxaspec/wz_mini_hacks/files/9130990/rcS-prerun.txt)

wz_user.log is looping with
```
+ '[' -e /tmp/wpa_supplicant.conf ]
+ wait_wlan
+ true
+ grep HWaddr
+ ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr 7C:78:B2:60:A9:6C  
+ echo 'wlan0 hwaddr is up'
wlan0 hwaddr is up
+ store_mac
+ echo 'store original mac'
store original mac
+ tr '[:lower:]' '[:upper:]'
+ cat /sys/class/net/wlan0/address
+ break
+ echo 'wpa_supplicant.conf ready'
wpa_supplicant.conf ready
+ echo 'wlan0 not ready yet...'
wlan0 not ready yet...
+ sleep 1
+ true
+ wpa_check usb0
+ '[' -e /tmp/wpa_supplicant.conf ]
```

virmaior, 2022-07-18:
It looks like a regression came up 
where wpa_check changed.

not working code in 07-18-2022:
```
wpa_check() ....
	if [ -e /tmp/wpa_supplicant.conf ]; then
		wait_wlan
		echo "wpa_supplicant.conf ready"
	else
```

I think the versions from 7-17-2022 were working:
https://github.com/gtxaspec/wz_mini_hacks/blob/cfcad72bbfe0513831360d200a2e66f44fa6af3f/SD_ROOT/wz_mini/etc/init.d/wz_user.sh

https://github.com/gtxaspec/wz_mini_hacks/blob/3f7344111603dda48180a99e348efe4f9c792d74/SD_ROOT/wz_mini/etc/init.d/wz_user.sh

were probably still working.

gtxaspec, 2022-07-18:
ouch, I'll take a look.  thanks

virmaior, 2022-07-19:
confirmed fixed!

closed 2022-07-19 by virmaior