gtxaspec/wz_mini_hacks #195: web server UI

gtxaspec, 2022-07-19:
@virmaior 

the web config interface is awesome, i  have enabled it on all my devices!

1. after editing the config, all the spaces are removed from the wz_mini.conf file, is this by design?
2. after clicking update, I see if reboots right away, can we make rebooting an option, as in, "config updated, do you want to reboot now or go back to the config screen"
3. maybe we could add some text that will display in a pop up balloon or something that explains every option?

just some ideas

virmaior, 2022-07-19:
For 1, sheer incompetence on my part but built around the idea that the webpage is responsive to the configuration file (i.e., all new entries in the configuration file automatically are editable in the webpage because it derives its list from the configuration file rather than requiring separate maintenance). (I'll look into the spaces issues)
For 2, sounds good (I'll see when I can get to it).
For 3, the basic function is there. If you add a file with OPTION_NAME.md to the cgi-bin directory then it will display that text to the right. I just have hadn't the chance to fill them all in with the contents we already have. Switching it from displaying on the right hand side to popup balloon isn't all that complicated -- especially if we're planning to use jquery or some other front end tools.

gtxaspec, 2022-07-19:
@virmaior one more, maybe we could add an argument for jpeg.cgi for either 1 or 0 to indicate if we want to pull a 1080p or 360p image?

gtxaspec, 2022-07-19:
I forgot to mention, we do include bash, I remember in one of your commits you were talking about the ash vs bash intricacies.

virmaior, 2022-07-19:
@gtxaspec  was planning to check that out ... and then updating to the latest version temporarily clobbered my camera due to a regression error (cough). But hopefully I'll be able to update it to the latest version later today and work on these ideas.

ash vs bash, that's interesting. the cgi server definitely ran ash (which lacks associative array support) and required basically creating oodles of variables to compare the current wz_mini.conf vs the new one when creating a new one.

virmaior, 2022-07-20:
jpeg.cgi and #3 (partially) from the list above done in latest pull request.

virmaior, 2022-07-21:
added a version of the "do you want to reboot or keep editing?" (I'm not partial to how it's done). 

Also fixed an issue where it was only making 1 backup. ash vs bash strikes again (maybe I don't know the path to the current bash but merely changing it to /usr/bin/bash lead to an auto 404).

From the list, I just need to figure out why it's clobbering blank lines. I've got an idea as to why but will need to find the time to test that hypothesis.

gtxaspec, 2022-07-21:
you can try `#!/opt/wz_mini/bin/bash`

Vendo232, 2022-07-21:
web UI config? tell me more pls, I`m all in

virmaior, 2022-07-22:
@Vendo232 download latest master set WEB_SERVER_ENABLED=true in the config file.
then browse to ip address of your camera.
Currently three pages:
1. config page
2. car page
3. current camera jpeg page (your choice between channel=0 high res and channel=1 low res)

virmaior, 2022-07-22:
I've figured out the space removal issue.

difference between the behavior of
```
  for ARGUMENT in $(cat $hack_ini) 
```
where at least in ash, it's dropping the blank lines (possibly I've got the wrong IFS)
vs
```
  while IFS= read -r ARGUMENT; do
...
done < $hack_ini
```
which keeps them.

will upload when I've got a few more ideas sorted.

virmaior, 2022-07-23:
I think the latest pull request hits all of the issues here (https://github.com/gtxaspec/wz_mini_hacks/pull/205).

gtxaspec, 2022-07-24:
@virmaior looking real good! thank you so much

gtxaspec, 2022-07-25:
from `http://0.0.0.0/cgi-bin/config.cgi`, if you click on the "Revert Menu" link on the bottom, you get a 404 error

virmaior, 2022-07-25:
indeed. should be fixed with this pull request (https://github.com/gtxaspec/wz_mini_hacks/pull/206)..

gtxaspec, 2022-08-20:
webserver is live, resolving

closed 2022-08-20 by gtxaspec