gtxaspec/wz_mini_hacks #207: Edit authorized_keys file while system is online

SomebodySysop, 2022-07-28:
I installed winscp but it required that I create another public key in the putty format (my original key is in the ssh-2 format).  I have a wyzecam v3 running wz_mini_hacks to which I would like to add the new key without having to remove the sdcard.  I can still ssh in, but I need to add the new key in order to use winscp with it.

Problem is, when I ssh to the cam and try to edit /media/mmc/wz_mini/etc/ssh/authorized_keys, the file is ReadOnly.  I take it that particular mount is read only.

Any way to make this modification to authorized_keys without having to access that camera physically?

Thanks!

virmaior, 2022-07-28:
can you clarify what you mean by "try to edit"? Do you mean use ssh and use nano or some other means?

SomebodySysop, 2022-07-29:
> can you clarify what you mean by "try to edit"? Do you mean use ssh and use nano or some other means?

Log in using ssh.  cd -> /media/mmc/wz_mini/etc/ssh/ and use "vi" to open authorized_keys.  Even though file says user and group have write permission, vi reports ReadOnly.

gtxaspec, 2022-07-29:
I was able to use `nano /opt/wz_mini/etc/ssh/authorized_keys` to edit, no permissions errors.

permissions on my v3 are as follows:

`-rwxrwxrwx    1 root     root            36 Jul 29 03:42 /opt/wz_mini/etc/ssh/authorized_keys`

gtxaspec, 2022-07-29:
check `dmesg` to see if there are any errors relating to the memory card, if there is significant corruption, the kernel will set the card to `read only`

SomebodySysop, 2022-07-29:
Get error using vi or nano.

Yes, geting this in dmesg:

[52317.247307] FAT-fs (mmcblk0p1): error, invalid access to FAT (entry 0x186e3cce)
[52317.262092] FAT-fs (mmcblk0p1): error, invalid access to FAT (entry 0xef367dea)
[52317.304261] FAT-fs (mmcblk0p1): error, invalid access to FAT (entry 0x69bdfc48)
[52317.314553] FAT-fs (mmcblk0p1): error, invalid access to FAT (entry 0x7a012e39)
[177527.523211] ispcore: irq-status 0x00000600, err is 0x200,0x3f8,084c is 0x0
[243953.134367] ispcore: irq-status 0x00000600, err is 0x200,0x3f8,084c is 0x0
[299611.574493] ispcore: irq-status 0x00000600, err is 0x200,0x3f8,084c is 0x0
[361017.159562] ispcore: irq-status 0x00000600, err is 0x200,0x3f8,084c is 0x0

How do you suggest I fix?  Just use another card?

gtxaspec, 2022-07-29:
I would pull the card, format it as FAT32,  then copy the files back and try again.  If you still get errors, try another card, or just use another card to save time if you have it.  Test the suspect card later.

gtxaspec, 2022-08-20:
looks like the issue was the sd card, resolving for now!

closed 2022-08-20 by gtxaspec

SomebodySysop, 2022-08-20:
Yes, sorry, the issue was the sd card.  When I looked at it in the pc it showed as full.  Anyway, replaced card and all is well.  Thank you!