gtxaspec/wz_mini_hacks #208: No RTSP streams on V3

RyanHope, 2022-07-29:
The web server works but I can not connect to any RTSP stream. A port scan shows 80,22 and 4000 as only open ports.

virmaior, 2022-07-30:
web server and rtsp are separate.

have you set the config and after that let the camera **reboot**?

```

RTSP_LOGIN="admin"
RTSP_PASSWORD="password"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"

```

virmaior, 2022-07-30:
Also what version of wyze's firmware do you have? (the most recent beta one clobbers the hook that was making RTSP possible)

RyanHope, 2022-07-30:
I have those options enabled. I have tried with empty password (using auto MAC pass), I have tried setting a password. No combination seemed to get RTSP started. Was running on 4.36.9.139.

virmaior, 2022-07-30:
hmm...
1. what url are you trying to use for RTSP?
2. there's an image in the upper right that successfully updates in the config webpage right?

RyanHope, 2022-07-30:
1. I have tried /unicast, /unicast_video1, /unicast_video2 (if I recall)
2. yes, the video was working in the web server, both in the config view and the car/pan view

virmaior, 2022-07-30:
That you can see the image in the webpages confirms that the hook is in place that can make RTSP work.

Two things you haven't directly confirmed so far:
1. You've rebooted the camera after making the setting changes to enable RTSP right? (the settings only take effect on reboot)
2. What's the full url (not just the part on the end)?

another possibility that I haven't at all checked myself is that I wonder if there's a potential conflict between the recent local dnsmasq setup and RTSP (seems unlikey and something people would mention pretty quick).

That's pretty much it from me in terms of ideas. If there's anything funky in the logs or something more complicated going on, hopefully @gtxaspec has some ideas.

gtxaspec, 2022-07-30:
can you run `gather_wz_logs.sh` and upload it here? we can take a look and see whats going on

RyanHope, 2022-07-30:
I just got all my "production" cams setup with the RTSP beta fw from Wyze but I do have some other undeployed cams I can still test with but I need to find or buy some more sd cards first. Will get back on this asap.

gtxaspec, 2022-07-30:
make sure you are using a newer, non rtsp firmware with wzmini for best rtsp results.  the rtsp beta fw is old and has some issues.

RyanHope, 2022-07-31:
Is there a way to manually flash back to 4.36.9.139?

gtxaspec, 2022-07-31:
[flash the latest versi](https://download.wyzecam.com/firmware/v3/demo_wcv3_4.36.9.139.zip)

virmaior, 2022-07-31:
directions for how to flash: https://support.wyze.com/hc/en-us/articles/360031490871-How-to-flash-your-Wyze-Cam-firmware-manually

In order to flash it, the SD card must be formatted as FAT32 (same is required for wz_mini to run), because the bootloader can't read ExFAT. (Most 32gb and larger cards are formatted as ExFAT by default, so if unsure whether your card is ExFAT or FAT32 use an 8gb or smaller SD card)

RyanHope, 2022-07-31:
> 

yeah, i flashed it manually to the RSTP version so I know how to get back, just didnt have a link to the fw as I was normally using the in-app update

bjs-pdx, 2022-08-10:
I'm having the same problem with a PanCam V2 and a regular V2.  The pan has the video image on the web page, the v2 does not.  The V2 is running firmware 4.9.8.1002 and the PanV2 is on 4.49.1.47.  I'm trying to access the stream using VLC and none of the strings for accessing rtsp work, with or without authentication.  Log for VLC just says connection refused.

baudneo, 2022-08-10:
Have to tried video6_unicast and video7_unicast endpoints?

baudneo, 2022-08-10:
Can you run `netstat -tulpn` on your wyze v2 and v2 pan to see what ports are open and what processes have those ports open.

Have you done the gather logs command and browsed logs to see if there are any clues 8m there?

bjs-pdx, 2022-08-10:
Running netstat -tulpn from ssh on both cameras shows no listing for port 8554.  What is the command needed for gathering the logs?  This is all very new to me as I only installed the stuff a couple of days ago.

baudneo, 2022-08-10:
gather_wz_logs.sh

dskaplan, 2022-08-17:
This is silly and not a real fix, but @bjs-pdx I was having the same issues and changed the port of the streams to something different and it worked. After rebooting with new port, I changed it back to the default `8554`, rebooted, and was good to go. Worked for 3 different v3s for me.

RyanHope, 2022-08-17:
I thought I remember testing different ports with no luck but I can revisit that. I have new SD cards again and can resume testing/playing with this.

vincom, 2022-08-19:
fyi, v3 cam wyze rtsp fw. using blueiris v5, could get the rtsp stream using the wyze rtsp credentials and url.  if v3 has the wyze rtsp fw, and i insert scard w/mini hack i couldnt get blueiris to get the rtsp stream either with rtsp or http url. tried numerous urls and settings. web server is displaying the stream.
went back to stock fw and can use http url in blueiris to get the rtsp stream with the mini hack sdcard inserted

virmaior, 2022-08-20:
@vincom  the wyze provided rtsp firmware is trash so that's unsurprising. better to use the stock firmware and enable rtsp using wz_mini

vincom, 2022-08-20:
> @vincom the wyze provided rtsp firmware is trash so that's unsurprising. better to use the stock firmware and enable rtsp using wz_mini

agreed, it was just a fyi, cause i know users are going to try this hack on the wyze rtsp fw, but its just a headache, go to stock and no issues with this hack or atleast ur removing one variable for trouble shooting. 
also onec configured with wyze firmware just reflashing fw doesnt reset your wyze app settings, "no" reconnecting to your wifi steps needed

gtxaspec, 2022-08-20:
@RyanHope let us know if you have any update to see if we can help further!

vincom, 2022-08-20:
> @RyanHope let us know if you have any update to see if we can help further!

all good, even got streaming via port 80 on blueiris and android apps

gtxaspec, 2022-09-02:
aging, closing, feel free to re open or make a new issue/discussion if further issues persist.

closed 2022-09-02 by gtxaspec