gtxaspec/wz_mini_hacks #217: devices with failed wlan hw

gtxaspec, 2022-08-05:
@juanboro

I noticed your fork with the changes you made for failed wlan hw.  Do you mind if I roll these changes into master?  I am about to merge in a new init.d system and It would probably be easier than with the old one.

thanks!

juanboro, 2022-08-05:
Sounds great.  Yeah – it’s been on my to-do to put in a PR.  The camera I have with the failed wlan HW is at a remote location I don’t have physical access to all the time – so I haven’t been able to test with the latest updates (wz-mini-hacks has been running stably untouched on it for the past month though 😊).

 

Thanks!

juanboro, 2022-08-11:
I've merged the new init onto my current master branch.  I'll be able to test it out in a couple weeks when I have physical access to the camera with the broken internal wifi.

juanboro, 2022-08-24:
Submitted as PR  #241