gtxaspec/wz_mini_hacks #222: H264 Mainline/Baseline encoding causing errors.

baudneo, 2022-08-07:
I am trying to use Janus WebRTC with the RTSP streams provided but the encoding options are incorrect for the spec from what research ive done. The 2nd stream (360p) gets farther along than the main stream does but both fail.

# SubStream
```
v=0
o=- 1659831422573847 1 IN IP4 10.0.1.67
s=LIVE555 Streaming Media v2022.04.26
i=LIVE555 Streaming Media v2022.04.26
t=0 0
a=tool:LIVE555 Streaming Media v2022.04.26
a=type:broadcast
a=control:*
a=range:npt=now-
a=x-qt-text-nam:LIVE555 Streaming Media v2022.04.26
a=x-qt-text-inf:LIVE555 Streaming Media v2022.04.26
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=4d0032;sprop-parameter-sets=J00AMudAUBf8s1AQEB8AAAMAAQAAAwAoNGAE4gB1N//wKA==,KO48gA==
a=x-dimensions:640,360
a=control:track1
m=audio 0 RTP/AVP 97
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:97 L16/44100
a=fmtp:97
a=control:track2

[janus-ZM] [Sat Aug  6 21:25:51 2022] SETUP answer:RTSP/1.0 200 OK
CSeq: 3
Date: Sun, Aug 07 2022 03:25:49 GMT
Transport: RTP/AVP;unicast;destination=10.0.1.7;source=10.0.1.67;client_port=10060-10061;server_port=6978-6979
Session: C128E013;timeout=65

[janus-ZM] [Sat Aug  6 21:25:51 2022]   -- Source (video): 10.0.1.67
[janus-ZM] [Sat Aug  6 21:25:51 2022]   -- RTP port (video): 6978
[janus-ZM] [Sat Aug  6 21:25:51 2022]   -- RTCP port (video): 6979
[janus-ZM] [Sat Aug  6 21:25:51 2022]   -- RTSP session timeout (video): 32000 ms
[janus-ZM] [Sat Aug  6 21:25:51 2022] RTSP video latching: 10.0.1.67:6978
[janus-ZM] [Sat Aug  6 21:25:51 2022]   -- RTCP: 10.0.1.67:6979
[janus-ZM] [Sat Aug  6 21:25:51 2022] Sending PLAY request...
[janus-ZM] [Sat Aug  6 21:25:51 2022] PLAY answer:RTSP/1.0 200 OK
CSeq: 4
Date: Sun, Aug 07 2022 03:25:49 GMT
Range: npt=0.000-
Session: C128E013
RTP-Info: url=rtsp://10.0.1.67:8554/video2_unicast/track1;seq=27647;rtptime=2856732743,url=rtsp://10.0.1.67:8554/video2_unicast/track2;seq=0;rtptime=0
```

# MainStream
```
CSeq: 1
Date: Sun, Aug 07 2022 03:44:20 GMT
WWW-Authenticate: Digest realm="LIVE555 Streaming Media", nonce="2431aaa7fb9fa620e8665597fd90b059"

RTSP/1.0 200 OK
CSeq: 2
Date: Sun, Aug 07 2022 03:44:20 GMT
Content-Base: rtsp://10.0.1.67:8554/video1_unicast/
Content-Type: application/sdp
Content-Length: 563

v=0
o=- 1659831422316587 1 IN IP4 10.0.1.67
s=LIVE555 Streaming Media v2022.04.26
i=LIVE555 Streaming Media v2022.04.26
t=0 0
a=tool:LIVE555 Streaming Media v2022.04.26
a=type:broadcast
a=control:*
a=range:npt=now-
a=x-qt-text-nam:LIVE555 Streaming Media v2022.04.26
a=x-qt-text-inf:LIVE555 Streaming Media v2022.04.26
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:96 H264/90000
a=fmtp:96
a=x-dimensions:1920,1080
a=control:track1
m=audio 0 RTP/AVP 97
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:97 L16/16000
a=fmtp:97
a=control:track2

[janus-ZM] [Sat Aug  6 21:44:22 2022] [ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_parse_sdp:6349] [2] cannot parse video fmtp...
[janus-ZM] [Sat Aug  6 21:44:22 2022] [ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
```

Substream is calling for profile 4dxxxxx whichis Mainline when the spec calls for Baseline. The mainstream does not seem to call a profile at all which fails outright.

Does the RTSP implementation allow for configuring the H264 encoding profiles?

gtxaspec, 2022-08-07:
@baudneo unfortunately, we are at the mercy of the ingenic media platform's mercy regarding the stream implementation. 

see https://github.com/cgrrty/Ingenic-SDK-T31-1.1.1-20200508/tree/main/doc

you can run `impdbg --enc_info` to see the stream parameters

The program we use to forward the streams is `v4l2rtspserver` https://github.com/mpromonet/v4l2rtspserver, but I don't believe that changes anything in the stream itself.

baudneo, 2022-08-07:
Ah ok, I'll play around a bit and see if anything is possible. I think janus also allows to override the ftmp line. Hopefully the encoding conforms to whatever chrome/Firefox allow.

Seems chrome only accepts 1 type -> profile-level-id=42e01f

closed 2022-08-07 by baudneo