gtxaspec/wz_mini_hacks #223: Unable to run compile_image.sh

bjs-pdx, 2022-08-08:
After cloning the repository, I try to run compile_image.sh in the v2_install directory and I get the following:

`compile_image.sh 
wz_mini: this will download the latest firmware version from the vendor and compile a modified demo.bin.  Are you sure? [y/N]y
checking for tools
tools OK
create temporary work directory
check for local zip
local archive not found
downloading latest firmware
build for v2
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum failed check, please manually supply file
`
Didn't see any specific instructions other than to run the script to create the files needed to install on the V2.  What am I missing?

gtxaspec, 2022-08-08:
can you give me some specifics on what OS you are running?

bjs-pdx, 2022-08-08:
Ubuntu 18.04.

I think I found the solution.  Manually downloaded the latest firmware zip for the V2 and renamed it to demo.zip.  The script seems to have completed and generated a demo.bin.  Will try it tomorrow.  Seems the script isn't finding the firmware file on the Wyze servers.

`compile_image.sh 
wz_mini: this will download the latest firmware version from the vendor and compile a modified demo.bin.  Are you sure? [y/N]y
checking for tools
tools OK
create temporary work directory
check for local zip
local archive found
extracting firmware to temporary work directory
Archive:  v2_ro/demo.zip
  inflating: ./v2_ro/demo_4.9.8.1002.bin  
  inflating: ./v2_ro/__MACOSX/._demo_4.9.8.1002.bin  
create firmware directory
unpack stock firmware image
+ ACTION=unpack
+ KERNEL_OFFSET=64
+ ROOTFS_OFFSET=2097216
+ DRIVER_OFFSET=5570624
+ APPFS_OFFSET=6225984
+ [ unpack = unpack ]
+ DEMO_IN=v2_ro/demo_4.9.8.1002.bin
+ OUT_DIR=v2_ro/fw_dir
+ dd if=v2_ro/demo_4.9.8.1002.bin of=v2_ro/fw_dir/kernel.bin skip=64 count=2097152 bs=1
2097152+0 records in
2097152+0 records out
2097152 bytes (2.1 MB, 2.0 MiB) copied, 3.99414 s, 525 kB/s
+ md5sum v2_ro/fw_dir/kernel.bin
bee7951307dc53edf59599cd15294a88  v2_ro/fw_dir/kernel.bin
+ dd if=v2_ro/demo_4.9.8.1002.bin of=v2_ro/fw_dir/rootfs.bin skip=2097216 count=3473408 bs=1
3473408+0 records in
3473408+0 records out
3473408 bytes (3.5 MB, 3.3 MiB) copied, 6.54794 s, 530 kB/s
+ md5sum v2_ro/fw_dir/rootfs.bin
c4ac3804e50cae555a21d2afbc63f26a  v2_ro/fw_dir/rootfs.bin
+ dd if=v2_ro/demo_4.9.8.1002.bin of=v2_ro/fw_dir/driver.bin skip=5570624 count=655360 bs=1
655360+0 records in
655360+0 records out
655360 bytes (655 kB, 640 KiB) copied, 1.23216 s, 532 kB/s
+ md5sum v2_ro/fw_dir/driver.bin
e9e8f7b18e57845bf6835a9c4b5788f1  v2_ro/fw_dir/driver.bin
+ uname -s
+ [ Linux = Darwin ]
+ stat -c %s v2_ro/demo_4.9.8.1002.bin
+ IMAGE_END=11075648
+ dd if=v2_ro/demo_4.9.8.1002.bin of=v2_ro/fw_dir/appfs.bin skip=6225984 count=4849664 bs=1
4849664+0 records in
4849664+0 records out
4849664 bytes (4.8 MB, 4.6 MiB) copied, 9.18957 s, 528 kB/s
+ md5sum v2_ro/fw_dir/appfs.bin
50deec4b53dde03c3efe98535e2f5b16  v2_ro/fw_dir/appfs.bin
replace factory kernel
pack firmware image with new kernel
+ ACTION=pack
+ KERNEL_OFFSET=64
+ ROOTFS_OFFSET=2097216
+ DRIVER_OFFSET=5570624
+ APPFS_OFFSET=6225984
+ [ pack = unpack ]
+ [ pack = pack ]
+ TMP_DIR=v2_ro/fw_dir
+ DEMO_OUT=demo.bin
+ dd if=/dev/zero of=v2_ro/fw_dir/kernel.bin bs=1 count=1 seek=2097151
1+0 records in
1+0 records out
1 byte copied, 0.000155224 s, 6.4 kB/s
+ cat v2_ro/fw_dir/kernel.bin v2_ro/fw_dir/rootfs.bin v2_ro/fw_dir/driver.bin v2_ro/fw_dir/appfs.bin
+ mkimage -A MIPS -O linux -T firmware -C none -a 0 -e 0 -n jz_fw -d v2_ro/fw_dir/flash.bin demo.bin
Image Name:   jz_fw
Created:      Sun Aug  7 20:59:50 2022
Image Type:   MIPS Linux Firmware (uncompressed)
Data Size:    11075584 Bytes = 10816.00 KiB = 10.56 MiB
Load Address: 00000000
Entry Point:  00000000
remove temporary work directory
demo.bin ready.  Please copy demo.bin to your memory card
`

bjs-pdx, 2022-08-10:
Had to manually download the firmware zip file to the v2 install directory and name it demo.zip.  The script then executes.

closed 2022-08-10 by bjs-pdx

SteveAtSWS, 2022-08-18:
I got the new firmware to build, now the camera won't take the update.

In their official flashing instructions it says with the camera unplugged and the SD card with firmware in place, hold down the Setup button and plug in the power and continue holding until both yellow and blue lights are on at the same time (should take about 6 seconds). In my case, the light never has both LED's on and if I hold the button too long, it goes back into the stock setup "Ready to connect".

SteveAtSWS, 2022-08-18:
> 

Figured out the problem. I had the SD card formatted as exFAT instead of FAT32.