gtxaspec/wz_mini_hacks #225: Regression in local_dns

clee604, 2022-08-08:
It seems in the latest version there is a regression for the DNS flood issue that was fixed in previous version via the enable_local_dns feature.

After updating, the `c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com` dns flood is back, and every five seconds.

Applying the old fix, `touch /tmp/test_hosts;echo -e "127.0.0.1 localhost\n54.185.42.140 [c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com](http://c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com/)" > /tmp/test_hosts; mount --bind /tmp/test_hosts /etc/hosts` works to silence the dns calls.

I've only tested this on two V3 cameras so far: 
1. connected via ethernet using the CDC driver, with bonding enabled (wifi fallback). This camera has DNS flood.
2. connected via usb-direct, connected to linux host using bridge, with bonding disabled. This camera does not have DNS flood.

gtxaspec, 2022-08-08:
fixed, updated master, try now

capncybo, 2022-08-08:
Curious to know... was the implementation of the init.d system to quickly solve issues which are dependant on boot-order ?

gtxaspec, 2022-08-08:
not really, it is more of a getting ready for overlayfs, easier to manager than giant scripts

clee604, 2022-08-08:
thanks, upgraded and testing now. 

There any chance we could get the resolv.dnsmasq nameservers settable via configuration or something that can survive upgrades? I like to have my cameras point to my pihole so I can monitor the DNS calls.

gtxaspec, 2022-08-08:
Sure, I'll add that to the upgrade script.

gtxaspec, 2022-08-14:
backup added in latest master

clee604, 2022-08-15:
sorry for the delay, cameras been running for a few days and DNS looks like it is resolved.

clee604, 2022-08-18:
@gtxaspec actually, after the backup was added to the latest master, I don't see any DNS calls to my pihole which is unusual. Trying to do `nslookup google.com` results in `nslookup: bad address '192.168.x.xx # wlan0'` (it correctly points to my DNS pihole server's IP)

Is there a way to check if the DNS server is set properly? I suspect it is not even though the resolve.dnsmasq file does have my DNS servers there.

gtxaspec, 2022-08-18:
check with `cat /etc/resolv.conf`

gtxaspec, 2022-08-18:
if it's set to 127.0.0.1 that means the local dnsmasq server is being used.

clee604, 2022-08-18:
everything seems in order... 127.0.01 for resolv.conf and my DNS set in resolv.dnsmasq

Is `nslookup `supposed to work?

gtxaspec, 2022-08-18:
it does, check the syntax on the command you are using.

try:

`nslookup google.com 127.0.0.1`

clee604, 2022-08-18:
interesting, thanks. 
```
nslookup google.com 127.0.0.1
nslookup google.com 192.168.x.xx
```
both work fine, just not nslookup without manually specifying the dns server.

Seems everything is fine and working as intended.

closed 2022-08-18 by clee604

gtxaspec, 2022-08-18:
when nslookup fails, can you verify that /etc/resolv.conf ONLY has the 127.0.0.1 entry?

clee604, 2022-08-18:
```
[root@WCV3:~]# cat /etc/resolv.conf
nameserver 127.0.0.1
search domain.home # wlan0
nameserver 192.168.x.xx # wlan0
nameserver 192.168.x.xx # wlan0
```
My name servers come up here

gtxaspec, 2022-08-18:
i see, thats why it's failing, because there is more than one entry.  it looks like there may be a timing issue with the dnsmasq script. 

edit `/opt/wz_mini/etc/resolv.conf` and ensure there is only the first line, 
`nameserver 127.0.0.1`

then edit `/opt/wz_mini/etc/network.d/S09dnsmasq`

right under `        if [[ "$ENABLE_LOCAL_DNS" == "true" ]]; then`
add:
```
                echo "wait for resolv.conf creation from iCamera"
                sleep 5
```

so the entire section should look like:

```
        if [[ "$ENABLE_LOCAL_DNS" == "true" ]]; then
                echo "wait for resolv.conf creation from iCamera"
                sleep 5
                dnsmasq -C /opt/wz_mini/etc/dnsmasq.conf
                rm -f /tmp/resolv.conf
                cp /opt/wz_mini/etc/resolv.conf /tmp/resolv.conf
                echo "dnsmasq enabled"
        fi
```

then reboot  and try nslookup again

clee604, 2022-08-18:
works after the fix.
cat /etc/resolv.conf brings only the name server also

clee604, 2022-08-19:
are there any other implications to this other than nslookup not working? ie. should I update all of my cameras with this timing fix?

gtxaspec, 2022-08-19:
can you run this fix for a bit and see if it holds? still testing on my end.

clee604, 2022-08-19:
sure i'll leave it on one camera for now. It doesn't get rebooted often except for updates usually but i'll just test doing a nslookup after every reboot