gtxaspec/wz_mini_hacks #240: Fix of: Regression in local_dns #225 seems to have effect on...

capncybo, 2022-08-13:
I have been regularly using this project on three version 3 Wyze cameras & happily upgrading via: upgrade-run.sh for a while now.
But In parallel with this project, I've been running the: https://github.com/mrlt8/docker-wyze-bridge project on a pi400.
I've been using the other project mainly for its web-based streaming to support the HTTP(in-browser) viewing.
Until recently I could easily update & reboot the wz_mini_hack firmware & after the reboot... the 3 camera streams would resume as expected.  However (to the best of my knowledge) shortly after you implemented this fix... Regression in local_dns #225
A firmware upgrade or even a camera reboot will prevent the "docker-wyze-bridge project" streams from resuming normally.

It is quite possible that perhaps it's caused by something the other project has done. (as they have been updating versions also)
But I thought I'd ask if you have any idea(s) as to why, or how to resolve it?

gtxaspec, 2022-08-14:
no clue, we still operate the same way: boot the camera, wait for icamera, start rtsp server, network online.

capncybo, 2022-08-14:
Thank you for the response...
It turns out the issue(s) was actually caused by the OTHER project & upgrading it to v1.8.3
Closing now ;-) THX

closed 2022-08-15 by capncybo