gtxaspec/wz_mini_hacks #241: V2 installation proposed changes

gtxaspec, 2022-08-14:
current method:

1. Run compile script
2. script downloads firmware
3. extracts firmware
4. replace kernel
5. repack firmware
6. flash demo.bin
7. boot wz_mini

The downside to this, is the need to modify the onboard flash with a modified kernel.  

proposed method:

1. Flash special demo.bin containing kernel only (ONLY ONCE, EVER, PER v2 DEVICE), which will flash a new version of u-boot, able to load a kernel directly from the memory card.
2. flash stock firmware demo.bin ( we need this step since step 1 erases the onboard system, root, and driver partitions )
3. boot wz_mini the same way you boot it on the t31 devices, it will load the kernel from memory card.

The downside to the proposed method, is the risk of **permanently** bricking the camera if something goes wrong during the u-boot flash procedure. (unless you know how to remove the flash memory chip and manually flash it using a programmer)

I will do more testing to see if I can run into this situation.

gtxaspec, 2022-08-14:
I have successfully flashed four v2 devices using this method.  If anyone would like to test, let me know!

gtxaspec, 2022-08-14:
Here is the new provisional installation directions for T20 devices which does not require compiling the firmware:

Installation ( **for PanV1/V2 ONLY** )

1. Format memory card, FAT32, DOS Partition type
2. copy `demo_uboot.bin` to root of memory card, and make sure it is the **only** file on the memory card
3. rename `demo_uboot.bin` to `demo.bin` on the memory card root, again making sure it is the  **only** file on the memory card
4. insert memory card into device.  hold the setup button while inserting the power cable.  the LED should light up purple, then you can let go of the button ( standard firmware flashing procedure )
5. once the flashing is complete, the camera will reboot, and the LED will start rapidly flashing, once the rapid flashing stops, the camera will reboot. 
6. remove the memory card, insert it back into your pc, delete the demo.bin file on the memory card
7. download the stock firmware for the device.  follow the standard flashing procedures.
8. you may now copy the contents of SD_ROOT to the memory card and boot wz_mini

Bob-AOL, 2022-08-15:
Outstanding. I'm game. And I just figured out how to run the other script after getting Ubuntu loaded onto a stick and shaking off over ten years of Linux cobwebs! LOL At least I have the platform to try this now.

And my version of demo.bin actually worked..., albeit on one of three cams only. That does have me slightly concerned about those two cams. Multiple cards across all three cams only seem to work with one of the cameras. The other two take the latest OEM firmware and the latest (last) Wyze RTSP firmware without issue, but won't cooperate with wz_mini.

Bob-AOL, 2022-08-15:
Is 7-Zip and acceptable alternative to enabling special permissions on the Windows folder and getting Samba running right on the Ubuntu thumb drive -- just to unzip and transfer 2 files back and forth? The tar command line utility in Windows 10 is not cooperating, either.

Bob-AOL, 2022-08-15:
Too anxious to wait for an answer on the 7-Zip question. It works fine for the tasks and essentially makes the V2 process almost as easy as the V3 process for Windows users -- just a couple of extra steps and no Linux intimidation involved.

The process as described worked perfectly for me -- again, on one of three of my V2s. All three took the wipe down and the OEM recovery without a hitch. The same two problem children simply will not play nice with wz_mini no matter what I've tried so far. No logs generated from that process to analyze.

They did take the latest version of Wyze RTSP firmware with no problem..., which raises a question about flashing other firmware now since doing the wipe down and recovery. Any considerations there?

A couple of minor questions about the instruction list:

"once the camera has been successfully flashed with the stock firmware, you may now proceed to install wz_mini...."

"Install" had me confused enough to actually try flashing it to one of my trouble cams. Silly me. Perhaps 'prepare' or 'stage' wz_mini' would be a more accurate and less confusing instruction there? I also wondered if the original demo.bin file may somehow still be required.

Copying the contents of SD_ROOT to the memory card would include the T31 factory file and may be confusing to some who may wonder if both factory files are necessary. Are they? (I included both.)

There has also been a question of whether just copying the wz_mini folder is sufficient, or if the files themselves need to be copied individually. I believe just copying the folder contents actually gave me an error in the log when I tried it indicating the wz_mini folder itself could not be found and may have aborted the process as a result? (The memory is a little hazy after trying so many things with my two troublesome cams.)

All in all I personally think this is a great idea and applaud the effort, sir. Well done. And thank you!

gtxaspec, 2022-08-15:
both files can be included, only one is used depending on the device.

the folder `wz_mini` must be copied to the root, not the subfolders inside wz_mini.

after the stock firmware is flashed, the demo.bin files are no longer needed

Bob-AOL, 2022-08-18:
< both files can be included, only one is used depending on the device.>

Some browsing elsewhere seems to suggest that an additional one may be required for the problem V2s that provides what the base u-boot is looking for on the SD card for these particular devices -- if I'm sorta understanding this sorta correctly.

gtxaspec, 2022-08-20:
@Bob-AOL 

If you could try some things:

1. download the .860 firmware for your v2.  flash your v2 with it
2. verify via the mobile app that your camera _**is**_ running on the .860 firmware
3. if it is, proceed to download the .1002 firmware for your v2.  flash your v2 with it.
4. verify via the mobile app that you are on the .1002 firmware
5. if it is, proceed to follow the new instructions to flash and install wz_mini
6. report back if you can with results.

Bob-AOL, 2022-08-20:
The new instructions procedure is what I originally followed as detailed previously. Doing it all over again starting at .860 instead of .1002 made no difference. Rather pointless since the uboot_demo.bin wiped the flash for a clean .1002 anyway, didn't it? At least the uboot version generated the attached log file that may shed some light. The SD card is still not recognized to generate a wz_mini initramfs.log file after flashing the new .1002 image and the camera just boots to the onboard firmware normally -- which is progress in one respect since the endless loop seems to have been remedied. Just still no recognition of wz_mini..

[wz_mini_u_boot_initramfs.log](https://github.com/gtxaspec/wz_mini_hacks/files/9387779/wz_mini_u_boot_initramfs.log)

Bob-AOL, 2022-08-20:
So I've gone back through all available layers of V2 firmware to 2020 and tried the conventional installation approach with each one. Still no joy

The one cooperative V2 is still cruising along nicely.

Bob-AOL, 2022-08-20:
@gtxaspec

I'm out of bullets on the basic trial and error front.

gtxaspec, 2022-08-20:
@Bob-AOL thanks for the valuable info and logs.  Let me analyze and I'll get back to you.

gtxaspec, 2022-08-23:
@Bob-AOL I may have a solution, would you be willing to flash a test bootloader?  I won't say it's without risk of bricking the camera.  Let me know!

Bob-AOL, 2022-08-24:
If it's no less dangerous than the other 'enhanced' method, sure. Have you tried it with success?

I'm also trying to establish a Linux platform that allows me to do everything in the Linux environment -- thinking there may be some degree of sensitivity by the problem cam flash to how I'm compiling and transferring the .sh file for use with the conventional method.

gtxaspec, 2022-08-27:
@Bob-AOL stand by, still working on things behind the scenes

Bob-AOL, 2022-08-27:
I know you're spread pretty thin and development can take time. Was just curious if there was any progress. I'd be willing to send you one of the cams if it would be at all helpful (or practical).

Again, your efforts in all of this are greatly appreciated!

maxfield-allison, 2022-09-08:
can confirm this worked for me on two v2's and one pan v1. ultimately makes life 10 times easier in the long run. Would be wonderful to automate the process even more but I understand the limitations.

maxfield-allison, 2022-09-08:
just did another v2. I was going to try to combine steps 7-10 but i accidentally copied the factory_ZMC6tiIDQN file for v3 and decided to just continue on. but it did flash the stock firmware with the SD_ROOT contents in the card. so odds are its safe to condense at least a little.

maxfield-allison, 2022-09-08:
One minor issue. Looks like I cant open the rtsp streams from any of these now. tried VLC and motioneye.

compeek, 2022-09-11:
I tried this today on one of my V2s, and it seems to be working great! (I didn't even both trying to current official method first, so I can't compare to that.)

Bob-AOL, 2022-09-11:
Still only one for three here.

maxfield-allison, 2022-09-11:
Could it be the cards or maybe the PC you're using to write to the cards?
Maybe even the download is corrupted? Just spit balling.

Bob-AOL, 2022-09-11:
Thanks. Been there. Done that..., multiple times. It's all pretty well covered above (and elsewhere). Our host is still working on it as is also covered above.

Bob-AOL, 2022-09-23:
[gtxaspec](https://github.com/gtxaspec)

Just checking back in on this. I've re-visited everything again from scratch with the same results.

gtxaspec, 2022-09-23:
hey Bob, I'll have an update this weekend for ya!

Bob-AOL, 2022-09-23:
Cool. I'm anxious to kick the tires and take it for a spin.

Sadie33, 2022-10-13:
Success, I've taken the plunge and used this latest method to install wz_mini_hacks on my Wyze Cam 2.

I now have wz_mini_hacks operational on 7 Wyze Cam 3 and 1 Wyze Cam 2.

Thanks You!!!

Bob-AOL, 2022-10-23:
@gtxaspec -- hey Bob, I'll have an update this weekend for ya!

Any update on this?

gtxaspec, 2022-10-25:
@Bob-AOL sorry for the delay, still need to test some things before I can get you a test image.  My test setup is offline at the moment.  I haven't given up on it yet!

Bob-AOL, 2022-10-26:
Thanks for the encouraging response. There is certainly no shortage of cool developments on several other more relevant fronts to keep you busy as well. I understand. And now V3 Pro...

But if you're still game, I certainly am. It seems like a problem in _need_ of an answer..., just because.

maxfield-allison, 2022-10-26:
I've got some old v2's I can donate or use to assist troubleshooting

Bob-AOL, 2022-10-26:
Very generous of you.

Perhaps you could try the new method to see if it's successful on a couple? Otherwise I'd be interested in a couple just to double check my method(s) if nothing else. I flashed my first one with no problems. The second two are not recognizing wz-mini from the same SD card that works perfectly in the first V2. 

There are three distinctively different hardware platforms among the three cameras. Two seem to present a compatibility issue with wz-mini for whatever reason(s). It would be interesting to determine what the differences actually are if hardware is indeed causing the issue.

thedudehimself, 2023-01-22:
Ok, T20 V2's, have about 5 with 3x NIB, and I'm stuck. The v2 posted in the Readme.mdm didn't work for me, regardless of which firmware I've tried. I think that's the challenge: using the steps above uboot installed and it generates logs upon boot. I can't get past that step: loading either demo_4.9.9.1574 or demo_4.9.5.115 - modified or not - fails to get anywhere.

Help?

Bob-AOL, 2023-01-22:
Good luck. Lots of variables with the v2s. I'd stick with the 4.9.8.1002 firmware for starters. Revisit the SD card formatting and content. Basic 32GB Class10 FAT32 card and should not even contain volume information. Just the demo.bin file for flashing and then just the wz_mini folder and T-20 Factory file for booting from the card.

Linux or Windows?

thedudehimself, 2023-01-22:
Ubuntu Linux, using a Class10 16GB FAT32. I have 1 running on Wyze firmware only, and 1 clicking every 3 seconds after flashing with uboot: it's frustrated with me I guess.

Bob-AOL, 2023-01-22:
As you've undoubtedly read above, I achieved a whopping 33% success rate for my _weeks_ of effort. 

Everything worked perfectly according to the modified plan for the first one and it was was rock solid..., until I flashed it back to the basic RTSP firmware and sold it along with the other two. I haven't looked back since -- unless you count now. 

Thanks for reminding me. LOL

Good luck.

Bob-AOL, 2023-01-22:
I did recently run across a Reddit thread about the different flavors of card reader/format issues with v2s. It was primarily focused on how to successfully format large capacity cards for v3s to FAT32 by using an ancillary step with a v2 to make it all happen. The gist there was using a v2 card reader to format a SD card somehow properly prepared it for then being formatted as FAT32 -- which it evidently was _not_ during the formatting by the v2. I believe it was vFAT maybe? I'll see if I can find the thread. May not fix anything, but could maybe explain some stuff.  v2s are clearly not created equal.

thedudehimself, 2023-01-22:
Thanks. Current score: 

1x Wyze v2 that stated the :loading bootloader" thing, but never returned, following the OG steps. 
1x Wyze v2 that will load the uboot.bin firmware, but not much else. I'll keep trying to fully destroy it.

2x Wyze v3's that just work every time with the RTSP firmware.

Bob-AOL, 2023-01-22:
Here's the link. No real solid anything, just food for thought about what could be happening from v2 to v2 and their seeming inconsistent behavior with wz_mini.

https://www.reddit.com/r/wyzecam/comments/keykos/guide_to_getting_wyze_cam_v3_working_with_microsd/

thedudehimself, 2023-01-22:
Thanks. update: the 1x v2 that I used the OG steps on is now SSH enabled, but RTSP isn't working yet - current working theory is I _Blame VIM_.

Hard formatting a SD with the v3, going to try that in the stuck V2 with UBOOT firmware.

thedudehimself, 2023-01-22:
I can get an RTSP audio stream, but not video from the V2's. Bugger.

Bob-AOL, 2023-01-22:
VLC player?

thedudehimself, 2023-01-22:
Oh, _VIM_ got the best of me today. Last update: **4 of 5 v2's are streaming over RTSP** and ready for customization.

The one I tried this with: I don't know where to go from here. The uboot firmware will generate a new log file: ID'ing that it's currently installed and no other action to take. Any Wyze firmware bootloops.

Bob-AOL, 2023-01-22:
Congratulations!

Out of curiosity, what Wyze firmware is actually on the camera?

thedudehimself, 2023-01-26:
I kept the v2's at .1002 for now, that's the first I got to work. Minus my uboot-n-looping amigo, Clicky.

I'll try taking one of the v2's to the latest, like the v3's later this week. I found some micro-network-power-dongles in the electronics bin this am, going to complete this ridiculous project for good!

trevmlt, 2023-05-13:
> Here are the new provisional installation directions for T20 devices which does not require compiling the firmware:
> 
> ### **_EXPERIMENTAL, MAY DAMAGE YOUR DEVICE. PROCEED AT YOUR OWN RISK. report back results if possible!_**
> Installation ( **for PanV1/V2 ONLY** )
> 
> 0. **_This step only needs to be done ONCE, ever, for t20 based devices_**
> 1. Format memory card, FAT32, DOS Partition type
> 2. extract the file named `demo_uboot.bin.gz`, and then copy the file named `demo_uboot.bin` to root of memory card, and make sure it is the **only** file on the memory card
> 3. rename the file named `demo_uboot.bin` that you copied in the previous step, to `demo.bin` in the root of the memory card, again making sure it is the **only** file on the memory card
> 4. insert memory card into device.  hold the setup button while inserting the power cable.  the LED should light up purple, then you can let go of the button ( standard firmware flashing procedure )
> 5. once the flashing is complete, the camera will reboot, and the LED will start rapidly flashing, once the rapid flashing stops, the camera will reboot.
> 6. remove the memory card, insert it back into your computer, delete file named `demo.bin` on the memory card
> 7. download the stock firmware for the device.  follow the standard flashing procedures.
> 8. once the camera has been successfully flashed with the stock firmware, you may now proceed to install wz_mini
> 9. you may now copy the contents of SD_ROOT to the memory card
> 10. extract `factory_ZMC6tiIDQN.gz` and copy it to the root of the memory card
> 11. you may now boot wz_mini on the t20 device.
> 
> after flashing, you may check the logs located at `<memory card root>/wz_mini_u_boot_initramfs.log` for debug info about the flashing result.
> 
> [demo_uboot.bin.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9334360/demo_uboot.bin.gz)
> 
> [factory_ZMC6tiIDQN.gz](https://github.com/gtxaspec/wz_mini_hacks/files/9334365/factory_ZMC6tiIDQN.gz)
> 
> if you believe these directions can be simplified, please let me know!

Just wanted to say thank you for this! Used the above on 2 x V2 and worked perfectly on FW 4.9.9.1851. Thanks again!

xraive, 2023-06-04:
I tried this method on a v2 and a Pan v1 and both devices are just stuck in a boot loop.  Pleas see the details and log result below.

Firmware = demo_v2_rtsp_4.28.4.49 (RTSP Beta firmware)

wz_init.log result

```
welcome to wz_init.sh
PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

basename: applet not found
##########
Replace stock busybox
Replace stock fstab
Replace stock inittab
Replace /etc/profile for local/ssh shells
Replace system hostname
#####S02tmpfs#####
Mount wz_mini tmpfs
Create workspace directory
Mounting global tmpfs
#####S03busybox#####
Install busybox applets
#####S04model#####
WYZEC1-JZ detected
#####S05mount#####
mounting /system
Current Mounts:
rootfs on / type rootfs (rw)
devtmpfs on /dev type devtmpfs (rw,relatime,size=49676k,nr_inodes=12419,mode=755)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /opt type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mtdblock2 on / type squashfs (ro,relatime)
/dev/mmcblk0p1 on /bin/busybox type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/fstab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/inittab type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/profile type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p1 on /etc/hostname type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /opt/wz_mini/tmp type tmpfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime)
/dev/mtdblock4 on /system type jffs2 (rw,relatime)
#####S06car#####
Checking for CAR FW
CAR not detected
#####S07inject#####
Copy stock rcS
Add verbose debugging to rcS
Inject init.d scripts to rcS
Modify Global Paths in rcS
Copy factory app_init.sh
Replace factory app_init.sh path
Enable libcallback
set path for iCamera
#####S08passwd#####
Replace stock password
#####S09firstboot#####
First boot, initializing
Volume is 50
Audio Out GetPubAttr samplerate:8000
Audio Out GetPubAttr   bitwidth:16
[... 971 lines cut ...]
K01network: Network HW not ready yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: Network HW not ready yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
ifconfig: wlan0: error fetching interface information: Device not found
K01network: Network HW not ready yet, try again in 5 seconds
S09coredump: wpa_supplicant not initialized yet, try again in 5 seconds
umount: tmpfs busy - remounted read-only
umount: can't remount /dev/mmcblk0p1 read-only
umount: can't remount /dev/mmcblk0p1 read-only
umount: sysfs busy - remounted read-only
umount: devtmpfs busy - remounted read-only

The system is going down NOW!

Sent SIGTERM to all processes

Sent SIGKILL to all processes

Requesting system reboot

```

And it continues to loop.  Anyone have an idea how to resolve this issue?

Bob-AOL, 2023-06-04:
> Firmware = demo_v2_rtsp_4.28.4.49 (RTSP Beta firmware)

Using _stock_ firmware is recommended a couple of times. 

Try 4.9.8.1002 for a v2 and see how that goes. I don't know what the Pan v1 version might be.

xraive, 2023-06-05:
@Bob-AOL I tried that firmware as well with the same results.  Anyway we can revert this?

Bob-AOL, 2023-06-05:
Revert what?

languy99, 2023-09-14:
> @Bob-AOL I tried that firmware as well with the same results. Anyway we can revert this?

same thing for me, gets stuck in boot loop. you can hear the ir lens moving over and over.

rocket59, 2024-02-02:
Is this a valid install process? I'd like to do this without having to use linux.

Bob-AOL, 2024-02-03:
Following the 'proposed' experimental directions should still be valid given a cooperative v.2 hardware version. As you've likely read, they don't all necessarily play nice and there are a couple of variants.

If you have a cooperative v.2 you should be fine. Otherwise, you should still have a working v.2 even if it doesn't support the later method of using wz_mini which can be accomplished without using Linux.

rocket59, 2024-02-03:
Thanks. What's the later method? The wiki install instructions tell us to compile the firmware image which of course requires linux to run.

Bob-AOL, 2024-02-03:
This entire thread addresses the 'later' method consisting of the proposed changes explained in the third post as compared to the earlier 'compile' method. 

I was assuming that was the one you were asking about still being 'valid' since it does not involve Linux, though it may or may not be the 'latest' variant of the method, hence my referring to it as the 'later' version discussed here.

I'm unaware of yet another 'later' version, so I guess we can just go with 'latest' for the one dicussed here.

My bad.  LOL

rocket59, 2024-02-03:
No problem, thanks again!

Bob-AOL, 2024-02-04:
Good luck. If you're successful and intend to use the USB/POE/ETH functionality, the v.2s have no data capability on the micro USB connection as do the v.3s. The v.2 USB 2.0 slot has both and I used a micro USB to USB A adapter and the UCTRONICS POE adapter successfully with my particular effort.

rocket59, 2024-02-17:
Sadly no joy. I managed to flash the uboot and 1002 bin files successfully. But the mini hacks setup is stuck in the dreaded reboot loop. 

The camera will still boot up properly with no SD card fortunately

Bob-AOL, 2024-02-17:
Bummer. I feel your pain. I eventually gave up after a couple of weeks trying to figure out why my first try went without a hitch on one v2 but two others simply would not recognize or accept the same identical process.

While my suspicion was different hardware in the working camera than the other two, I never received any actual explanation from anyone indicating why that would matter. 

There are indeed at least two distinctly different hardware versions of the v2. Who knows what else may be involved as well. Most would not consider it a very worthwhile cause to explore..., obviously.

I simply reverted back to the Wyze RTSP firmware and eventually ended up selling all my v2s.

rocket59, 2024-02-17:
yea, not the end of the world, as the camera still works. I have one more v2 to try it out on later and  like you will replace them eventually with v3s. I already have a couple of them working great with mini hacks.

[3 more replies]