gtxaspec/wz_mini_hacks #250: v2 setup error

SteveAtSWS, 2022-08-17:
Doing a fresh install and I can't get the setup to run. Below is the output from running compile_image.sh

```
user@laptop:~/wz_mini_hacks/$ sudo ./v2_install/compile_image.sh 
wz_mini: this will download the latest firmware version from the vendor and compile a modified demo.bin.  Are you sure? [y/N]y
checking for tools
tools OK
create temporary work directory
check for local zip
local archive not found
downloading latest firmware
build for v2
wget: missing URL
Usage: wget [OPTION]... [URL]...

Try `wget --help' for more options.
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum: ./v2_ro/demo.zip: No such file or directory
md5sum failed check, please manually supply file
```

Any ideas what I did wrong?

virmaior, 2022-08-18:
I don't have a v2 but your issue looks identical to https://github.com/gtxaspec/wz_mini_hacks/issues/223
try manually downloading to supply the file.

Bob-AOL, 2022-08-18:
I had no problem using the most recent release. You may also want to consider #241 as an alternative solution. It avoids that issue altogether. Read the instructions and disclaimer carefully, then go for it. 

The cam will then boot from the memory card with full OEM functionality like the V3s and can revert to factory firmware by simply removing the SD card. 

The flash process worked flawlessly for three of my V2s, though only one would boot from the memory card afterward. It's a known issue with certain versions of the V2, evidently including the new black one. You can still resort to the OEM RTSP firmware if desired until a fix is found for the other issue.

Bob-AOL, 2022-08-19:
The problem persists with two of my three V2s not loading wz_mini. After quite a bit of reading, it would seem that u-boot wants to see a different version of the factory file for these particular cameras to even recognize the SD card, let alone boot from it. 

I've used both the traditional installation process and the new 'enhancement' version without success on these two cameras. Both approaches work flawlessly on the third camera using both the same and different SD cards and sets of files tested across all three cameras.

Since the enhanced process for establishing the latest Wyze firmware essentially wipes the flash memory prior to re-loading the firmware, could the issue be one of proper hardware identification in the factory file or u-boot wanting to see a different factory file to properly recognize the hardware in these problem cameras?

The frustration mounts...., LOL.

gtxaspec, 2022-08-19:
@Bob-AOL with the new install method, there should be log files on the root of the SD card, can you check?

Bob-AOL, 2022-08-19:
Will do. But so far, that's been the Catch-22. The SD card doesn't seem to even be recognized. 

And I'm certainly open to anything you suggest that's within my capability to investigate. Even dusted off my nano chops a little bit the other day. LOL

'Preciate the help.

Bob-AOL, 2022-08-19:
A swap.gz appears in the wz_mini folder. No log files in root. 

wz_mini.conf is the default with no user added settings, btw.

bjs-pdx, 2022-08-19:
If I remember my sd card, check /opt/wz_mini/log.  Can't remember what was there but there is a log file on mine.

Bob-AOL, 2022-08-19:
No such file. The log folder just has the empty .gitignore file. Here's a non-working version.

[image]

[image]

This is a functioning version.

[image]

[image]

All log files populated as probably expected.

Camera MAC addresses all share the same prefixes up to a point except the cam that takes wz_mini. 2C:AA:8E;DF:xx:xx for the problem children and 2C:AA:8E;_24_:xx:xx for the one that boots from the card.

Distinctly different hardware between the two MAC address ranges with the one on the right being the working one, but also the one that still used a connector on the board for the antenna -- presumably the older of the three for that reason alone.

[image]

bjs-pdx, 2022-08-19:
I don't see any log files on either listing...did you check /wz_mini/log/?

Bob-AOL, 2022-08-19:
As mentioned, there are log files as one would expect where they should probably be expected in the second two pics which are from the working v2.

Zero log files anywhere in the first two pics from a SD card boot attempt resulting in basically non-recognition of the SD card at all in the non-cooperative cameras. The same card will boot right up on the working camera. The card from the working camera  does nothing on the non-cooperating cameras. It would seem obvious that the wz_mini SD card and the troublesome cameras are simply not communicating properly -- if at all. The lack of any record of it would seem to support that theory.

I'm putting my money on a 2nd factory file being required for whatever the hardware differences are between these two versions of V2s

Bob-AOL, 2022-08-19:
BTW, I'm doing this bone clean with wifi only.

Bob-AOL, 2022-08-19:
Pretty good odds the answer lies somewhere in this thread...

https://www.reddit.com/r/wyzecam/comments/jyc9jo/does_the_new_version_of_the_wyze_cam_v2_hardware/

bjs-pdx, 2022-08-20:
Not sure exactly what's going on there but the post is 2yrs old.  I'll see if I have something in that range, 2020, to test.

gtxaspec, 2022-08-20:
does anyone have the ability to connect one of the non functioning camera's uart (serial port) to a  serial port on a pc or raspberry pi to read the debug output?

Bob-AOL, 2022-08-20:
The age of the post is sorta the point -- especially given the fact that the general discussion is about the impact of V2 hardware changes on the Xiaomi hack. Looks like lots of pertinent info there which could shed some light on the wz_mini's apparent inability to run on certain V2 hardware as well.

Will try .860 firmware this morning as has been suggested by our host., The only other firmware listed is two 2020 versions. The rest are presumably unavailable for download due to the recently exposed security issues.

bjs-pdx, 2022-08-20:
I'm slightly sceptical but open to the idea of a hardware change.  However, I'd like to point out that later versions of the V2 don't seem to have any problems.  So, did Wyze change the hardware again?

Bob-AOL, 2022-08-20:
If it's not a hardware issue, it's firmware issue. Guess that depends on whether you're the developer or the designer. 

One thing is absolutely certain..., it's a _compatibility_ issue between the two.

endertable, 2022-08-21:
@Bob-AOL If you do the following command on the "bad" cam:

dd if=/dev/mtd0 of=/tmp/mtd0

What is the size of the file /tmp/mtd0?

Also, what is the md5um (md5sum /tmp/mtd0)

gtxaspec, 2022-08-21:
@endertable the md5sum is in the u_boot log he posted

endertable, 2022-08-21:
What i noticed is that the md5sum of applied image on /dev/mtd0 was different from file.  Was curious if maybe the size of his boot partition was different?

Bob-AOL, 2022-08-21:
How would I access the directory structure of a stock firmware cam to check?

Bob-AOL, 2022-08-21:
I presume this is where the magic happens? More than likely RX, TX, and V for a serial connection from what I've learned from a great YT video on the subject. Not exactly sire I want to jump into it, but it's pretty cool to see how this stuff actually evolves.

[image]

Also some board info for what it may be worth. This particular one has the V3 style antenna which means I possibly have a third version of hardware -- or at least three different layouts. Haven't popped the backs off the other ones to see version numbers. But considering how long these have been around, I'd say 2020 is probably pushing the end of production for the white ones and v2.05 would be considered among the last versions. It would be interesting to see what the black ones are.

[image]

endertable, 2022-08-21:
You can use a RPi to access and the pin leads in order of the picture are:  GND, TX, RX

Looking at your log file, it "seems" you don't have the new uboot on the cam.  It looks like it runs into md5sum errors and restores your original one so if that is the case, viewing the startup won't do you any good at this point unless you force the cam to keep the new uboot irregardless of errors.  
Then again, I could be reading the log file wrong, hehe.

Bob-AOL, 2022-08-21:
I was under the impression the script wiped the flash and there no longer is (was?) an original uboot to revert to?

gtxaspec, 2022-08-22:
the process is as follows:

backup the stock uboot
verify the uboot file
if it passes, flash it to the flashrom
once flashed, verify it
if it fails, restore the stock uboot

interestingly, on @Bob-AOL's camera, once flashed the bootloader does not pass verification.  I didn't experience this on any of my v2 hardware.

Bob-AOL, 2022-08-22:
At least we now know that much.

Any possible areas that I could explore further on my end?  About the only thing I can think of is trying a different method of un-compressing the new files that may be more user friendly to the problem cams. I used 7-Zip on Windows which obviously worked ok for the one cam. Could the hardware on the other cams simply be less tolerant of minor irregularities in the un-zipping process than the one that took the files?

AllenBiehle, 2023-01-08:
@Bob-AOL did you ever find a way to get your problematic v2 to support the process? I seem to have a similar issue with one of my v2 devices, and was wondering if I should just throw in the towel on it. 🤔

Bob-AOL, 2023-01-08:
I ultimately abandoned the effort, re-flashed the Wyze RTSP firmware, sold all three and haven't looked back.