gtxaspec/wz_mini_hacks #251: Unsure how web server image capture works

cbass27, 2022-08-18:
I enabled the web server and click the Current Screen link to view/download the current image capture but my browser asks me to download the jpeg.cgi file. Am I doing something wrong or is it not supported in Microsoft Edge browser?

Thanks!

virmaior, 2022-08-18:
hmm.. I use brave, chrome, and safari and it hasn't asked me to download the file rather than see it in browser. Edge is also based on chromium/webkit so it should behave the same way ...

Two possibilities come to mind:
1. There's a bug for v2 cameras that's not related to the www code which breaks the image capture (https://github.com/gtxaspec/wz_mini_hacks/issues/170)
2.  in an earlier version, I had accidentally left some test code in that borked it for some uses (It didn't appear to affect browsers but broke it when using mod_proxy).

Do either of these seem like they might be the issue?

cbass27, 2022-08-19:
It could be the v2 camera bug. Any idea how to fix this or should I just wait for a fix in a newer release? Thank you!

gtxaspec, 2022-08-19:
still researching v2 compatibility, stay tuned.

cbass27, 2022-08-20:
@gtxaspec I shall. Thank you!

gtxaspec, 2022-08-20:
converting to discussion for visibility