gtxaspec/wz_mini_hacks #256: wz_mini won't install on older V2

bjs-pdx, 2022-08-19:
I have an older V2 camera (2019) and tried to install wz_mini on it today.  Ran the firmware installation process, heard the announcement regarding first boot and cache expansion then...nothing.  Light went out on camera and it never connected to the wifi.  Looking at the logs, it appears that the wifi NIC doesn't initialize nor connect to anything.  Attaching the logs.

[.wz_mini_initramfs.log](https://github.com/gtxaspec/wz_mini_hacks/files/9378944/default.wz_mini_initramfs.log)

[wz_init.log](https://github.com/gtxaspec/wz_mini_hacks/files/9378947/wz_init.log)

gtxaspec, 2022-08-19:
did you manually supply the firmware, or auto download with the compile script?

i can see:

`cp: can't stat '/system/init/app_init.sh': No such file or directory`
the /system partition didn't mount for some reason.

bjs-pdx, 2022-08-19:
Not sure what you are asking.  This is the same demo.bin used on another, newer V2 which worked fine.  I had to download the zip file manually from Wyze, rename it demo.zip and then run the compile.  The auto download isn't working, or at least it wasn't when I tried to run it.

gtxaspec, 2022-08-19:
@bjs-pdx I just added wz_mini to add additional logging information, can you download the latest version, and try again?

bjs-pdx, 2022-08-19:
sure...it will be tomorrow.

bjs-pdx, 2022-08-19:
Well, ran the updated version this morning and installed without any problem.  My suspicion would be an issue with the files on the SD card.  I formatted the card to FAT32 then copied all the new files to it.  I put the card in the camera, ran the firmware update process and the wz_mini files loaded.  Can't think of anything else.

bjs-pdx, 2022-08-19:
I'm going to close this since the camera appears to be working correctly now.  I'm going to put it down to user (me) error.

closed 2022-08-19 by bjs-pdx