gtxaspec/wz_mini_hacks #268: Failed to start RTSP streams on Pan V2 - missed libsetunbuf.so

homemain, 2022-08-21:
First of all - thanks a lot for this project! That does exactly what many DIY people would expect from the camera software!

I have an issue with RTSP stream, that you could probably help me with.
During RTSP startup I receive following message in logs and RTSP server doesn't start

```
root@WCV3:14]# cat /media/mmc/wz_mini/log/v4l2rtspserver.log

/opt/wz_mini/bin/v4l2rtspserver: can't load library '/system/lib/libsetunbuf.so'
```

My /system/lib folder looks like this, there is really no libsetunbuf.so
```
[root@WCV3:14]# ls /system/lib/
PHY_REG_PG.txt            libalog.so                libimp.so                 liblogserver.so           librtmp_push.so.0.0.1     libzbar.so.0.0.99
firmware                  libaudioProcess.so        libjzdl.so                libmp4rw.so               libsCHL.so
libAVAPIs.so              libcproducer.so           liblocalsdk.so            libperson_tracker_inf.so  libsysutils.so
libIOTCAPIs.so            libhevc2mp4.so            liblocalsdk_motor.so      librtmp_push.so           libzbar.so
libP2PTunnelAPIs.so       libiaac.so                liblocalsdk_watchdog.so   librtmp_push.so.0         libzbar.so.0
```

I  have Wyze Cam Pan V2 with 4.49.1.20 firmware

Web server works ok, I can see picture from the cam. Also attaching diagnostic logs
[Diagnostics T31(HL_PAN2) on running wz_mini Sat Aug 13 05_23_12 PM PDT 2022 as WCV3.pdf](https://github.com/gtxaspec/wz_mini_hacks/files/9388835/Diagnostics.T31.HL_PAN2.on.running.wz_mini.Sat.Aug.13.05_23_12.PM.PDT.2022.as.WCV3.pdf)

gtxaspec, 2022-08-21:
your firmware is too old.  try upgrading to a newer firmware

homemain, 2022-08-22:
Yeah. Upgrade from 4.49.1.20 to 4.49.1.47 solves this issue as well! so it makes a difference :)

closed 2022-08-22 by homemain