gtxaspec/wz_mini_hacks #269: FAT file system on SD Card gets corrupted

homemain, 2022-08-21:
Almost immediately after the start of Wyze Cam Pan V2 with wz_mini_hacks SD card gets into read-only mode and there are following messages in the logs

```
[   33.643405] FAT-fs (mmcblk0p1): error, fat_get_cluster: invalid cluster chain (i_pos 2793505)
[   33.675478] FAT-fs (mmcblk0p1): Filesystem has been set read-only
[   33.712380] FAT-fs (mmcblk0p1): error, clusters badly computed (1 != 2)
[   33.754379] FAT-fs (mmcblk0p1): error, fat_bmap_cluster: request beyond EOF (i_pos 2793505)
[   33.783699] FAT-fs (mmcblk0p1): error, clusters badly computed (2 != 3)
[   56.282274] FAT-fs (mmcblk0p1): error, clusters badly computed (3 != 4)
[  140.382266] FAT-fs (mmcblk0p1): error, clusters badly computed (4 != 5)
...
```
I run fsck.vfat and problem seems to be fixed, but only until next reboot

```
[root@WCV3:14]# fsck.vfat  /dev/mmcblk0p1 
fsck.fat 4.2 (2021-01-31)
/WZ_MINI/LOG/dmesg.log
  Contains a free cluster (7026). Assuming EOF.
/WZ_MINI/LOG/dmesg.log
  File size is 12023 bytes, cluster chain length is 0 bytes.
  Truncating file to 0 bytes.
Free cluster summary wrong (936742 vs. really 936197)
1) Correct
2) Don't correct
[12?q]? 1

*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/mmcblk0p1: 512 files, 7675/943872 clusters
[root@WCV3:14]# 
[root@WCV3:14]# fsck.vfat  /dev/mmcblk0p1 
fsck.fat 4.2 (2021-01-31)
/dev/mmcblk0p1: 512 files, 7675/943872 clusters

```

Windows check disk tool also sees this errors, fix them, but they appears again after card is inserted into the cam. 
SD Card it self is healthy. I plugged it into the my torrents server for a day and after random r/w operation during the day no errors on file system was found. 

I have Wyze Cam Pan V2 with 4.49.1.20 firmware

Full log attached
[Diagnostics T31(HL_PAN2) on running wz_mini Sat Aug 13 05_23_12 PM PDT 2022 as WCV3.pdf](https://github.com/gtxaspec/wz_mini_hacks/files/9388847/Diagnostics.T31.HL_PAN2.on.running.wz_mini.Sat.Aug.13.05_23_12.PM.PDT.2022.as.WCV3.pdf)

gtxaspec, 2022-08-21:
two things:

your firmware is too old, try upgrading to a newer one ( yes it makes a difference =D )
do you have another memory card to test?

homemain, 2022-08-22:
Yeah. Upgrade from 4.49.1.**20** to 4.49.1.**47** solves the problem! so it makes a difference :)

closed 2022-08-22 by homemain