gtxaspec/wz_mini_hacks #278: WIFI driver roaming in wpa_cli - 'FAIL'

claudobahn, 2022-08-24:
@gtxaspec my wifi environment has multiple stations serving the same SSID, allowing clients to roam. I noticed that since switching to the newer wifi drivers a while back, triggering roaming using `wpa_cli` no longer seems to be possible, all I get is a `FAIL`. Before the new drivers, issuing a `roam <BSSID>` in `wpa_cli` would sort of work, though not always associate successfully.

Wondering if that's a driver feature that's turned off right now?

fyi, in this case talking about the 8189fs driver.

Also tried the following w/ iwconfig, which presumably fails for the same reason (not sure)?
```
iwconfig wlan0 ap any
Error for wireless request "Set AP Address" (8B14) :
    SET failed on device wlan0 ; Operation not permitted.
```

claudobahn, 2022-08-26:
Interestingly when I use purely command-line, the following sequence currently mostly works for me:
* `wpa_cli -i wlan0 scan`
* `wpa_cli -i wlan0 scan_results`
* `wpa_cli -i wlan0 roam <BSSID>`

Sometimes says `OK` and ends up re-associating with a completely different BSSID, but at least it doesn't outright fail from the get-go.

Still curious to find out if you have any insight into the general roaming behavior of these cameras, since mine often get stuck on a poorly performing BSSID / channel when a much better option is available.

gtxaspec, 2022-08-27:
i'll take a look at the driver

gtxaspec, 2022-09-02:
@claudobahn can you enable debugging output (modify the file `/opt/wz_mini/etc/init.d/S10kmod` and set the log level: `rtw_drv_log_level=5`, check the section corresponding to your camera) and see if there is any output in dmesg when you try to do the wpa_cli that fails?

thanks

claudobahn, 2022-09-02:
I've done this: `mv /opt/wz_mini/bin/iwlist /opt/wz_mini/bin/iwlist.disabled` and will see if the cameras behave better. Though  "better" will be difficult to quantify :)

claudobahn, 2022-09-02:
Hm, never mind, according to your comment, that only impacts the V2. And even thought running `iwlist wlan0 scan` outputs all the networks in the area, `wpa_cli -i wlan0 scan_results` only shows the currently connected AP afterwards. Until I run `wpa_cli -i wlan0 scan`, then doing `wpa_cli -i wlan0 scan_results` does show other access points / networks.