gtxaspec/wz_mini_hacks #279: Cameras not connecting through Wyze app outside LAN

MomosX, 2022-08-24:
First of all let me thank you for this great hack.  Since i found it i converted 9 Wyze V3 to wyze_mini_hacks, and moved all to Ethernet connections which i must say is one of the best features:)

Since some weeks ago my cameras do not connect through wyze app when i'm outside the LAN.  I have the latest 4.36.9.139 FW, wyze_mini_hacks working prefectly. I am pretty sure this used to work a month ago, but not 100% sure ...

No idea how to troubleshoot this issue so if you have any ideas please help:)

Also, as a side note, the bonding of network interfaces does not seem to function properly.  When bonding is enabled, the Ethernet gets a 10 Mbps connection.  If i disable bonding and reboot it gets a 100 Mbps full-duplex connection.

Thank you very much.

bjs-pdx, 2022-08-25:
Pretty sure that's an issue with VPN or another setting on your phone.  If you check the Wyze forum and look for that topic I think you'll find an answer.  It happens without wz_mini.

MomosX, 2022-08-25:
Thank you for the answer. Unfortunatelly that's not it.  I double and triple checked and there is no VPN active.  I even deleted all the VPN profiles i had just to make sure.  All the cameras get stuck at connecting camera (1/3) and then error 27 or error 20010 which would suggest the camera can not connect to the internet.

bjs-pdx, 2022-08-25:
Are you running android?  Did you take a look at the last link I posted?  IP V6 can cause problems too. The issue isn't with the cameras it is with the phone.  If the cameras were unable to connect to the internet they wouldn't boot and you couldn't access them on the local network since Wyze servers are involved either way.

MomosX, 2022-08-25:
Yes i looked and checked - i do not have any IPv6 on my android, all APN are set to IPv4 only.  I'm just trying to get to the bottom of this so i figured a question would not hurt. 

What i did to further troubleshoot the problem:  i inserted in my android phone a second SIM card from another carrier.  Left voice on the first one but switched data to the new (second SIM) - and it works !!

So the conclusion would be that the first mobile carrier is somehow blocking the access to the cams , right ? 
APN's for both SIMS are set to IPv4 only.

endertable, 2022-08-26:
So if yo pull SD card out of a camera and run the stock firmware, can you connect to the cam from "the outside".  Easiest way to  test is by turning wifi off on your phone so it uses cellular service.

MomosX, 2022-08-26:
I'll close this as i'm pretty sure after even more testing that there is an issue with the mobile carrier.  Thank you :)

closed 2022-08-26 by MomosX