gtxaspec/wz_mini_hacks #286: [Question] Automatic IR Day/Night

Jonathan-Garber, 2022-08-30:
Is there any on board software that handles automatic day/night IR LED control?

claudobahn, 2022-08-30:
If you use SSH to log on to the camera, type `cmd` to get help or `cmd irled on` or `cmd irled off` specifically to turn them on/off.

For automatic control, the Wyze supplied app/firmware handles that. So go into the Wyze app and set up (or turn off) IR auto and which LED (near/far) to use.

kohrar, 2022-10-05:
For automatic control, see #309 and look at the `night_` options that you can set in `/configs/.user_config`. `iCamera` will automatically control the IR lights based on those settings.

baudev, 2022-10-22:
Indeed, it is important to understand that this hack runs in parallel with the Wyze firmware.

If we activate the RTSP stream of the hack without "turning on" the camera in the Wyze application (or using `ENABLE_SELFHOST_MODE=true`), then there will be no automatic change of the IR leds. In use it is quite surprising but technically understandable.

Maybe you should close the issue since @Jonathan-Garber haven't answered since August..?

closed 2022-10-22 by Jonathan-Garber