gtxaspec/wz_mini_hacks #287: Cameras freezing

MomosX, 2022-08-30:
Hi,

I have a bunch of Wyze V3, all on the latest FW and with wizi_mini_hacks installed.   Most of them are hardwired using Ugreen Asix Eth 2.0 adaptors, the rest are on wifi.

For 2 weeks i have been trying to find out why some of the cameras simply stop working after some time (a few hours to 24h).  Both hardwired and on wifi.  I have changed the SD cards, the Ugreen adaptors, changed Eth ports on the router also.  Nothign seems to work. Reboot does not help, SSH not active anymore, the only way to get them working again is by removing the SD card and rebooting.  after i get them going again, make a new SD card, same happens again.

I have enabled syslog maybe someone can figure out what the problem is.

[dmesg.log](https://github.com/gtxaspec/wz_mini_hacks/files/9449602/dmesg.log)
[syslog.log](https://github.com/gtxaspec/wz_mini_hacks/files/9449603/syslog.log)
[v4l2rtspserver.log](https://github.com/gtxaspec/wz_mini_hacks/files/9449604/v4l2rtspserver.log)
[wz_init.log](https://github.com/gtxaspec/wz_mini_hacks/files/9449605/wz_init.log)

I do understand much from these logs, so hopefully someone can help out :)

thank you !!!

C2BB, 2022-08-30:
This was happening to me as well on my V3. I need more time to verify (only 24 hours in at the moment) but for me the "fix" was to turn off the native Wyze recording to SD. When I had that enabled and the wz_mini_hacks setting ENABLE_MP4_WRITE set to true my SD card filled up quickly leaving it in a state where wz_mini_hacks could no longer function (and hence why in your case things work again with SD card removed thus wz_mini_hacks is removed). May not be your exact issue but based on your description it sounds a lot like my experience when the SD card gets filled by "something"

C2BB, 2022-08-30:
I'm running into the exact same problem with my V3 (wifi)

C2BB, 2022-08-30:
Looking through your logs I see that FPS configuration apply fails (and I assume your config has it set at 20 based on the error message):

`#####S18fps#####
Seting system FPS if greater than default
sh: 20: unknown operand`

I also noticed that in the web config editor that if an FPS value is set, it cannot be unset after...perhaps the two are correlated. 

My camera just started acting up again as you've described so this time instead of re-writing the SD card I modified the config file directly on the SD card, setting the FPS back to default (empty) and I also set the FSCK option to true. Put the SD back into my camera and it booted up as expected. I'll report back if the issue returns.

MomosX, 2022-08-30:
I have not set any FPS in config file, as i understood that default camera FPS will be used if left blank. However i set Night_
Drop_Disable = true.  But the cameras fail in daytime also.
FSCK is on.
1 hour ago 2 more cameras became unresponsive, no SSH, nothing.  Cutting the power does not help.

C2BB, 2022-09-01:
Are you trying to use this with Home Assistant / Frigate by chance?

MomosX, 2022-09-01:
I am using them with Frigate and a Coral TPU yes.  I have 8 Wyze v3 working perfectly, and 4 giving me a very hard time.... My next step will be to change the power supply for those 4 to see what happens as i am running out of things to change.

C2BB, 2022-09-01:
Can you share your Frigate config?...Over the last few days of triaging this I have narrowed down my issues which were exactly as you've described yours and it ended up being my Frigate configuration (was trying to RTMP stream at 1080P which as noted in the docs isnt going to work well). I have Frigate running smoothly now with V2 and V3 both a mix of official RTSP firmware, and some latest firmware using the wz_mini_hack.

MomosX, 2022-09-01:
`  TerasaSpate: 
    ffmpeg: 
      inputs:
        - path: rtsp://admin:spectra@192.168.X.XXX:8554/unicast
          roles:
            - record  
            - detect 
            - rtmp                            
    detect:
      width: 1920
      height: 1080
      fps: 5
    objects: 
       track:
         - person
         - cat
         - dog
    record:
      enabled: True
      events:
        objects:
          - person
          - cat
          - dog
        retain:
          default: 2
          mode: motion  `

Nothing fancy, standard config. All are pretty much the same, some having some masks also

C2BB, 2022-09-01:
Yep, thats the gist of what I had when I was running into the same issues as you. After carefully going back through the Frigate docs I noticed a key point in the "Stream Viewing" section per HA: https://docs.frigate.video/guides/camera_setup . Then looking at the camera docs I figured out how to split the streams up. Note you will need to enable both streams in your wz_mini_hack config. Here is my firgate config for reference, using Frigate Beta:

> ui:
>   use_experimental: true
> mqtt:
>   host: core-mosquitto
>   user: homeassistant
>   password: upee5aimah9soh4Thie9OhHahhieba5Kai7chai6shishoJaefiukie7tat6leci
> detectors:
>   cpu1:
>     type: cpu
>     num_threads: 2
>   cpu2:
>     type: cpu
>     num_threads: 2
> ffmpeg:
>   hwaccel_args:
>     - -hwaccel
>     - vaapi
>     - -hwaccel_device
>     - /dev/dri/renderD128
>     - -hwaccel_output_format
>     - yuv420p
> cameras:
>   front:
>     ffmpeg:
>       inputs:
>         - path: 'rtsp://192.168.2.70:8554/video2_unicast'
>           roles:
>             - rtmp
>             - detect
>         - path: 'rtsp://192.168.2.70:8554/video1_unicast'
>           roles:
>             - record
>     detect:
>       width: 640
>       height: 360
>     rtmp:
>       enabled: true

C2BB, 2022-09-02:
Hope that solves it for you! One more link I came across where the author of Frigate made comment specific to lower cost cams such as Wyze and issues running many cams and/or HD streams and detection: https://www.reddit.com/r/homeassistant/comments/runlac/wyze_cam_v3_and_frigate_getting_errors/

baudneo, 2022-09-03:
I am having this issue as well. The weird thing is that the rtsp wz mini stream Conti ues to broadcast and send/receive. The wyze app only gets to 2/3 authenticating but does not connect. If I ssh into the camera and reboot then things start working as they should.

Usually how I find this issue is happening again is in the morning. The camera keeps its or filter engaged which keeps the image b&w.

I also have night drop disabled so the fps stays the same day/night. I have fps set to 15 in the .conf file. This only happens on wyzev3, v2 keeps chugging along nicely.

Pics for reference. I use zoneminder for an NVR system.

[image]
[image]

C2BB, 2022-09-04:
Can't speak to zoneminder. Try testing with only the low definition feed. For wyze App issues, clear the app cache, solves many connection issues.

MomosX, 2022-09-04:
Ok reporting back. 48h and not a single drop/freeze in any of my Wyze cameras.  They work flawlessly no matter what resolution i set the stream/streams on. for testing purposes i left some with only main stream and some with dual stream.  Performance wise i can say i'm more happy with the ones having a single stream at max resolution.

What i did was change the power supplies on all those that were giving me a hard time.

C2BB, 2022-09-05:
Great news! That's the first I've heard of the stock power supplies causing stability issues. Curious, did you replace with the same amperage power supply or go with something bigger?

MomosX, 2022-09-05:
Well they were stock and not so stock. I've bought them from Amazon US and i'm in Europe, so they came with those 2 pins that do not fit into power outlets here. So i opened them and cut those 2 pins to solder some wires as i had to mount them into junction boxes anyhow.  Guess i didn't do such a good job on a few of them as they did not work as expected. I replaced them with some small modular USB power outlets of 5V, 2A.

closed 2022-09-05 by MomosX