gtxaspec/wz_mini_hacks #288: Editing wz_mini.conf on Microsoft Windows breaks the configuration entirely

KevinKadow, 2022-08-31:
It isn't currently possible to create/modify a working SD card from a Windows host, due to artifacts from editing "wz_mini.conf".

The errors after editing the configuration from Windows are due to DOS linefeeds added to the end of each line in wz_mini.conf when modified using common Windows apps (notepad, etc).   

One way to prevent this would be to modify the scripts which "source" the conf file (especially etc/rc.common) to instead  use dos2unix to create a clean copy of the configuration file in a temporary location at first use, and then source that cleaned copy instead of the original.

MomosX, 2022-08-31:
I do it on a win11 machine without any problems.  Just use Notepadd++ to edit the files.

maxfield-allison, 2022-09-08:
it is currently possible if you use a text editor that doesnt save encoding info. notepad++ or sublime are great options

Bob-AOL, 2022-10-16:
Notepad works just fine.

gtxaspec, 2023-09-26:
support for line endings from editing in windows has been added:
https://github.com/gtxaspec/wz_mini_hacks/commit/930cbe2e9d78379f5baaf193f96118d29c3a723f

closed 2023-09-26 by gtxaspec